intel/intel-device-plugins-for-kubernetes #55: qat_plugin: additional review comments

mythi, 2018-07-25:
I've tested the QAT plug-in on my host and read the code a little bit. Here's my observations/improvement ideas:

- while re-running the plug-in multiple times, I eventually run "out of resources" because the devices remain bound to the dpdk user IO driver. The driver should probably implement:
``` for i in `ls -d /sys/bus/pci/drivers/vfio-pci/0000:*`; do echo `basename $i` > /sys/bus/pci/drivers/vfio-pci/unbind; done``` (for the given ```dpdk-driver```) in its init
- ```maxNumdevice``` check looks wrong. 1) the loop gives ```maxNumdevice```from each kernel driver resulting N_drivers * ```maxNumdevice``` devices. 2) the loop continues for all files even if ```maxNumdevice``` is already reached

mythi, 2018-07-25:
@swatisehgal I wrote some review feedback still.

swatisehgal, 2018-08-03:
Thanks @mythi for your feedback!! There has been some code refactoring done recently which takes care of the maxNumdevice issue that you have pointed out. And as far as the out of resource issue is concerned, I will have a look into that to see what can be done to gracefully handle this case. You are right that this is tricker than unbind as we need to be aware that there might be devices bound to the dpdk driver configured with the intention to be used on the host and are not assigned to the kubernetes for device allocation. So randomly unbinding those devices to the kernel would not be correct.  I think for the time being we can update the README to mention this limitation while we work towards overcoming this.

mythi, 2018-08-06:
ping @MCamp859. FYI, see the proposed README updates.

@swatisehgal perhaps the plugin could log out the devices it binds for easier manual 'recovery'.

rojkov, 2018-08-06:
Is there a way to detect the devices that are already in use on the host so the plugin could exclude them from exposing?

mythi, 2018-08-06:
@rojkov the plugin already "excludes" them (them= host dpdk users): it only looks devices that are bound to the QAT kernel driver module and re-binds those to UIO/vfio-pci.

rojkov, 2018-08-06:
@mythi This isn't the case after my refactoring: now it looks for devices that are bound to kernel and DPDK drivers.

If the only indication of the fact the device is taken by some other process is that it's bound to uio/vfio-pci then we'd better add unbinding upon the plugin's exit. Though I can't say that's a perfect solution because I feel it's prone to device leakages.

rojkov, 2018-08-06:
@mythi True. My assumption was that we don’t care about the use case where some devices are taken by host processes. And the FPGA plugin has been developed without this use case in mind.

mythi, 2019-01-08:
Obsoleted by #145 and #146

closed 2019-01-08 by mythi