intel/intel-device-plugins-for-kubernetes #81: QAT: crypto-perf image fails to build

bart0sh, 2018-08-10:
Here is what I'v got when I tried to build crypto-perf image:

```
$ cd demo
$ ./build-image.sh crypto-perf
++ dirname ./build-image.sh
+ CWD=.
+ IMG=crypto-perf
+ '[' -z crypto-perf ']'
+ '[' '!' -d crypto-perf ']'
+ PROXY_VARS='http_proxy https_proxy'
+ BUILD_ARGS=
+ for proxy in '$PROXY_VARS'
+ '[' -v http_proxy ']'
++ tr -d ' '
++ echo http://proxy-chain.intel.com:911
+ val=http://proxy-chain.intel.com:911
+ BUILD_ARGS=' --build-arg http_proxy=http://proxy-chain.intel.com:911'
+ RUN_ARGS=' -e http_proxy=http://proxy-chain.intel.com:911'
+ for proxy in '$PROXY_VARS'
+ '[' -v https_proxy ']'
++ tr -d ' '
++ echo http://proxy-chain.intel.com:911
+ val=http://proxy-chain.intel.com:911
+ BUILD_ARGS=' --build-arg http_proxy=http://proxy-chain.intel.com:911 --build-arg https_proxy=http://proxy-chain.intel.com:911'
+ RUN_ARGS=' -e http_proxy=http://proxy-chain.intel.com:911 -e https_proxy=http://proxy-chain.intel.com:911'
+ docker build -t crypto-perf --build-arg http_proxy=http://proxy-chain.intel.com:911 --build-arg https_proxy=http://proxy-chain.intel.com:911 ./crypto-perf/
Sending build context to Docker daemon  2.048kB
Step 1/8 : FROM centos:7
 ---> 5182e96772bf
Step 2/8 : ENV SHELL /bin/bash
 ---> Using cache
 ---> 19e95b4e9030
Step 3/8 : COPY ./resolv.conf /etc/resolv.conf
COPY failed: stat /var/lib/docker/tmp/docker-builder406254565/resolv.conf: no such file or directory
```

kad, 2018-11-07:
@swatisehgal ping

swatisehgal, 2018-11-08:
Thanks for pinging me on this! We will assign resources to get this fixed.

przemeklal, 2018-11-20:
PR #122 should fix this and other potential issues (like missing directory with DPDK sources)