intel/intel-device-plugins-for-kubernetes #138: Support for GVT-g vGPUs

qmfrederik, 2018-12-04:
I'm interested in using [GVT-g](https://github.com/intel/gvt-linux/wiki/GVTg_Setup_Guide) to virtualize the GPUs in my cluster nodes and assign them to pods (essentially for use with qemu in [kubevirt](http://kubevirt.io)).

I came up with a very [naïve implementation](https://github.com/intel/gvt-linux/wiki/GVTg_Setup_Guide), and wanted to check with you whether there's any interest in making this part of the "official" device plugin?

mythi, 2018-12-05:
You should be able to experiment with https://github.com/intel/sriov-network-device-plugin where the ```resourceList``` is be configured to expose your graphics card to a vfio device.

I've not tried with kubevirt but I'm using a similar mechanism to isolate (QAT) devices to Kata Containers backed pods.

qmfrederik, 2018-12-05:
Thanks! I gave it a try, but I'm not entirely sure whether this is to work with virtualized GPUs, or would allow me to pass through an individual (physical) GPU?

I have created two vGPUs, which show up as `/dev/vfio/9` and `/dev/vfio/10`.
When I deploy the daemonset for the sriov-network-device-plugin, only one device is listed.
Creating a pod which uses this device fails: `Error: failed to start container "appcntr1": Error response from daemon: linux runtime spec devices: error gathering device information while adding custom device "/dev/vfio/1": no such file or directory`.

Here's the configuration file:
```json
{
    "resourceList":
    [
        {
            "resourceName": "gpu",
            "rootDevices": [ "00:02.0" ],
            "sriovMode": false,
            "deviceType": "vfio"
        }
    ]
}
```

And the output from describe pod:

```
I1205 13:36:09.720121      15 server.go:85] Allocate() called with &AllocateRequest{ContainerRequests:[&ContainerAllocateRequest{DevicesIDs:[0000:00:02.0],}],}
I1205 13:36:09.720239      15 vfioPool.go:76] vfio GetDeviceSpecs() called
I1205 13:36:09.720255      15 vfioPool.go:54] vfio GetEnvs() called
I1205 13:36:09.720272      15 vfioPool.go:70] vfio GetMounts() called
I1205 13:36:09.720279      15 server.go:94] AllocateResponse send: &AllocateResponse{ContainerResponses:[&ContainerAllocateResponse{Envs:map[string]string{PCIDEVICE_INTEL_COM_GPU: 0000:00:02.0,},Mounts:[],Devices:[&DeviceSpec{ContainerPath:/dev/vfio/vfio,HostPath:/dev/vfio/vfio,Permissions:mrw,} &DeviceSpec{ContainerPath:/dev/vfio/1,HostPath:/dev/vfio/1,Permissions:mrw,}],Annotations:map[string]string{},}],}
```

Let me know if missed something.

mythi, 2018-12-05:
@qmfrederik I was thinking about GVT-d so my idea won't work, sorry.

The link to your implementation points to GVT-g wiki.

qmfrederik, 2018-12-06:
@mythi Makes sense, thanks.

The link should point to https://github.com/kubedroid/intel-device-plugins-for-kubernetes/blob/vgpu/cmd/vgpu_plugin/vgpu_plugin.go, I've updated the comment.

msivosuo, 2021-09-14:
Inactive for several years, if still valid please reopen.

closed 2021-09-14 by msivosuo