intel/intel-device-plugins-for-kubernetes #147: can't build fpga_plugin

ldcastell, 2019-01-07:
What's the golang version I should use? I can't pass the fpga_plugin build: 
with golang 1.6 I get the following error:
```
../../internal/deviceplugin/server.go:18:2: cannot find the package "context" in any of:
$GOROOT
$GOPATH
```
With Golang 1.7 and 1.8 I get:
```
*server does not implement v1beta1.DevicePluginServer (wrong type for Allocate method)
		have Allocate("context".Context, *v1beta1.AllocateRequest) (*v1beta1.AllocateResponse, error)
		want Allocate("github.com/intel/intel-device-plugins-for-kubernetes/vendor/golang.org/x/net/context".Context, *v1beta1.AllocateRequest) (*v1beta1.AllocateResponse, error)
```

and With latest I get:
```
import cycle not allowed
package github.com/intel/intel-device-plugins-for-kubernetes/cmd/fpga_plugin
	imports flag
	imports errors
	imports runtime
	imports internal/bytealg
	imports internal/cpu
	imports runtime
Makefile:35: recipe for target 'fpga_plugin' failed
make: *** [fpga_plugin] Error 1
```

rojkov, 2019-01-08:
Hm... I can't reproduce the issue with the latest golang. In .travisci with put go-1.11.x as a runtime and it has been working so far. go-1.10 is insufficient because one of the FPGA components uses a templating feature available in go >= 1.11.

Did you try a Docker-based build with `make intel-fpga-plugin`?

ldcastell, 2019-01-09:
Golang >= 1.11 did the trick, thanks!

closed 2019-01-09 by ldcastell