intel/intel-device-plugins-for-kubernetes #148: Can’t run the demo on k8s node with two FPGAs

khrd, 2019-01-08:
Hi.
I'm trying the FPGA plugin and I have 2 questions related to a multi-FPGA k8s node.
If any one here can give me some hint I will be very appreciate.

I have a DELL R740 server with two Intel PAC (FPGA Board), and I use this server as a k8s node.
BTW, I also have a DELL R640 with single Intel PAC and I do not have problem with this one.

## Question1
I want to create two pod, each pod attached with one PAC.
But inside pod, I can see both two PAC with `lspci` command. 
Is this behavior correct? 
```
$ lspci  | grep accelerators
86:00.0 Processing accelerators: Intel Corporation Device 09c4
d8:00.0 Processing accelerators: Intel Corporation Device 09c4
```

## Question2
When I run the nlb3 of `ubuntu-demo-opae` with FPGA Card, 
the demo fails with `Error: device enumeration failed.` in af mode although the AFU id is correct.
And `Error: couldn't open device.` in both af and region mode.

The only way to run the demo I found is to change `af mode` to `region mode` and also specify the bus number like `./nbl3 -B 0x86`.

Is it required to specify the bus number on a multi-FPGA node?
If so, how do I know the allocated FPGA bus number from the pod? Because I can not tell which is the one with the `lspci` command. 

### pod creation
I modified the yaml and created my pod as following

- test-fpga-region2.yml (copy of test-fpga-region.yml)
```diff
-name: test-fpga-region
+name: test-fpga-region2
-command: ["sh", "/usr/bin/test_fpga.sh"]
+command: ["sleep", "3600"]
-fpga.intel.com/arria10-nlb3: 1 
+fpga.intel.com/arria10: 1　# To use region mode
```

### nlb3 execution result
I executed my pod as following

```
## execute nlb3  (region mode)

$ kubectl exec -ti test-fpga-region2 /usr/bin/nlb3
[22][WARN][accelerator::open] Errors encountered while opening accelerator resource
Error: couldn't open device.
command terminated with exit code 102

## execute nlb3 with Bus number option (region mode)

$ kubectl exec -ti test-fpga-region2 /usr/bin/nlb3 -- -B 0x86

Cachelines Read_Count Write_Count Cache_Rd_Hit Cache_Wr_Hit Cache_Rd_Miss Cache_Wr_Miss   Eviction 'Clocks(@200 MHz)'   Rd_Bandwidth   Wr_Bandwidth
         1          1           0            0            0             0             0          0              155     0.083 GB/s     0.000 GB/s

VH0_Rd_Count VH0_Wr_Count VH1_Rd_Count VH1_Wr_Count VL0_Rd_Count VL0_Wr_Count
           1            1            0            0            0            0

## execute nlb3  (af mode)
ubectl exec -ti test-fpga-region /usr/bin/nlb3
[74][WARN][accelerator::open] Errors encountered while opening accelerator resource
Error: couldn't open device.
command terminated with exit code 102

## execute nlb3 with Bus number option (af mode)
ubectl exec -ti test-fpga-region /usr/bin/nlb3 -- -B 0x86
Error: device enumeration failed.
Please make sure that the driver is loaded and that a bitstream for
AFU id: F7DF405C-BD7A-CF72-22F1-44B0B93ACD18 is programmed.
command terminated with exit code 102

```

### k8s status

```
$ kubectl get no
NAME     STATUS   ROLES    AGE   VERSION
r740     Ready    <none>   31d   v1.12.3
master   Ready    master   40d   v1.12.3

$ kubectl get po
NAME                                            READY   STATUS    RESTARTS   AGE
intel-fpga-webhook-deployment-98d745549-nf8v5   1/1     Running   0          17h

$ kubectl describe node r740 | grep fpga.intel.com
fpga.intel.com/device-plugin-mode: region
fpga.intel.com/region-9926ab6d6c925a68aabca7d84c545738:  2
fpga.intel.com/region-9926ab6d6c925a68aabca7d84c545738:  2

$ kubectl describe node r740 | grep hugepage
 hugepages-1Gi:                                       0
 hugepages-2Mi:                                       2Gi
 hugepages-1Gi:                                       0
 hugepages-2Mi:                                       2Gi
 
$ cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/afu_id
f7df405cbd7acf7222f144b0b93acd18

$ cat /sys/class/fpga/intel-fpga-dev.1/intel-fpga-port.1/afu_id
f7df405cbd7acf7222f144b0b93acd18
```

bart0sh, 2019-01-08:
> But inside pod, I can see both two PAC with lspci command. Is this behavior correct?

Yes, it's correct.
lspci output is not masked inside containers as far as I know at least. You can see all PCI devices in the output, not just FPGA accelerators.

bart0sh, 2019-01-08:
@khrd 
> When I run the nlb3 of ubuntu-demo-opae with FPGA Card, the demo fails with Error: device enumeration failed.

Can you run these commands inside the container and show me the output, please?
```
lsmod
ls -la /dev/intel-fpga*
```

khrd, 2019-01-09:
Thanks for the reply. 
here are the results(I tried both mode).

## inside the container (af mode)
### lsmod
```
# lsmod
Module                  Size  Used by
ipt_REJECT             12541  0
nf_reject_ipv4         13373  1 ipt_REJECT
veth                   13410  0
vxlan                  49241  0
ip6_udp_tunnel         12755  1 vxlan
udp_tunnel             14137  1 vxlan
xt_statistic           12601  2
xt_nat                 12681  7
xt_comment             12504  24
xt_mark                12563  5
ipt_MASQUERADE         12678  4
nf_nat_masquerade_ipv4    13412  1 ipt_MASQUERADE
nf_conntrack_netlink    40449  0
nfnetlink              14696  2 nf_conntrack_netlink
iptable_nat            12875  1
nf_conntrack_ipv4      15053  6
nf_defrag_ipv4         12729  1 nf_conntrack_ipv4
nf_nat_ipv4            14115  1 iptable_nat
xt_addrtype            12676  3
iptable_filter         12810  1
xt_conntrack           12760  5
nf_nat                 26787  3 nf_nat_ipv4,xt_nat,nf_nat_masquerade_ipv4
nf_conntrack          133387  6 nf_nat,nf_nat_ipv4,xt_conntrack,nf_nat_masquerade_ipv4,nf_conntrack_netlink,nf_conntrack_ipv4
br_netfilter           22209  0
bridge                136173  1 br_netfilter
stp                    12976  1 bridge
llc                    14552  2 stp,bridge
overlay                51863  10
rpcrdma                86152  0
sunrpc                348674  1 rpcrdma
ib_isert               50770  0
iscsi_target_mod      302966  1 ib_isert
ib_iser                47813  0
libiscsi               57233  1 ib_iser
scsi_transport_iscsi    99909  2 ib_iser,libiscsi
ib_srpt                48170  0
target_core_mod       367918  3 iscsi_target_mod,ib_srpt,ib_isert
ib_srp                 48454  0
scsi_transport_srp     20993  1 ib_srp
scsi_tgt               20027  1 scsi_transport_srp
ib_ipoib              110142  0
rdma_ucm               26841  0
ib_ucm                 22589  0
ib_uverbs              64636  2 ib_ucm,rdma_ucm
ib_umad                22080  0
rdma_cm                54426  4 rpcrdma,ib_iser,rdma_ucm,ib_isert
ib_cm                  47287  5 rdma_cm,ib_srp,ib_ucm,ib_srpt,ib_ipoib
iw_cm                  46260  1 rdma_cm
mlx5_ib               171306  0
ib_core               211874  14 rdma_cm,ib_cm,iw_cm,rpcrdma,mlx5_ib,ib_srp,ib_ucm,ib_iser,ib_srpt,ib_umad,ib_uverbs,rdma_ucm,ib_ipoib,ib_isert
iTCO_wdt               13480  0
iTCO_vendor_support    13718  1 iTCO_wdt
dcdbas                 14847  0
skx_edac               13258  0
edac_core              58151  1 skx_edac
intel_powerclamp       14419  0
vfat                   17461  1
coretemp               13444  0
fat                    65950  1 vfat
intel_rapl             19362  0
iosf_mbi               13523  1 intel_rapl
kvm_intel             170086  0
[... 200 lines cut ...]
igb                   205773  0
drm                   370825  4 ttm,drm_kms_helper,mgag200
ahci                   34042  0
libahci                31992  1 ahci
devlink                30193  1 mlx5_core
libata                238896  2 ahci,libahci
intel_fpga_pci         26519  2 intel_fpga_afu,intel_fpga_fme
megaraid_sas          149357  3
ptp                    19231  2 igb,mlx5_core
pps_core               19057  1 ptp
dca                    15130  1 igb
i2c_algo_bit           13413  2 igb,mgag200
i2c_core               40756  7 drm,igb,i2c_i801,ipmi_ssif,drm_kms_helper,mgag200,i2c_algo_bit
dm_mirror              22124  0
dm_region_hash         20813  1 dm_mirror
dm_log                 18411  2 dm_region_hash,dm_mirror
dm_mod                123303  10 dm_log,dm_mirror

```
### ls -la /dev/intel-fpga*
```
# ls -la /dev/intel-fpga*
crw-------. 1 root root 244, 0 Jan  9 00:24 /dev/intel-fpga-port.a

```

bart0sh, 2019-01-09:
@khrd  thanks. The output looks ok except of one thing: 
```
# ls -la /dev/intel-fpga*
crw-------. 1 root root 244, 0 Jan  9 00:24 /dev/intel-fpga-port.a
```

I have never seen this before. Usually port devices are named /dev/intel-fpga-port.N, where N is a number. Are you sure it's /dev/intel-fpga-port.a ?

Anyway, I'm not able to reproduce this issue on my setup. I have the same setup - 2 Arria 10 DCP1.1 cards on a  DELL R720 machine flashed with nlb3 green bitstream:

```
> cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/pr/interface_id 
9926ab6d6c925a68aabca7d84c545738
> cat /sys/class/fpga/intel-fpga-dev.1/intel-fpga-fme.1/pr/interface_id 
9926ab6d6c925a68aabca7d84c545738
> cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/afu_id
f7df405cbd7acf7222f144b0b93acd18
> cat /sys/class/fpga/intel-fpga-dev.1/intel-fpga-port.1/afu_id
f7df405cbd7acf7222f144b0b93acd18
```

The plugin is running in region mode:
```
> kubectl get pod -n kube-system
NAME                                             READY   STATUS    RESTARTS   AGE
coredns-86c58d9df4-g7hpn                         1/1     Running   4          24h
coredns-86c58d9df4-r42sp                         1/1     Running   4          24h
etcd-r720-1                                      1/1     Running   4          24h
intel-fpga-plugin-lncxv                          1/1     Running   2          20h
intel-fpga-webhook-deployment-58845f7bfc-sq22p   1/1     Running   1          20h
kube-apiserver-r720-1                            1/1     Running   1          20h
kube-controller-manager-r720-1                   1/1     Running   16         24h
kube-proxy-lzp2m                                 1/1     Running   4          24h
kube-scheduler-r720-1                            1/1     Running   15         24h
weave-net-rjxcb                                  2/2     Running   7          24h

> kubectl logs intel-fpga-plugin-lncxv -n kube-system
Overriding mode to  region
FPGA device plugin started in  region  mode
Start server for region-9926ab6d6c925a68aabca7d84c545738 at: /var/lib/kubelet/device-plugins/fpga.intel.com-region-9926ab6d6c925a68aabca7d84c545738.sock
Device plugin for region-9926ab6d6c925a68aabca7d84c545738 registered
```

Here is my modified spec. Please, note it differs from yours. You removed AFU id from the resource for some reason, I didn't.
```
> cat test-fpga-region2.yml 
apiVersion: v1
kind: Pod
metadata:
  name: test-fpga-region2
spec:
  containers:
  - name: test-container
    image: ubuntu-demo-opae:devel
    imagePullPolicy: IfNotPresent
    command: ["sleep", "1000000"]
    securityContext:
      capabilities:
        add:
          [IPC_LOCK]
    resources:
      limits:
        fpga.intel.com/arria10-nlb3: 1
        cpu: 1
        hugepages-2Mi: 20Mi

  restartPolicy: Never
```

Here is how I started the pod:
```
> kubectl create -f test-fpga-region2.yml 
pod/test-fpga-region2 created

> kubectl get pods
NAME                READY   STATUS    RESTARTS   AGE
test-fpga-region2   1/1     Running   0          13m
```

Here is how I run nlb3:
```
> kubectl exec test-fpga-region2 /usr/bin/nlb3

Cachelines Read_Count Write_Count Cache_Rd_Hit Cache_Wr_Hit Cache_Rd_Miss Cache_Wr_Miss   Eviction 'Clocks(@200 MHz)'   Rd_Bandwidth   Wr_Bandwidth
         1          1           0            0            0             0             0          0              140     0.091 GB/s     0.000 GB/s

VH0_Rd_Count VH0_Wr_Count VH1_Rd_Count VH1_Wr_Count VL0_Rd_Count VL0_Wr_Count 
           1            1            0            0            0            0 
```

As you see everything works as expected, which is good. We just need to figure out the difference between your and my setups.

My current idea is that you installed latest version(1.2) of kernel drivers. Our plugins have been tested with OPAE 1.1.0 from here https://github.com/OPAE/opae-sdk/releases/tag/1.1.0-2 

I'd propose you to do this:
- forget about af mode for now. The demo uses region mode and I have working setup in region mode. Let's concentrate on this until it works for you.
- show interface ids for both your cards here: ```cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/pr/interface_id  && cat /sys/class/fpga/intel-fpga-dev.1/intel-fpga-fme.1/pr/interface_id``` 
- Use my test-fpga-region2.yml as yours is not correct
- Make sure you have OPAE kernel drivers v 1.1.0-2 installed and loaded. Reinstall them if needed and power cycle your server just to be sure hardware is setup correctly.
- Try to repeat what I did
- Report here if it works for you or not. Please do it in either case.

Hope it helps,
Ed

khrd, 2019-01-10:
I appreciate your help with this.

Actually, the OPAE kernel driver v1.0.2-2 was installed.
So, I updated `opae-*-driver-1.1.0-2*.rpm` and `opae*1.1.0-2*.rpm`.
But, I found that one of FPGA has trouble ( It failed diagnostics with NLB_3 on the host machine).
I might it take a long to resolve this problem.
I'll try asking vender support.

Anyway, Thank you for your support!

kad, 2019-08-25:
This should be fixed with earlier #179 and now updated with #199

closed 2019-08-25 by kad