intel/intel-device-plugins-for-kubernetes #163: Support for Intel Movidius Neural Compute Stick?

gianlazz, 2019-02-27:
Any chance of seeing a plugin for supporting using the ncs with kubernetes? I'd really like to be able to use it in my cluster.

kad, 2019-03-21:
Do you have more than one Movidius NCS per node? Do you have also any other actively used USB devices on the same node?

tkhred, 2019-06-25:
Do you have a plan to offer the plugin for intel HDDL card (e.g. IEI Mustang) ?

ababushk, 2019-10-18:
It would be great to manage myriad sticks through kubernetes pod settings.
We're using Kubernetes with Jenkins Kubernetes plugin to run tests on CPU, GPU and Myriad, but for the latter we use 'hacky' solution implemented via Jenkins Pipeline lock() step because two apps can't use one myriad stick at the same time but we still want to run CPU or GPU test on a node to which NCS installed

ababushk, 2019-10-21:
Oops, I was wrong. It seems for my purpose Kubernetes extended resources functionality is enough (https://kubernetes.io/docs/tasks/administer-cluster/extended-resource-node/)

rojkov, 2020-01-24:
@tkhred wrt HDDL you might be interested in #273.

alekdu, 2020-01-25:
Yes, the HDDL should work for even single NCS stick. A kind of overkill, but it works.

tkhred, 2020-01-26:
Thanks.
I'll check #273

thkukuk, 2020-02-04:
HDDL seems to be more than overkill. It would be really nice to see direct support for a NCS2 stick.

mythi, 2020-02-04:
> HDDL seems to be more than overkill. It would be really nice to see direct support for a NCS2 stick.

From what I've understood it gives us device abstraction and deals with usb device resets etc. Would you have an idea how to deal with those in a device plugin and allow non-privileged workloads to use the device?

thkukuk, 2020-02-04:
> From what I've understood it gives us device abstraction and deals with usb device resets etc. 

But it is binary only.

pisymbol, 2020-03-01:
I have a Raspberry Pi 4 based cluster and a few NCS sticks. Can this be supported?

alekdu, 2020-03-03:
@pisymbol theoretically, yes. It should support. But you may need handle some NCS reset logic. The bsl_reset with the hddldaemon may need replaced with your own NCS reset code. I guess you have some GPIOs to reset NCS right ? The default bsl_reset support several GPIO reset method like, i2c GPIO expander, i2c HID driver.

alekdu, 2020-03-03:
The bsl_reset source code is available at: /opt/intel/openvino/deployment_tools/inference_engine/external/hddl/hddl-bsl/src
if you pull the hddldaemon container from: openvisualcloud/vcaca-ubuntu1804-analytics-hddldaemon

thkukuk, 2020-03-04:
> I have a Raspberry Pi 4 based cluster and a few NCS sticks. Can this be supported?

@pisymbol Works fine (https://github.com/thkukuk/demos/tree/master/Hackweek19), but unfortunately not with this plugin. I did build the Myriad plugin from the dldt source with "NO_BOOT" and wrote a cli to boot the NCS stick before starting kubelet/the containers.
https://github.com/thkukuk/myriadctl
Version 0.3 is based on dldt 2019.R3, the current head on dldt 2020.1

pisymbol, 2020-03-04:
@thkukuk Not to thread crap here but... so you installed dldt from source on a worker node and then advertised using extended resources (I see you just made something up to advertise the dongle with resource 1). Is that right?

Btw, feel free to email me (gmail.com) if you want to take this discussion offline. I'd really like to get my NCS stick to work with my cluster

mythi, 2021-06-17:
closing due to inactivity

closed 2021-06-17 by mythi