intel/intel-device-plugins-for-kubernetes #165: Dockerfile for openssl-qat-engine fails to build with debian:sid as builder but works with ubuntu:latest

eadamsintel, 2019-03-06:
I am not sure what exactly is going on but the ca-certificates package refuses to install.  Below at 1 is what the failure looks like using the Dockerfile as is. Below at 2 is when I modified the docker file to try to specify installing ca-certificates right after the apt-get update.  This failed on 3 different Clear Linux machines with builds from December and from March. These machines are all behind a proxy. However, another clear linux machine that is behind a transparent proxy (Don't need to set proxy variables) works fine for building.  If I change the Dockerfile so that the builder is from ubuntu:latest vs from debian:sid it builds just fine.  This worked for me a couple days ago with debian:sid so I have no idea what might have changed and can't find any recent issues from others on installing this package on debian. 

Note: When I manually specify to install the ca-certificates I get a lot of errors about TERM not set but if I set "ENV DEBIAN_FRONTEND="noninteractive"" then those errors don't come up. The image still fails to build from debian though.

**1**
```
Setting up libalgorithm-merge-perl (0.08-3) ...
Processing triggers for libc-bin (2.28-7) ...
Errors were encountered while processing:
 ca-certificates
E: Sub-process /usr/bin/dpkg returned an error code (1)
The command '/bin/sh -c apt-get update &&     apt-get install -y git build-essential wget libssl-dev openssl libudev-dev pkg-config autoconf autogen libtool &&     git clone https://github.com/intel/QAT_Engine &&     git clone -b OpenSSL_1_1_1-stable https://github.com/openssl/openssl.git &&     wget https://01.org/sites/default/files/downloads/intelr-quickassist-technology/$QAT_DRIVER_RELEASE.tar.gz &&     tar zxf $QAT_DRIVER_RELEASE.tar.gz' returned a non-zero code: 100
```

**2**
```
Unpacking libssl1.1:amd64 (1.1.1b-1) ...
Selecting previously unselected package openssl.
Preparing to unpack .../openssl_1.1.1b-1_amd64.deb ...
Unpacking openssl (1.1.1b-1) ...
Selecting previously unselected package ca-certificates.
Preparing to unpack .../ca-certificates_20190110_all.deb ...
Unpacking ca-certificates (20190110) ...
Setting up libssl1.1:amd64 (1.1.1b-1) ...
Setting up openssl (1.1.1b-1) ...
Setting up ca-certificates (20190110) ...
Updating certificates in /etc/ssl/certs...
mv: cannot move '/tmp/ca-certificates.crt.tmp.sL7u4e' to a subdirectory of itself, 'ca-certificates.crt'
dpkg: error processing package ca-certificates (--configure):
 installed ca-certificates package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.28-7) ...
Errors were encountered while processing:
 ca-certificates
E: Sub-process /usr/bin/dpkg returned an error code (1)
The command '/bin/sh -c apt-get install -y ca-certificates' returned a non-zero code: 100
```

mythi, 2019-03-06:
@eadamsintel I ran a fresh build and I'm unable to reproduce this. My log output is:

```
Setting up ca-certificates (20190110) ...
debconf: unable to initialize frontend: Dialog
debconf: (TERM is not set, so the dialog frontend is not usable.)
debconf: falling back to frontend: Readline
Updating certificates in /etc/ssl/certs...
128 added, 0 removed; done.
```

eadamsintel, 2019-03-06:
I just setup Clear Linux again and it happens again so I can consistently reproduce it.   I can give you credentials to check this out if you want.  It does build successfully on other machines but I now have 3 servers this fails on.

mythi, 2019-04-10:
@eadamsintel Is this still an issue or was it the runc vs kata setup you had?

eadamsintel, 2019-04-10:
This can be closed. Clear Linux configures Kata as the default runtime which was causing the issue.

mythi, 2019-04-10:
@eadamsintel thanks for the confirmation, closing

closed 2019-04-10 by mythi