intel/intel-device-plugins-for-kubernetes #166: OpenCL does not work on servers with two PACs

khrd, 2019-03-14:
Hi. Using this plug-in, 
I'm trying OpenVINO container with FPGA.

When run OpenVINO on my DELL R640 with one PAC,
I could execute inference with FPGA without any problem.

But, on my DELL R740 which has 2 PAC,
I could not execute OpenVINO inference or OpenCL memory diagnostics.

I attached some experiments log.
Please find it.

# log summary

please find detailed log of these 6 patterns below.

- DELL R640 (It has one PAC)
    - 0. on the pod
        - "It works"
       
- DELL R740 (It has two PAC)
    - 1. on the host
    - 2. on the pod (each pod is attached 1 PAC)
    - 3. on the pod (single pod is attached 2 PAC)
    - 4. on docker container without k8s (container is attached 2 PAC)
        - "It works, but 2 PAC are necessary..."
    - 5. on docker container without k8s (container is attached 1 PAC)

## enviroments

```
DELL R640 opae version 1.0.2-1
DELL R740 opae version 1.0.2-1
```

## R640

### 0. on the pod
It works
```
[kaz@r640-2 openvino]$ k apply -f openvino-pod.yaml
pod/openvino-pod created

[kaz@r640-2 openvino]$ k get po
NAME                                            READY   STATUS    RESTARTS   AGE
intel-fpga-webhook-deployment-98d745549-27cnx   1/1     Running   2          33d
openvino-pod-nfwrc                              1/1     Running   0          11s

[kaz@r640-2 openvino]$ k exec -ti openvino-job-nfwrc bash
root@openvino-pod-nfwrc:/#

root@openvino-pod-nfwrc:/# aocl diagnose
--------------------------------------------------------------------
Device Name:
acl0

BSP Install Location:
/opt/a10_gx_pac_ias_1_1_pv/opencl/opencl_bsp

Vendor: Intel Corp

Physical Dev Name   Status            Information

pac_a10_f300000     Passed            PAC Arria 10 Platform (pac_a10_f300000)
                                      PCIe 216:00.0
                                      FPGA temperature = 52 degrees C.

DIAGNOSTIC_PASSED
--------------------------------------------------------------------

[... 381 lines cut ...]
Error opening AFC: no driver available
--------------------------------------------------------------------
Device Name:
acl0

BSP Install Location:
/opt/a10_gx_pac_ias_1_1_pv/opencl/opencl_bsp

Vendor: Intel Corp

Physical Dev Name   Status            Information

pac_a10_ee00001     Passed            PAC Arria 10 Platform (pac_a10_ee00001)
                                      PCIe 216:00.0
                                      FPGA temperature = 43 degrees C.

DIAGNOSTIC_PASSED
--------------------------------------------------------------------

Call "aocl diagnose <device-names>" to run diagnose for specified devices
Call "aocl diagnose all" to run diagnose for all devices
root@3d83eeca62e8:/# aocl diagnose all
Error opening AFC: no driver available
Error opening AFC: no driver available
```

bart0sh, 2019-03-14:
Thank you for reporting the issue. Can you provide more info on your configuration, please? Which version of Acceleration Stack for Runtime do you use? Does the error persist with the latest OPAE version?

khrd, 2019-03-18:
Thank you for replying. I tried `opae v1.0.2-1` and `opae v1.1.0-2`.
Scheduling has succeeded, pod will also start. 
However, OpenCL commands do not work.

kad, 2019-08-25:
With PR #199 merged, this should work with OpenCL bitstreams as well in multi-card configuration.

bart0sh, 2019-08-26:
@kad, @khrd Should we close this issue then?

kad, 2019-08-26:
@bart0sh We need to verify it in our Lab, and it would be good if @khrd can verify that in their setup as well. But in theory, yes, it should be closed.

khrd, 2019-09-03:
No problem.
I will open this issue again when I find error.
thank you!

closed 2019-09-03 by khrd

belanasaikiran, 2021-12-03:
> Hi. Using this plug-in, I'm trying OpenVINO container with FPGA.
>
> When run OpenVINO on my DELL R640 with one PAC, I could execute inference with FPGA without any problem.
>
> But, on my DELL R740 which [...]

can you please share the dockerfile?

mythi, 2021-12-04:
> Has this issue been resolved? can you please share the Dockerfile ?

@belanasaikiran The original reporter was OK closing. I'd first like to turn the question back to you: do you see failures or errors?

belanasaikiran, 2021-12-06:
> > Has this issue been resolved? can you please share the Dockerfile ?
> 
> @belanasaikiran The original reporter was OK closing. I'd first like to turn the question back to you: do you see failures or errors?

Yes @mythi , I was not able to program the FPGAs through Docker Container. It shows the FPGAs available with `fpgainfo fme ` command. 
```
root@69b7aac9f145:/# fpgainfo fme
Board Management Controller, microcontroller FW version 26895
Last Power Down Cause: POK_CORE
Last Reset Cause: External reset
Power-on-reset
//****** FME ******//
Object Id                     : 0xED00001
PCIe s:b:d:f                  : 0000:AF:00:0
Device Id                     : 0x09C4
Socket Id                     : 0x00
Ports Num                     : 01
Bitstream Id                  : 0x124000200000367
Bitstream Version             : 1.2.4
Pr Interface Id               : 38d782e3-b612-5343-b934-2433e348ac4c
Boot Page                     : user
Board Management Controller, microcontroller FW version 26895
Last Power Down Cause: POK_CORE
Last Reset Cause: External reset
Power-on-reset
//****** FME ******//
Object Id                     : 0xED00000
PCIe s:b:d:f                  : 0000:86:00:0
Device Id                     : 0x09C4
Socket Id                     : 0x00
Ports Num                     : 01
Bitstream Id                  : 0x124000200000367
Bitstream Version             : 1.2.4
Pr Interface Id               : 38d782e3-b612-5343-b934-2433e348ac4c
Boot Page                     : user

```

But when I use `aocl diagnose`  or program the FPGA with `aocl program <device_id> hello_world.aocx ` , The following errors occur. 

```
root@69b7aac9f145:~/intelrtestack# aocl diagnose
--------------------------------------------------------------------
ICD System Diagnostics                                              
--------------------------------------------------------------------

Using the following location for ICD installation: 
        /etc/OpenCL/vendors

Found 1 icd entry at that location:
        /etc/OpenCL/vendors/Altera.icd

the following OpenCL libraries are referenced in the icd files:
        libalteracl.so

checking LD_LIBRARY_PATH for registered libraries:
        libalteracl.so was registered on the system at /opt/opencl_rte/aclrte-linux64/host/linux64/lib

Using the following location for fcd installations:
        /opt/Intel/OpenCL/Boards

Found 1 fcd entry at that location:
        /opt/Intel/OpenCL/Boards/dcp_bsp.fcd

the following OpenCL libraries are referenced in the fcd files:
        /root/intelrtestack/a10_gx_pac_ias_1_2_1_pv/opencl/opencl_bsp/linux64/lib/libintel_opae_mmd.so

checking LD_LIBRARY_PATH for registered libraries:
        /root/intelrtestack/a10_gx_pac_ias_1_2_1_pv/opencl/opencl_bsp/linux64/lib/libintel_opae_mmd.so was registered on the system.

ERROR: OpenCL host failed
--------------------------------------------------------------------
ICD diagnostics FAILED                                              
--------------------------------------------------------------------
--------------------------------------------------------------------
BSP Diagnostics                                                     
--------------------------------------------------------------------
Error opening AFC: no driver available
Error opening AFC: no driver available
Error opening AFC: no driver available
--------------------------------------------------------------------
Device Name:
acl0

BSP Install Location:
/root/intelrtestack/a10_gx_pac_ias_1_2_1_pv/opencl/opencl_bsp

Vendor: Intel Corp

DIAGNOSTIC_PASSED
--------------------------------------------------------------------
Error opening AFC: no driver available
--------------------------------------------------------------------
Device Name:
acl1

BSP Install Location:
/root/intelrtestack/a10_gx_pac_ias_1_2_1_pv/opencl/opencl_bsp

Vendor: Intel Corp

DIAGNOSTIC_PASSED
--------------------------------------------------------------------

Call "aocl diagnose <device-names>" to run diagnose for specified devices
Call "aocl diagnose all" to run diagnose for all devices

```

bart0sh, 2021-12-07:
@belanasaikiran FPGA programming is done in a different way. The plugin doesn't expose FME devices inside a container for security reasons. That's why you can't program devices directly by aocl or similar tools inside a container.
The programming could only be done in [region mode](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/fpga_plugin/README.md#fpga-modes). Please, follow the above setup instructions and let us know if that works for you.