intel/intel-device-plugins-for-kubernetes #186: Use FPGA in container

Huihuihh, 2019-06-05:
I can use FPGA in a  physical machine with opencl, but I can not use it  the same way in
 container.  And I see, I can not aocl program a aocx file in container. Can someone tell me how to use FPGA in a container, whether it doesn't make sense to use it in containers. If not, can  give me a artificial intelligence demo to use FPGA in container

bart0sh, 2019-06-05:
Can you provide a little bit more info about your setup, please? What kind of FPGA hardware, kubernetes version, distro, CRI-O version do you use? How did you setup Intel FPGA device plugin and what kind of errors are you getting/how exactly it doesn't work for you?

You can find a screencast demonstrating setup and usage of FPGA plugin in a kubernetes cluster inside demo/ directory: https://github.com/intel/intel-device-plugins-for-kubernetes/tree/master/demo#intel-fpga-device-plugin-demo-video. The demo shows how to use FPGA plugin in orchestration-programmed mode. It also demonstrates programming and usage of GZIP aocx bitstream in a kubernetes environment.

bart0sh, 2019-06-18:
closed due to inactivity

closed 2019-06-18 by bart0sh