intel/intel-device-plugins-for-kubernetes #196: Any chance you of Intel RealSense device plugin?

mv78, 2019-06-19:
We are currently building solution on top of Kubernetes and would like to plugin RealSesne cameras all the way into container. It would be really good to see support for USB devices like RealSense.

kad, 2019-08-01:
Can you share a bit more details?
E.g. which version of the RealSense hardware you're using?
Have you already tried to use RealSense with e.g. Docker command line?

mv78, 2019-08-02:
We are using Intel® RealSense™ Depth Camera D435i
Yes, using librealsense inside docker by using hostpath mount.
However, would like to plug USB device as device plugin and utilize Kubernete heals check mechanism to monitor and recover camera if camera fails etc.

kad, 2019-08-05:
Can you share of your current yaml file with hostpath mounts?

mv78, 2019-08-07:
We have not yet tried it in the Kubernetes POD; however, I assume it would not be difficult to do given we can use librealsense drivers from inside of the Docker container in this way:
docker run -it --privileged -v /dev:/dev ...
The same would equate to elevated privilege container and r/w hostpath to /dev

kad, 2019-08-22:
Well, running it in privileged mode and especially exposing host OS /dev to workload inside container is not secure. Practically it means that malicious software inside container can do whatever with host OS and any other workloads. That's why I was curious on what is minimal required privileges for realsense camera. I unfortunately don't have that hardware in hand to try it myself.

danfairs, 2020-01-22:
For what it's worth - I'm also interested in this, and have things working in a container much as @mv78 did (albeit mounting `/dev/bus/usb`, not the whole of `/dev`. Running with `--privileged` is certainly not ideal of course, but in our use case we control the whole stack so is not a showstopper.

HugoHJH, 2020-04-11:
> For what it's worth - I'm also interested in this, and have things working in a container much as @mv78 did (albeit mounting `/dev/bus/usb`, not the whole of `/dev`. Running with `--privileged` is certainly not ideal of course, but in our use case we control the whole stack so is not a showstopper.
> 
> I'll try to find some time over the next couple few weeks to pin down exactly which privileges are required.

Did you have any luck working out what privileges are required?

AndreV84, 2020-05-27:
hi guys,
is there any way to run 435i from within docker container without using the privileged mode?

msivosuo, 2021-06-17:
Currently no plans to implement realsense plugin. Contributions are welcome.

closed 2021-06-17 by msivosuo