intel/intel-device-plugins-for-kubernetes #198: demo/ubuntu-demo-opencl  looks like can't work

shaoyuta, 2019-06-24:
Hi,
I try to build  demo/ubuntu-demo-opencl, and use it to test gpu_plugin. But it looks like that this demo can't work correctly.
1) If only run this image using docker in host os, this app crashed.
$ docker run -it --rm -v /dev:/dev -v /etc:/etc -v /opt:/opt --privileged ubuntu-demo-opencl:devel
# ./run-opencl-example.sh /root/6-1/fft
+ WORK_DIR=/root/6-1/fft
+ cd /root/6-1/fft
+ ./fft
Segmentation fault (core dumped)

and at the same time, It looks like "clinfo" could not get the gpu infomation.
# clinfo 
Number of platforms                               0

2) If copy "/root/6-1/fft/*" to the host environment and execute it, the app can work correctly.

I'm not sure whether there are something is wrong in my steps. Or need to do some extra setup operations?

rojkov, 2019-06-26:
@taotriad could you please take a look?

shaoyuta, 2019-07-04:
Hi @taotriad, can you reproduce this issue? or give me some advice about it ?
Thanks.

dweineha, 2019-07-04:
I haven't tried the demo; we successfully use the plugin on our test cluster though, so if there's a bug it's likely in the demo rather than the plugin.

But if clinfo cannot get your GPU info it seems that the issue is further down in the stack; is your OpenCL function properly, and does it support the hardware you have?

rojkov, 2019-07-04:
I would rather not mount your host's `/etc` and `/opt` to a container, because the demo image have its own OpenCL runtime installed in these directories which may differ from what you have in your host (e.g. beignet on the host and non-beignet in the container).

dweineha, 2019-07-04:
I think in general you should *never* mount the host's /etc in your container. There are cases where mounting /opt might makes sense.

eero-t, 2019-07-09:
I think simplest test for GPU access is  this Dockerfile (named "test-vainfo" in run example below):
```
FROM ubuntu:18.04
RUN apt-get update && apt-get -y install vainfo
```
i965 media/VAAPI GPU driver gets pulled in to the container as "vainfo" dependency .  That supports HW from Haswell to Coffeelake (whereas new iHD driver available from Github would support HW from BDW upwards).

> $ docker run -it --rm -v /dev:/dev -v /etc:/etc -v /opt:/opt --privileged ubuntu-demo-opencl:devel

This worked as expected. Container prevented GPU device writes when that wasn't explicitly enabled.

**If you run docker directly instead of using Kubernetes and GPU plugin, you need to do same thing as what the GPU plugin does, i.e. provide GPU device access to the container** (you don't need to mount /dev or run container in privileged mode).

Like this:
`$ docker run -it --rm --device /dev/dri/renderD128:/dev/dri/renderD128 -t test-vainfo vainfo`

If GPU access works, vainfo prints out what HW you have and available VA-API profiles.  If you remove the device mapping, i.e. prevent GPU access, vainfo doesn't detect what HW is in question and doesn't print any info.

> I think in general you should never mount the host's /etc in your container. There are cases where mounting /opt might makes sense

Sometimes it's useful to mount specific files from host /etc  to containers (e.g. passwd and/or group file so that names for numeric user & group IDs match host).

rojkov, 2019-08-01:
I'm closing this issue as it seems to be caused by misuse of docker volumes, not a problem in the plugin. Please reopen if correct volumes don't solve the problem for you.

closed 2019-08-01 by rojkov