intel/intel-device-plugins-for-kubernetes #201: intel-fpga-plugin dial tcp i/o timeout

Antsypc, 2019-07-30:
intel-fpga-plugin pod get error
```shell
$ kubectl logs intel-fpga-plugin-5dppj -n kube-system
ERROR: Get https://110.1.0.1:443/api/v1/nodes/node10: dial tcp 110.1.0.1:443: i/o timeout
```

I created Kubernetes cluster as follow. There are two worker nodes in the cluster, node09(110.1.1.109), node10(110.1.1.110)
```shell
kubeadm init --kubernetes-version=1.14.2 \
  --apiserver-advertise-address=110.1.1.108  \
  --image-repository registry.aliyuncs.com/google_containers  \
  --service-cidr=110.1.0.0/16  --pod-network-cidr=110.244.0.0/16
```

I created intel-fpga-plugin DaemonSet as [intel-device-plugins-for-kubernetes/cmd/fpga_plugin](https://github.com/intel/intel-device-plugins-for-kubernetes/tree/master/cmd/fpga_plugin). And I finished all the guides: FPGA device plugin, FPGA admission controller webhook, FPGA prestart CRI-O hook.

However, logs in intel-fpga-plugin pods and intel-fpga-webhook pod get errors. Theses pods try to access https://110.1.0.1:443, but there is no such a pod with this ip.

Could you guys help me figure out how to fix it? Am I wrong to create cluster with wrong `service-cidr`?

kad, 2019-07-30:
What does your "kubectl get nodes" shows? Do you have node in Ready state and do you have networking properly running between your nodes?

110.1.0.1 is from your service range. You can see if it matches with output of `kubectl -n kube-system get service" for `intel-fpga-webhook-svc`.

Another possibility for this timeout errors might be with proxy configuration. Do you have something in environment variables set regarding proxy before running kubeadm?

kad, 2019-08-25:
No activity. Closing. Please re-open if problem persists.

closed 2019-08-25 by kad