intel/intel-device-plugins-for-kubernetes #205: Wrong FPGA device plugin af/region ID

Antsypc, 2019-08-22:
I followed the doc [cmd/fpga_plugin](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/cmd/fpga_plugin/README.md).

Run FPGA device plugin as administrator in `af` mode
```shell
# kubectl describe node node08 node09 node10 | grep fpga.intel.com
                    fpga.intel.com/device-plugin-mode: af
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
  fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18      0           0
  fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee  0           0
                    fpga.intel.com/device-plugin-mode: af
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
  fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18      0          0
  fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee  0          0
                    fpga.intel.com/device-plugin-mode: af
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
  fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18      0          0
  fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee  0          0
```

the `af` ID is `af-f7df405cbd7acf7222f144b0b93acd18`, BUT when I run the [demo](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/demo/test-fpga-region.yml) and replaced `fpga.intel.com/arria10-nlb3` to `fpga.intel.com/arria10-nlb0`. I got
```shell
[root@node08 demo]# kubectl apply -f test-fpga-region-ypc.yml
pod/test-fpga-region-ypc2 created
[root@node08 demo]#
[root@node08 demo]# kubectl describe pod test-fpga-region-ypc2
Name:               test-fpga-region-ypc2
Namespace:          default
Priority:           0
PriorityClassName:  <none>
Node:               <none>
Labels:             <none>
Annotations:        kubectl.kubernetes.io/last-applied-configuration:
                      {"apiVersion":"v1","kind":"Pod","metadata":{"annotations":{},"name":"test-fpga-region-ypc2","namespace":"default"},"spec":{"containers":[{...
Status:             Pending
IP:
Containers:
  test-container:
    Image:      ubuntu-demo-opae:devel
    Port:       <none>
    Host Port:  <none>
    Command:
      sh
      /usr/bin/test_fpga.sh
    Limits:
      cpu:                                                 1
      fpga.intel.com/af-d8424dc4a4a3c413f89e433683f9040b:  1
      hugepages-2Mi:                                       20Mi
    Requests:
      cpu:                                                 1
      fpga.intel.com/af-d8424dc4a4a3c413f89e433683f9040b:  1
      hugepages-2Mi:                                       20Mi
    Environment:                                           <none>
    Mounts:
      /var/run/secrets/kubernetes.io/serviceaccount from default-token-9mrqb (ro)
Conditions:
  Type           Status
  PodScheduled   False
Volumes:
  default-token-9mrqb:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  default-token-9mrqb
    Optional:    false
QoS Class:       Burstable
Node-Selectors:  <none>
Tolerations:     node.kubernetes.io/not-ready:NoExecute for 300s
                 node.kubernetes.io/unreachable:NoExecute for 300s
Events:
  Type     Reason            Age   From               Message
  ----     ------            ----  ----               -------
  Warning  FailedScheduling  3s    default-scheduler  0/3 nodes are available: 3 Insufficient fpga.intel.com/af-d8424dc4a4a3c413f89e433683f9040b, 3 Insufficient hugepages-2Mi.
```

Such situation appeared when I use `region` mode too. My region id is `fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee`. BUT when I run the demo, reported that it requests `region-9926ab6d6c925a68aabca7d84c545738`

So how to set right configuration in yaml. 
```yaml
spec:
  containers:
  - name: test-container
    ...
    resources:
      limits:
        fpga.intel.com/arria10-nlb3: 1  # here
        cpu: 1
        hugepages-2Mi: 20Mi
```

Thank you!

bart0sh, 2019-08-22:
Hi,

From the first look you might be using different FPGA firmware. Can you show the output of the below commands. I assume that you've programmed the device with nlb0 bitstream.

```
$ cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/pr/interface_id 
$ cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/afu_id 
$ kubectl describe node <node name> |grep -A10 Allocatable:
```

bart0sh, 2019-08-22:
@Antsypc I can already see from the issue description that your region id is a9f2d0f3b39857b0b34fd226bf364fee

Currently plugin supports only dcp1.0 and dcp1.1 interface ids. You can see them in deployments/fpga_admissionwebhook/mappings-collection.yaml

We'd be happy to accept PR implementing support for other dcp versions.

bart0sh, 2019-08-22:
@Antsypc btw, which FPGA device model are you using?

Antsypc, 2019-08-22:
@bart0sh Thanks for your help! I'm really appreciate that!

Here are outputs
```shell
[root@node08 ~]# cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-fme.0/pr/interface_id
a9f2d0f3b39857b0b34fd226bf364fee

[root@node08 ~]# cat /sys/class/fpga/intel-fpga-dev.0/intel-fpga-port.0/afu_id
f7df405cbd7acf7222f144b0b93acd18

[root@node08 ~]# kubectl describe node node08 | grep -A10 Allocatable
Allocatable:
 cpu:                                                     56
 ephemeral-storage:                                       451175774690
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      0
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
 hugepages-1Gi:                                           0
 hugepages-2Mi:                                           0
 memory:                                                  230469272Ki
 nvidia.com/gpu:                                          0
 pods:                                                    110
System Info:

[root@node08 ~]# kubectl describe node node09 | grep -A10 Allocatable
Allocatable:
 cpu:                                                     56
 ephemeral-storage:                                       451175774690
 fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:      1
 fpga.intel.com/region-a9f2d0f3b39857b0b34fd226bf364fee:  0
 hugepages-1Gi:                                           0
 hugepages-2Mi:                                           0
 memory:                                                  230475168Ki
 nvidia.com/gpu:                                          2
 pods:                                                    110
System Info:
[root@node08 ~]#
```

I'm not sure what do you mean `FPGA device model`, how can I get that information from my centOS7. How can I get version of dcp interface ids of my devices? Sorry for my poor knowledge.

Thanks again!

bart0sh, 2019-08-22:
@Antsypc Interface Id of your device is a9f2d0f3b39857b0b34fd226bf364fee. You can add it to mappings-collection.yaml and apply the changes with ```kubectl apply -f mappings-collection.yaml```. That should make the mapping working.

By device model I mean FPGA hardware model, e.g. Arria 10 or Stratix 10.

kad, 2019-08-22:
According to release notes, this interface ID seems to belong to 2.0 (Pre-Beta) Statix10 (d5005) PAC card: https://www.intel.com/content/www/us/en/programmable/documentation/edj1542148561811.html#pao1544211266875

kad, 2019-08-25:
@Antsypc  you can check examples of updating mappings for different cards in https://github.com/intel/intel-device-plugins-for-kubernetes/pull/199/commits/6395b1626d1e82754de6ae4a884a732dbc3d163b

closed 2019-09-14 by kad