intel/intel-device-plugins-for-kubernetes #225: VCA1(VCA1283LVV) support？

mdjxyf, 2019-09-20:
I installed the plug-in in the VCA card , but when I started gpu_plugins, I saw the following error through debug mode:
[image]

Is VCA1 supported?

kad, 2019-09-20:
From debug output that you shown, yes, device plugin was able to detect correctly `card0`.
Do you observe any issue with it if your workload tries to use the GPU?

mdjxyf, 2019-09-23:
Thank, Kad.  Since I am new to the VCA card, do you have any suggestions for me to verify if the GPU is working?

mdjxyf, 2019-09-25:
Each NODE node on VCA card communicates with NAT. Can it be bridged? If not, how does the node on VCA communicate with other physical hosts?

kad, 2019-09-26:
@mdjxyf to verify that it is working, you can do `kubectl describe node XXXX` where your plugin runs. you should see resources exposed by plugin. 
For network question, unfortunately I'll not be able to help much here, haven't tried myself this particular version of the card. In theory, on the host OS, you can do bridge from your VCA "veth" to host's physical ethernet.

mdjxyf, 2019-10-25:
Thank , Kad . On the node of VCA, install the node program of k8s, but after installing docker, the node cannot be connected through SSH after rebooting. I guess it's because docker has modified the network part, but I don't know where it is. Do you know?

mythi, 2021-06-17:
it seems the GPU plugin was working correctly, closing but please reopen if still valid

closed 2021-06-17 by mythi