intel/intel-device-plugins-for-kubernetes #233: Current doc is all about how to deploy a device plugin. Could you please add more description about the device and how to use the plugin?

LuyaoZhong, 2019-11-19:


mythi, 2019-11-19:
@LuyaoZhong demo folder has several application deployments that request resources provided by the plugins. The screencast videos linked in the README might also help.

mythi, 2019-12-10:
@LuyaoZhong is this still valid?

LuyaoZhong, 2019-12-12:
@mythi Sorry for this late reply. I mean could you please add more description about what every step is doing(e.g. for FPGA, what is admission webhook doing, what is CRI-O hook doing), some basic notions about the devices, and some useful links which can help newbies/me. :D

msivosuo, 2020-01-13:
In addition we should:
- explain the two drivers, two modes
- Update the screencast.

msivosuo, 2020-01-14:
Updated screencast: https://github.com/intel/intel-device-plugins-for-kubernetes/pull/262

grahamwhaley, 2020-01-15:
I also think we could do with a bunch more info on the front page of the repo describing more of what the framework is and how it plugs into k8s (personally I like diagrams as well).
The way the ToC entries for the sub-components (fpga, gpu etc.) leap off to other documents further down the tree is really strange as well - the ToC should point to sections on the main page, that then probably have a brief description and then leap off to the sub-section.

We need to make the front (landing) page more informative and easy for folks to leap off from - it is the first thing they will see when they land in the repo.

grahamwhaley, 2020-01-21:
See #268 for improvements to FPGA documentation.

msivosuo, 2020-01-29:
@kad, @rojkov -could you take a look at #268, thanks!

msivosuo, 2020-02-12:
Closing as this is now addressed, @LuyaoZhong please reopen or create a new issue if you still feel there are gaps.

closed 2020-02-12 by msivosuo