intel/intel-device-plugins-for-kubernetes #254: Kustomize

msivosuo, 2020-01-13:
Add kustomization for deployments.

askervin, 2020-01-29:
More detailed goals:

1. qat plugin to be deployable with `kubectl apply -k qat_plugin/VERSION` (`VERSION` being devel or a release tag)

2. gpu plugin to be deployable with `kubectl apply -k gpu_plugin/...`, possibly configurable command line parameters if needed

3. fpga plugin to be deployable with `kubectl apply -k fpga_plugin/...`, different kustomizations for different run modes (preprogrammed vs orchestrated)

askervin, 2020-01-29:
Reasoning for the work:
- provide end users an easy and consistent way to deploy the plugins and add their own customizations
- improve deployment maintainability by removing duplication

mythi, 2020-01-29:
@askervin FYI, deployments from the repo works:

```
kubectl apply -k github.com/intel/intel-device-plugins-for-kubernetes.git/deployments/qat_dpdk_app/test-crypto1?ref=master
```

mythi, 2020-01-30:
> (VERSION being devel or a release tag)

This may not be necessary since we can already do

```
$ kubectl kustomize github.com/intel/intel-device-plugins-for-kubernetes.git/deployments/qat_dpdk_app/test-crypto1?ref=release-0.15
$ kubectl kustomize github.com/intel/intel-device-plugins-for-kubernetes.git/deployments/qat_dpdk_app/test-crypto1?ref=master
```

grahamwhaley, 2020-01-30:
I did consider doing the direct-from-URL deploys in the docs updates, which in the case of using the default images and setup would avoid pulling the sources down. But, one change at a time....

askervin, 2020-02-07:
PRs contributing to this issue:
- QAT plugin - PR #286
- GPU plugin - PR #288

closed 2020-02-28 by mythi