intel/intel-device-plugins-for-kubernetes #259: fpga_plugin is failing to get deployed

hectoragb, 2020-01-13:
System summary: 

```sudo swupd info
Distribution:      Clear Linux OS
Installed version: 32070
Version URL:       https://cdn.download.clearlinux.org/update
Content URL:       https://cdn.download.clearlinux.org/update
kubectl version
Client Version: version.Info{Major:"1", Minor:"17", GitVersion:"v1.17.0", GitCommit:"70132b0f130acc0bed193d9ba59dd186f0e634cf", GitTreeState:"archive", BuildDate:"2019-12-10T21:58:38Z", GoVersion:"go1.13.5", Compiler:"gc", Platform:"linux/amd64"}
Server Version: version.Info{Major:"1", Minor:"17", GitVersion:"v1.17.0", GitCommit:"70132b0f130acc0bed193d9ba59dd186f0e634cf", GitTreeState:"clean", BuildDate:"2019-12-07T21:12:17Z", GoVersion:"go1.13.4", Compiler:"gc", Platform:"linux/amd64"}
```

In order to reproduce the issue, the following steps are used:

```export GOPATH=/home/clear/go
export PATH=$PATH:/usr/local/go/bin:$GOPATH/bin
go get -u github.com/cloudflare/cfssl/cmd/cfssl
cd scripts ; ./webhook-deploy.sh --namespace kube-system --mode orchestrated # Deploy webhook
kubectl create -f ${REPO_ROOT}/deployments/fpga_plugin/fpga_plugin_service_account.yaml # Create service account for the plugin 
kubectl annotate node --all 'fpga.intel.com/device-plugin-mode=region' # Set region mode for the plugin 
kubectl create -f ${REPO_ROOT}/deployments/fpga_plugin/fpga_plugin.yaml # Create plugin daemonset 
kubectl wait --for=condition=Ready pod --all --timeout=5m --namespace kube-system && sleep 3m
```
 
`kubectl create -f /home/clear/workspace/bmaas-test-intel-device-plugins-kubernetes/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/fpga_plugin.yaml` is used to deploy the fpga plugin. However, after a while the command `kubectl get pods -n kube-system` return something similar to the following 

```NAME                                                                          READY   STATUS                 RESTARTS   AGE
canal-5dw45                                                                   2/2     Running                0          8m58s
coredns-6955765f44-55wjn                                                      1/1     Running                0          8m58s
coredns-6955765f44-smqjh                                                      1/1     Running                0          8m58s
etcd-device-plugins-kubernetes-clearlinux-14-4.novalocal                      1/1     Running                0          9m15s
intel-fpga-plugin-7r8db                                                       0/1     CreateContainerError   0          4m27s
intel-fpga-webhook-deployment-84c48784c-jd7nm                                 1/1     Running                0          4m29s
kube-apiserver-device-plugins-kubernetes-clearlinux-14-4.novalocal            1/1     Running                0          9m15s
kube-controller-manager-device-plugins-kubernetes-clearlinux-14-4.novalocal   1/1     Running                0          9m15s
kube-proxy-mv2ld                                                              1/1     Running                0          8m58s
kube-scheduler-device-plugins-kubernetes-clearlinux-14-4.novalocal            1/1     Running                0          9m15s
```

By inspecting the pod using `kubectl describe pod intel-fpga-plugin-7r8db -n kube-system`, the following output is  got 

```Name:         intel-fpga-plugin-7r8db
Namespace:    kube-system
Priority:     0
Node:         device-plugins-kubernetes-clearlinux-14-4.novalocal/192.168.185.55
Start Time:   Mon, 13 Jan 2020 18:02:39 +0000
Labels:       app=intel-fpga-plugin
              controller-revision-hash=78cc488ccd
              pod-template-generation=1
Annotations:  cni.projectcalico.org/podIP: 10.244.0.5/32
Status:       Pending
IP:           10.244.0.5
IPs:
  IP:           10.244.0.5
Controlled By:  DaemonSet/intel-fpga-plugin
Init Containers:
  intel-fpga-initcontainer:
    Container ID:   cri-o://246d98949db65e20b34b74b43b63f54a842218a3f3a54148b13c5184b4d4e77d
    Image:          intel/intel-fpga-initcontainer:devel
    Image ID:       cloud-native-image-registry.westus.cloudapp.azure.com/intel/intel-fpga-initcontainer@sha256:3a790b3e3a25db65d5064e8d13d2cdfc4773e2cf79bb6134e30511f075756d76
    Port:           <none>
    Host Port:      <none>
    State:          Terminated
      Reason:       Completed
      Exit Code:    0
      Started:      Mon, 13 Jan 2020 18:04:56 +0000
      Finished:     Mon, 13 Jan 2020 18:04:56 +0000
    Ready:          True
    Restart Count:  0
    Environment:    <none>
[... 57 lines cut ...]
                 node.kubernetes.io/not-ready:NoExecute
                 node.kubernetes.io/pid-pressure:NoSchedule
                 node.kubernetes.io/unreachable:NoExecute
                 node.kubernetes.io/unschedulable:NoSchedule
Events:
  Type     Reason     Age               From                                                          Message
  ----     ------     ----              ----                                                          -------
  Normal   Scheduled  4m29s             default-scheduler                                             Successfully assigned kube-system/intel-fpga-plugin-7r8db to device-plugins-kubernetes-clearlinux-14-4.novalocal
  Normal   Pulling    4m28s             kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Pulling image "intel/intel-fpga-initcontainer:devel"
  Normal   Pulled     2m12s             kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Successfully pulled image "intel/intel-fpga-initcontainer:devel"
  Normal   Created    2m12s             kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Created container intel-fpga-initcontainer
  Normal   Started    2m12s             kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Started container intel-fpga-initcontainer
  Normal   Pulling    2m12s             kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Pulling image "intel/intel-fpga-plugin:devel"
  Normal   Pulled     52s               kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Successfully pulled image "intel/intel-fpga-plugin:devel"
  Warning  Failed     51s               kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Error: container create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"rootfs_linux.go:58: mounting \\\"/var/lib/kubelet/pods/d7262db5-e3fc-4b7b-bc2e-da245f600c4b/containers/intel-fpga-plugin/cddd0f76\\\" to rootfs \\\"/var/lib/containers/storage/overlay/edd75bb94b1b4cf93ae1ea5c064945169fb329d0abdb56b7621cddfc721f6eda/merged\\\" at \\\"/var/lib/containers/storage/overlay/edd75bb94b1b4cf93ae1ea5c064945169fb329d0abdb56b7621cddfc721f6eda/merged/dev/termination-log\\\" caused \\\"open /var/lib/containers/storage/overlay/edd75bb94b1b4cf93ae1ea5c064945169fb329d0abdb56b7621cddfc721f6eda/merged/dev/termination-log: read-only file system\\\"\""
  Warning  Failed     50s               kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Error: container create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"rootfs_linux.go:58: mounting \\\"/var/lib/kubelet/pods/d7262db5-e3fc-4b7b-bc2e-da245f600c4b/containers/intel-fpga-plugin/50f72a5c\\\" to rootfs \\\"/var/lib/containers/storage/overlay/ebd8db253257472f31d801410d21b77da652b90e3fc7a94c796341f252d97c04/merged\\\" at \\\"/var/lib/containers/storage/overlay/ebd8db253257472f31d801410d21b77da652b90e3fc7a94c796341f252d97c04/merged/dev/termination-log\\\" caused \\\"open /var/lib/containers/storage/overlay/ebd8db253257472f31d801410d21b77da652b90e3fc7a94c796341f252d97c04/merged/dev/termination-log: read-only file system\\\"\""
  Warning  Failed     39s               kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Error: container create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"rootfs_linux.go:58: mounting \\\"/var/lib/kubelet/pods/d7262db5-e3fc-4b7b-bc2e-da245f600c4b/containers/intel-fpga-plugin/a94de925\\\" to rootfs \\\"/var/lib/containers/storage/overlay/8ba817f8ac9f553fc88b1be220f8bd7e41e1995765977342e5c8239648fa2898/merged\\\" at \\\"/var/lib/containers/storage/overlay/8ba817f8ac9f553fc88b1be220f8bd7e41e1995765977342e5c8239648fa2898/merged/dev/termination-log\\\" caused \\\"open /var/lib/containers/storage/overlay/8ba817f8ac9f553fc88b1be220f8bd7e41e1995765977342e5c8239648fa2898/merged/dev/terminatio [... cut ...] ination-log: read-only file system\\\"\""
```

By removing lines [43](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/237/files#diff-4bb9f4362d128f1faf357339862c1ec2R43)
and [46](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/237/files#diff-4bb9f4362d128f1faf357339862c1ec2R46) from [fpga_plugin.yaml](https://github.com/intel/intel-device-plugins-for-kubernetes/pull/237/files#diff-4bb9f4362d128f1faf357339862c1ec2), the fpga_plugin gets then deployed correctly.

hectoragb, 2020-01-13:
@bart0sh @mythi

bart0sh, 2020-01-14:
@hectoragb thank you for the report.

I'm not yet able to reproduce it on my system though:

```
> kubectl get pods --namespace kube-system |grep intel-fpga-plugin
intel-fpga-plugin-p89vf          1/1     Running   0          20h

> kubectl describe pod $(kubectl get pods --namespace kube-system |grep intel-fpga-plugin|cut -f1 -d' ') --namespace kube-system
Name:         intel-fpga-plugin-p89vf
Namespace:    kube-system
Priority:     0
Node:         r720-1/10.237.71.129
Start Time:   Mon, 13 Jan 2020 14:32:28 +0200
Labels:       app=intel-fpga-plugin
              controller-revision-hash=78cc488ccd
              pod-template-generation=1
Annotations:  <none>
Status:       Running
IP:           10.88.0.5
IPs:
  IP:           10.88.0.5
  IP:           1100:200::5
Controlled By:  DaemonSet/intel-fpga-plugin
Init Containers:
  intel-fpga-initcontainer:
    Container ID:   cri-o://5cdc923d60dda01714dce6f4c08e73ae9760d25178ed7b2abf900515e575251d
    Image:          intel/intel-fpga-initcontainer:devel
    Image ID:       docker.io/intel/intel-fpga-initcontainer@sha256:3a790b3e3a25db65d5064e8d13d2cdfc4773e2cf79bb6134e30511f075756d76
    Port:           <none>
    Host Port:      <none>
    State:          Terminated
      Reason:       Completed
      Exit Code:    0
      Started:      Mon, 13 Jan 2020 14:32:54 +0200
      Finished:     Mon, 13 Jan 2020 14:32:54 +0200
    Ready:          True
    Restart Count:  0
    Environment:    <none>
    Mounts:
      /etc/containers/oci/hooks.d from oci-hooks-config (rw)
      /opt/intel/fpga-sw from intel-fpga-sw (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from intel-fpga-plugin-controller-token-vnwjx (ro)
Containers:
  intel-fpga-plugin:
    Container ID:   cri-o://1e0dd20b1ad123dc7b54b328e208348afc21009d30823b43198467a2bad95363
    Image:          intel/intel-fpga-plugin:devel
    Image ID:       docker.io/intel/intel-fpga-plugin@sha256:27f1395807d69027043f8b64922572fe28880032195695be3b8ff3a4b4fd184d
    Port:           <none>
    Host Port:      <none>
    State:          Running
      Started:      Mon, 13 Jan 2020 14:33:17 +0200
    Ready:          True
    Restart Count:  0
    Environment:
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /dev from devfs (ro)
      /sys/class from sysfs (ro)
      /var/lib/kubelet/device-plugins from kubeletsockets (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from intel-fpga-plugin-controller-token-vnwjx (ro)
Conditions:
  Type              Status
  Initialized       True 
  Ready             True 
  ContainersReady   True 
  PodScheduled      True 
Volumes:
  devfs:
    Type:          HostPath (bare host directory volume)
    Path:          /dev
    HostPathType:  
  sysfs:
    Type:          HostPath (bare host directory volume)
    Path:          /sys/class
    HostPathType:  
  kubeletsockets:
    Type:          HostPath (bare host directory volume)
    Path:          /var/lib/kubelet/device-plugins
    HostPathType:  
  intel-fpga-sw:
    Type:          HostPath (bare host directory volume)
    Path:          /opt/intel/fpga-sw
    HostPathType:  DirectoryOrCreate
  oci-hooks-config:
    Type:          HostPath (bare host directory volume)
    Path:          /etc/containers/oci/hooks.d
    HostPathType:  DirectoryOrCreate
  intel-fpga-plugin-controller-token-vnwjx:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  intel-fpga-plugin-controller-token-vnwjx
    Optional:    false
QoS Class:       BestEffort
Node-Selectors:  <none>
Tolerations:     node.kubernetes.io/disk-pressure:NoSchedule
                 node.kubernetes.io/memory-pressure:NoSchedule
                 node.kubernetes.io/not-ready:NoExecute
                 node.kubernetes.io/pid-pressure:NoSchedule
                 node.kubernetes.io/unreachable:NoExecute
                 node.kubernetes.io/unschedulable:NoSchedule
Events:          <none>
```

I'll investigate further. If you spot the difference in your and my configurations/environment please let me know.

bart0sh, 2020-01-14:
It seems that /dev/termination-log doesn't exist on your system and cri-o tries to create it and fails. Tweaking terminationMessagePath for the pod should solve it. I'll come up with the fix ASAP.

PR #261 (2020-01-15): fpga_plugin: move termination-log out of /dev
runtime uses /dev/termination-log to write container termination
messages. If this file doesn't exist on the host the runtime tries
to create it. As /dev is read-only for intel-fpga-plugin container
attempt to create /dev/termination-log fails with this error:

```
Warning Failed kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Error:
  container create failed: container_linux.go:345: starting container process caused
  "process_linux.go:430: container init caused \"rootfs_linux.go:58:
  mounting \\\"/var/lib/kubelet/pods/d7262db5-e3fc [... cut]
commit 205d32a6 (2020-01-15): fpga_plugin: move termination-log out of /dev

runtime uses /dev/termination-log to write container termination
messages. If this file doesn't exist on the host the runtime tries
to create it. As /dev is read-only for intel-fpga-plugin container
attempt to create /dev/termination-log fails with this error:

Warning Failed kubelet, device-plugins-kubernetes-clearlinux-14-4.novalocal  Error:
  container create failed: container_linux.go:345: starting container process caused
  "process_linux.go:430: container init caused \"rootfs_linux.go:58:
  mounting \\\"/var/lib/kubelet/pods/d7262db5-e3fc-4b [... cut]
closed 2020-01-15 by rojkov