intel/intel-device-plugins-for-kubernetes #260: GPU plugin fails on CoreOS

poussa, 2020-01-14:
```bash
kubectl get pod -n kube-system intel-gpu-plugin-gktbr -o wide
NAME                     READY   STATUS             RESTARTS   AGE   IP          NODE             NOMINATED NODE   READINESS GATES
intel-gpu-plugin-gktbr   0/1     CrashLoopBackOff   7          13m   10.44.0.1   w-1-k8s-node-1   <none>           <none>
spoussa@cloud-manager:~$ kubectl logs -n kube-system intel-gpu-plugin-gktbr
GPU device plugin started
Device scan failed: open /sys/class/drm: no such file or directory
Can't read sysfs folder
main.(*devicePlugin).scan
	/intel-device-plugins-for-kubernetes/cmd/gpu_plugin/gpu_plugin.go:83
main.(*devicePlugin).Scan
	/intel-device-plugins-for-kubernetes/cmd/gpu_plugin/gpu_plugin.go:69
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*Manager).Run.func1
	/intel-device-plugins-for-kubernetes/pkg/deviceplugin/manager.go:96
runtime.goexit
	/usr/lib/golang/src/runtime/asm_amd64.s:1357
```

```bash
w-1-k8s-master ~ # uname -a
Linux w-1-k8s-master 4.19.86-coreos #1 SMP Mon Dec 2 20:13:38 -00 2019 x86_64 QEMU Virtual CPU version 2.5+ GenuineIntel GNU/Linux
w-1-k8s-master ~ # cat /etc/os-release
NAME="Container Linux by CoreOS"
ID=coreos
VERSION=2303.3.0
VERSION_ID=2303.3.0
BUILD_ID=2019-12-02-2049
PRETTY_NAME="Container Linux by CoreOS 2303.3.0 (Rhyolite)"
ANSI_COLOR="38;5;75"
HOME_URL="https://coreos.com/"
BUG_REPORT_URL="https://issues.coreos.com"
COREOS_BOARD="amd64-usr"
```

mythi, 2020-01-20:
Don't crash/exit but report no devices?
```
files, err := ioutil.ReadDir(dp.sysfsDir)
	if err != nil {
		return nil, errors.Wrap(err, "Can't read sysfs folder")
	}
```

mythi, 2020-01-22:
/cc @grahamwhaley

PR #272 (2020-01-30): gpu: do not fail if no GPU devices found
If we do not find any GPU devices, for whatever reason
(such as missing or unparsable /dev or /sys files), do not
error and quit, just return an empty device tree (indicating
we have no devices on this node).
This is preferable as we will then not enter a retry-death-spin
type scenario.

Fixes: #260
commit 6537e384 (2020-01-30): gpu: do not fail if device scanning fails

If we fail to scan for GPU devices (note, that is potentially
different from not finding any devices during a scan), then
warn on it, and go around the poll loop again. Do not treat
it as a fatal error or we might end up in a re-launch death
deploy loop...

Of course, getting a warning in your logs every 5s could also
be annoying, but is somewhat 'less fatal'.

Fixes: #260
Fixes: #230
closed 2020-01-30 by rojkov