intel/intel-device-plugins-for-kubernetes #274: gpu: update k8s 'schedule GPUs' website page

grahamwhaley, 2020-01-27:
The k8s docs page https://kubernetes.io/docs/tasks/manage-gpus/scheduling-gpus/ details how to schedule GPUs. Presently there is no mention of Intel GPUs (but are mentions of other vendors).

At the appropriate time we should discuss/Issue/PR to get that page updated, which may mean either:
- adding Intel GPUs to that page
- or, if the trend is to remove vendor specific information from such pages, remove all vendor specifics.

The source of the page lives at:
https://github.com/kubernetes/website/blob/master/content/en/docs/tasks/manage-gpus/scheduling-gpus.md

msivosuo, 2021-06-17:
@uniemimu - What do you think?

eero-t, 2021-06-17:
Especially as Intel GPU plugin supports now (hanks to Ukri) fractional resources that the other plugins (according to the page) do not...

msivosuo, 2022-10-05:
Intel reference added to the site (https://kubernetes.io/docs/tasks/manage-gpus/scheduling-gpus/). Closing.

closed 2022-10-05 by msivosuo