intel/intel-device-plugins-for-kubernetes #278: make fails to build vpu plugin

bart0sh, 2020-01-29:
Make fails with this error:
```
$ make
cd cmd/fpga_admissionwebhook; go build -tags ""
cd cmd/fpga_crihook; go build -tags ""
cd cmd/fpga_plugin; go build -tags ""
cd cmd/fpga_tool; go build -tags ""
cd cmd/gpu_plugin; go build -tags ""
cd cmd/qat_plugin; go build -tags ""
cd cmd/vpu_plugin; go build -tags ""
# pkg-config --cflags  -- libusb-1.0
Package libusb-1.0 was not found in the pkg-config search path.
Perhaps you should add the directory containing `libusb-1.0.pc'
to the PKG_CONFIG_PATH environment variable
Package 'libusb-1.0', required by 'world', not found
pkg-config: exit status 1
Makefile:46: recipe for target 'vpu_plugin' failed
make: *** [vpu_plugin] Error 2
```

It can probably be fixed by installing libusb-1.0-0-dev, but it should be handled better I believe.

bart0sh, 2020-01-29:
/cc @alekdu

alekdu, 2020-01-29:
Right, in the merged patch, both Jenkins and .Travis.xml added libusb-1.0-0-dev. That is the dependency of gousb module I think.

mythi, 2020-01-30:
> but it should be handled better I believe.

Makefile to run pre-checks?

bart0sh, 2020-02-03:
build inside a container?

mythi, 2020-02-03:
@bart0sh but container builds we have already. Do you think the default build target should be `images`?

```
$ build/docker/build-image.sh intel-vpu-plugin
$ make images
```

bart0sh, 2020-03-17:
I meant that we can build vpu plugin in a container, but that would probably increase build time.
If you think it's not worth trying please close this issue.

msivosuo, 2021-06-17:
Closing as proposed.

closed 2021-06-17 by msivosuo