intel/intel-device-plugins-for-kubernetes #290: unknown revision v0.0.0

CyberPoison, 2020-02-09:
Hello everyone, i'm new on golang so sorry if this is the wrong github to publish my issue.

Well my issue is than when i try to install intel device plugins for kubernetes.
i get this issue

`go get -d -u github.com/intel/intel-device-plugins-for-kubernetes
go get: github.com/intel/intel-device-plugins-for-kubernetes@v0.15.0 requires
        k8s.io/api@v0.0.0: reading k8s.io/api/go.mod at revision v0.0.0: unknown revision v0.0.0
`

this is my go env

`go env
GO111MODULE="on"
GOARCH="amd64"
GOBIN=""
GOCACHE="/root/.cache/go-build"
GOENV="/root/.config/go/env"
GOEXE=""
GOFLAGS=""
GOHOSTARCH="amd64"
GOHOSTOS="linux"
GONOPROXY=""
GONOSUMDB=""
GOOS="linux"
GOPATH="/root/go"
GOPRIVATE=""
GOPROXY="https://proxy.golang.org,direct"
GOROOT="/usr/local/go"
GOSUMDB="sum.golang.org"
GOTMPDIR=""
GOTOOLDIR="/usr/local/go/pkg/tool/linux_amd64"
GCCGO="gccgo"
AR="ar"
CC="gcc"
CXX="g++"
CGO_ENABLED="1"
GOMOD="/dev/null"
CGO_CFLAGS="-g -O2"
CGO_CPPFLAGS=""
CGO_CXXFLAGS="-g -O2"
CGO_FFLAGS="-g -O2"
CGO_LDFLAGS="-g -O2"
PKG_CONFIG="pkg-config"
GOGCCFLAGS="-fPIC -m64 -pthread -fmessage-length=0 -fdebug-prefix-map=/tmp/go-build637408058=/tmp/go-build -gno-record-gcc-switches"
`

go version latest (v1.13.7)

Is there a way to workarround my issue ?

Kind Regards,

CyberPoison, 2020-02-15:
Any help?

mythi, 2020-02-15:
@CyberPoison would `git clone` the repository unblock you? What's your goal?

I was able to reproduce the issue even with:
`go get -d -u github.com/intel/intel-device-plugins-for-kubernetes@master`

CyberPoison, 2020-02-16:
My goal is to use hw transcoding on plex pods but on rke but since the plug-in doesn't install because I get the error mentioned above, i tried many thing BTW, how can I install Intel plug-in kubernets by git clone without using go get?

mythi, 2020-02-17:
@CyberPoison The plugin images can be found in Dockerhub (intel/intel-XXX-plugin). Would those work?

You can `git clone` to `cd $GOPATH/src/github.com/intel/intel-device-plugins-for-kubernetes` and then follow [the build instructions](https://github.com/intel/intel-device-plugins-for-kubernetes/tree/master/cmd/gpu_plugin#build-the-plugin-image)

FWIW, you've reported a real problem. We'll look into it.

CyberPoison, 2020-02-17:
By using ```git clone ``` to cd $GOPATH/src/github.com/intel/intel-device-plugins-for-kubernetes
and after followed the build instruction have worked fine for me.

Many thanks ;)

PR #299 (2020-02-24): READMEs: use git clone to get the code
go get'ing does not work due to our k8s.io/kubernetes dependency
so guide users to use git clone to get the code.

Fixes: #290
commit f145541c (2020-02-24): READMEs: use git clone to get the code

go get'ing does not work due to our k8s.io/kubernetes dependency
so guide users to use git clone to get the code.

Fixes: #290
closed 2020-02-24 by bart0sh