intel/intel-device-plugins-for-kubernetes #303: webhook throws "tls: bad certificate" error

bart0sh, 2020-02-25:
Admission controller webhook throws this error when trying to modify pod referencing FPGA resource.
The pod spec: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/demo/test-fpga-region.yaml
webhook log:
```
> kubectl logs $(kubectl get pods | grep intel-fpga-webhook | awk '{print $1}')
DEBUG fpga_admissionwebhook.go:280 Webhook started
Starting controller
Started controller workers
DEBUG patcher.go:341 created new patcher for namespace default
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.2-nlb0  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.2-nlb0 886d9ebd-1443-4653-a08d-e5e99fdef6ea 7864836 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {d8424dc4a4a3c413f89e433683f9040b}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.2-nlb0 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.2-nlb3  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.2-nlb3 93c2c539-380b-45b3-ae59-e5e340d9a2dd 7864837 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {f7df405cbd7acf7222f144b0b93acd18}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.2-nlb3 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.0-nlb3  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.0-nlb3 d8b5a361-5574-4bde-96e4-2cff95202f7e 7864833 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {f7df405cbd7acf7222f144b0b93acd18}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.0-nlb3 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.0-nlb0  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.0-nlb0 44a2b794-c807-4a5f-ab9b-3aea1063d5a5 7864832 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {d8424dc4a4a3c413f89e433683f9040b}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.0-nlb0 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.1-nlb0  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.1-nlb0 3157a596-ccd2-4712-8234-eb65e7422cc7 7864834 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {d8424dc4a4a3c413f89e433683f9040b}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.1-nlb0 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.1-nlb3  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.1-nlb3 a24e5310-ef17-469b-a579-fd9c107766a9 7864835 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {f7df405cbd7acf7222f144b0b93acd18}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.1-nlb3 af}'
DEBUG controller.go:197 Received &{{ } {d5005-nlb0  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/d5005-nlb0 46283fa3-aa47-4682-9e32-2cd3a747c3e4 7864838 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {d8424dc4a4a3c413f89e433683f9040b}}
DEBUG controller.go:156 Successfully synced '{default/d5005-nlb0 af}'
DEBUG controller.go:197 Received &{{ } {d5005-nlb3  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/d5005-nlb3 7497523c-4c87-4638-85d5-6c82080aa51b 7864839 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {f7df405cbd7acf7222f144b0b93acd18}}
DEBUG controller.go:156 Successfully synced '{default/d5005-nlb3 af}'
DEBUG controller.go:197 Received &{{ } {arria10.dcp1.0-compress  default /apis/fpga.intel.com/v1/namespaces/default/acceleratorfunctions/arria10.dcp1.0-compress 942215dc-bc09-4bc5-b7cb-90b3285cd560 7864831 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {946c21d1e49704a5e5daa0805bc6b0785e1765bf}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.0-compress af}'
DEBUG controller.go:231 Received &{{ } {arria10.dcp1.0  default /apis/fpga.intel.com/v1/namespaces/default/fpgaregions/arria10.dcp1.0 06bbe910-7350-4392-85df-e4a3446d6cef 7864840 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {ce48969398f05f33946d560708be108a}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.0 region}'
DEBUG controller.go:231 Received &{{ } {arria10.dcp1.1  default /apis/fpga.intel.com/v1/namespaces/default/fpgaregions/arria10.dcp1.1 eeca3df8-7b85-424c-bf41-776643d907b7 7864841 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {9926ab6d6c925a68aabca7d84c545738}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.1 region}'
DEBUG controller.go:231 Received &{{ } {arria10.dcp1.2  default /apis/fpga.intel.com/v1/namespaces/default/fpgaregions/arria10.dcp1.2 cb97966d-d2f9-4b44-9478-c4878a03da62 7864842 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {69528db6eb31577a8c3668f9faa081f6}}
DEBUG controller.go:156 Successfully synced '{default/arria10.dcp1.2 region}'
DEBUG controller.go:231 Received &{{ } {d5005  default /apis/fpga.intel.com/v1/namespaces/default/fpgaregions/d5005 90764c1f-71d6-45f8-aa07-1b1b2f153788 7864843 1 2020-02-25 15:14:21 +0000 UTC <nil> <nil> map[] map[] [] []  []} {bfac4d851ee856fe8c95865ce1bbaa2d}}
DEBUG controller.go:156 Successfully synced '{default/d5005 region}'
2020/02/25 15:15:56 http: TLS handshake error from 10.237.71.129:41488: remote error: tls: bad certificate
```

As a result resource is not mapped:
```
> kubectl describe pod test-fpga-region
Name:         test-fpga-region
Namespace:    default
Priority:     0
Node:         <none>
Labels:       <none>
Annotations:  <none>
Status:       Pending
IP:           
IPs:          <none>
Containers:
  test-container:
    Image:      intel/opae-nlb-demo:devel
    Port:       <none>
    Host Port:  <none>
    Limits:
      cpu:                                 1
      fpga.intel.com/arria10.dcp1.2-nlb3:  1
      hugepages-2Mi:                       20Mi
    Requests:
      cpu:                                 1
      fpga.intel.com/arria10.dcp1.2-nlb3:  1
      hugepages-2Mi:                       20Mi
    Environment:                           <none>
    Mounts:
      /var/run/secrets/kubernetes.io/serviceaccount from default-token-gz5m5 (ro)
Conditions:
  Type           Status
  PodScheduled   False 
Volumes:
  default-token-gz5m5:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  default-token-gz5m5
    Optional:    false
QoS Class:       Burstable
Node-Selectors:  <none>
Tolerations:     node.kubernetes.io/not-ready:NoExecute for 300s
                 node.kubernetes.io/unreachable:NoExecute for 300s
Events:
  Type     Reason            Age        From               Message
  ----     ------            ----       ----               -------
  Warning  FailedScheduling  <unknown>  default-scheduler  0/1 nodes are available: 1 Insufficient fpga.intel.com/arria10.dcp1.2-nlb3.
  Warning  FailedScheduling  <unknown>  default-scheduler  0/1 nodes are available: 1 Insufficient fpga.intel.com/arria10.dcp1.2-nlb3.
```

Steps to reproduce:
1. ./scripts/fpga-plugin-prepare-for-kustomization.sh
2. kubectl create -k deployments/fpga_plugin/overlays/region
3. kubectl create -f demo/test-fpga-region.yaml
4. kubectl logs $(kubectl get pods | grep intel-fpga-webhook | awk '{print $1}')
5. kubectl describe pod test-fpga-region

rojkov, 2020-02-25:
I remember I saw this error when CA_BUNDLE was in wrong format: it should be a PEM encoded once again in base64 in ONE line (i.e. piped through `base64 -w 0`).

askervin, 2020-02-26:
When I'm comparing `caBundle:` value injected to the yaml by kustomize, and `base64 -w 0` output injected to the yaml by `webhook-deploy.sh`, there doesn't seem to be any difference.
```bash
$ kubectl get configmap -n kube-system extension-apiserver-authentication -o=jsonpath='{.data.client-ca-file}' | ( base64 -w 0 ; echo ) | md5sum
04100b750f7fd15ddfe35c31ea7060d5  -

$ kubectl kustomize deployments/fpga_plugin/overlays/region | awk '/caBundle:/{print $2}' | md5sum
04100b750f7fd15ddfe35c31ea7060d5  -
```

Do you think there could be some other explanation, too?

askervin, 2020-02-26:
@bart0sh , @rojkov , root caused, fix in PR #305

bart0sh, 2020-02-26:
Should we consider using getopt[s] to avoid this kind of things in future?

PR #305 (2020-02-26): FPGA kustomization: fix signing certificate
The certificate got signed for wrong service name (fallback default
"webhook-svc") due to incorrectly passed parameter. This showed as
    "http: TLS handshake error from ... tls: bad certificate"
error in webhook pod logs.

Fixes: #303
commit 7e2044b8 (2020-02-26): FPGA kustomization: fix signing certificate

The certificate got signed for wrong service name (fallback default
"webhook-svc") due to incorrectly passed parameter. This showed as
    "http: TLS handshake error from ... tls: bad certificate"
error in webhook pod logs.

Fixes: #303
closed 2020-02-26 by mythi