intel/intel-device-plugins-for-kubernetes #323: Klog logging

msivosuo, 2020-03-10:


msivosuo, 2020-03-11:
@grahamwhaley - could you take a look?

grahamwhaley, 2020-03-11:
Sure. We probably want to consult with @pohly to ensure we align or prepare for future k8s infra logging changes - I'm thinking in particular wrt to https://github.com/kubernetes/enhancements/pull/1367 for instance.

pohly, 2020-03-11:
It looks like the direction in Kubernetes now is to enhance klog incrementally, so starting to use it is a good first step.

grahamwhaley, 2020-03-11:
Going to make some notes as I do some prelim investigation...

First note then, is that afaict all components (framework and existing plugins) today use fmt.Print directly. That means all logs end up in their relevant container logs, just like a normal container.

Second note - from [this page](https://kubernetes.io/docs/concepts/cluster-administration/logging/), we see it says:

> On machines with systemd, the kubelet and container runtime write to journald. If systemd is not present, they write to .log files in the /var/log directory. System components inside containers always write to the /var/log directory, bypassing the default logging mechanism. They use the klog logging library. You can find the conventions for logging severity for those components in the development docs on logging.

That reads to me that our components, being system components inside containers, should use `klog`, and should write directly to `/var/log`.

I had a look on my `minikube` cluster, and did not see this - but, if we go look at the [kubernetes kube-proxy daemonset](https://github.com/kubernetes/kubernetes/blob/master/cluster/addons/kube-proxy/kube-proxy-ds.yaml#L41-L55) as an example system component container, we see:

```yaml
        command:
        - /bin/sh
        - -c
        - kube-proxy {{cluster_cidr}} --oom-score-adj=-998 {{params}} 1>>/var/log/kube-proxy.log 2>&1
        env:
        - name: KUBERNETES_SERVICE_HOST
          value: {{kubernetes_service_host_env_value}}
        {{kube_cache_mutation_detector_env_name}}
          {{kube_cache_mutation_detector_env_value}}
        securityContext:
          privileged: true
        volumeMounts:
        - mountPath: /var/log
          name: varlog
          readOnly: false
```

That makes me think we should probably pursue mounting `/var/log` into the plugins and redirecting their `klog` output by default, but do this via our kustomize methods, so it is easily over-rideable...

mythi, 2020-03-12:
> That reads to me that our components, being system components inside containers, 

I don't consider our plugins as system components and would prefer we don't add any extra `hostPath` mounts. `kubectl logs` works.

closed 2020-04-01 by mythi