intel/intel-device-plugins-for-kubernetes #324: Intel Devices Operator

msivosuo, 2020-03-10:
Implement one lifecycle operator for all intel device plugins
- [ ] Study
- [ ] Prototype
- [ ] 0.1 version
- [ ] Publish on operator hub

msivosuo, 2020-04-08:
@rojkov, @askervin  - would either of you have time to pick this up? @mythi can provide background information.

askervin, 2020-04-09:
@msivosuo , I'll talk with @mythi about this.

msivosuo, 2020-04-09:
@askervin  - I talked with @rojkov already and we agreed he's taking a look. But thanks anyway!

rojkov, 2020-04-22:
braindump on possible functionality: https://docs.google.com/document/d/1raZQKpCIgqmdiI41hCNJL1kmZAYz8xJMzy7OsLJNyVQ/edit#heading=h.fnlack9sz6iv

msivosuo, 2020-06-03:
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/376

closed 2020-07-01 by mythi