intel/intel-device-plugins-for-kubernetes #349: VPU Plugin Go unit tests coverage

mythi, 2020-03-24:
@alekdu we're improving Go unit test coverage for our plugins.

VPU plugin coverage is <50%. Please check if that could be improved still.

Steps:
1.  `make test WHAT=cmd/vpu_plugin`
2. review `cmd/vpu_plugin/coverage.html`

There's not much code in the plugin so while many things are already covered the coverage is poor.

Some ideas:
1. move `sharedDevNum` checking to `newDevicePlugin` and add a test case for 0 value
2. test without `/var/tmp/hddl_service.sock`
3. test `Scan()` API as it was just added in FPGA plugin
4. `os.Exit(1)` is now redundant 

Other:
`/var/tmp/hddl_service.sock` is not cleaned up after testing

alekdu, 2020-03-26:
hmm... I will look into this.

alekdu, 2020-03-30:
https://github.com/intel/intel-device-plugins-for-kubernetes/pull/352 please review

mythi, 2020-04-01:
these changes were made...

closed 2020-04-01 by mythi