intel/intel-device-plugins-for-kubernetes #351: qat0plugin is not working with igb_uio

sindhura730, 2020-03-27:
I'm trying to make QAT VFs (Atom Processor C3000) available to my dpdk based app running in a kubernetes cluster. 
I deployed the plugin and it was able to bind the VFs to igb_uio driver and I see intel.qat.com/generic devices becoming available in the node object.
I brought up my app in a pod requesting 4 qat resources. But it throws an error, 
`
EAL: PCI device 0000:01:02.3 on NUMA socket -1
EAL:   Invalid NUMA socket, default to 0
EAL:   probe driver: 8086:19e3 qat
EAL: Cannot open /sys/class/uio/uio13/device/config: No such file or directory
EAL: Requested device 0000:01:02.3 cannot be used
`
Basically the /sys/class/uio/<uiox>/device on the node is being mounted as /sys/class/uio in the container, so it can not find the file. I verified that permissions are not a problem.
And that seems to be consistent with code. In getDpdkMounts(), I see this,

`       uioMountPoint := path.Join(uioMountPath, dpdkDeviceName, "/device")
        return []pluginapi.Mount{
            {
                HostPath:      uioMountPoint,
                ContainerPath: uioMountPath,
            },
`
What is the reason for this mount path? 
I tried with the qat-dpdk app provided for the demo, I get the same error.

mythi, 2020-03-27:
@sindhura730 thanks for the report. I don't have C3000 online right now but I'll try to compile `igb_uio` to see the result here.

@swatisehgal any comments to:

> What is the reason for this mount path?

sindhura730, 2020-03-30:
@mythi I don't think it is specific to C3000. I'll try without the plugin and see if it works.

mythi, 2020-03-30:
@sindhura730 I believe so too. What is the dpdk command that fails for you?

sindhura730, 2020-03-30:
I'm trying the demo pod, qat-dpdk mentioned in the documentation.
The following command fails.

dpdk-test-crypto-perf -l 6-7 -w $QAT1 \
-d /usr/lib64/librte_mempool_ring.so.1.1 \
-d /usr/lib64/librte_pmd_qat.so.1.1 -- \
--ptest throughput --devtype crypto_qat \
--optype cipher-only --cipher-algo aes-cbc --cipher-op encrypt \
--cipher-key-sz 16 --total-ops 10000000 --burst-sz 32 --buffer-sz 64

sindhura730, 2020-03-30:
I made some changes, and now it is working. Basically mounts are wrong.

`diff --git a/cmd/qat_plugin/dpdkdrv/dpdkdrv.go b/cmd/qat_plugin/dpdkdrv/dpdkdrv.go
index 2ff1420..ec640e7 100644
--- a/cmd/qat_plugin/dpdkdrv/dpdkdrv.go
+++ b/cmd/qat_plugin/dpdkdrv/dpdkdrv.go
@@ -136,11 +136,10 @@ func (dp *DevicePlugin) getDpdkDeviceSpecs(id string) ([]pluginapi.DeviceSpec, e
        // TODO: case "pci-generic" and "kernel":
        case "igb_uio":
                //Setting up with uio
-               uioDev := path.Join(uioDevicePath, dpdkDeviceName)
                return []pluginapi.DeviceSpec{
                        {
-                               HostPath:      uioDev,
-                               ContainerPath: uioDev,
+                               HostPath:      uioDevicePath,
+                               ContainerPath: uioDevicePath,
                                Permissions:   "rw",
                        },
                }, nil
@@ -178,7 +177,7 @@ func (dp *DevicePlugin) getDpdkMounts(id string) ([]pluginapi.Mount, error) {
                return []pluginapi.Mount{
                        {
                                HostPath:      uioMountPoint,
-                               ContainerPath: uioMountPath,
+                               ContainerPath: uioMountPoint,
                        },
                }, nil
        case "vfio-pci":
diff --git a/go.mod b/go.mod
index e8d7be1..09c979b 100644
--- a/go.mod
+++ b/go.mod
`

mythi, 2020-03-30:
@sindhura730 thanks

```
        case "igb_uio":
                //Setting up with uio
-               uioDev := path.Join(uioDevicePath, dpdkDeviceName)
                return []pluginapi.DeviceSpec{
                        {
-                               HostPath:      uioDev,
-                               ContainerPath: uioDev,
+                               HostPath:      uioDevicePath,
+                               ContainerPath: uioDevicePath,
                                Permissions:   "rw",
```

We don't want to provide full `/dev/` but the requested VF devices only.

mythi, 2020-03-30:
OK, I'm able to reproduce this and see what's wrong. However,

```
                return []pluginapi.Mount{
                        {
                                HostPath:      uioMountPoint,
-                               ContainerPath: uioMountPath,
+                               ContainerPath: uioMountPoint,
                        },
```

is not good because it makes the uio visible in the container... I'll fix this.

mythi, 2020-03-30:
> is not good because it makes the uio visible in the container... I'll fix this.

I just learnt that cri-o/containerd automatically mounts sysfs read-only in containers so it looks `getDpdkMounts()` is not needed.

mythi, 2020-03-31:
@sindhura730 does #353 work for you?

sindhura730, 2020-04-01:
I'll test and get back to you.

sindhura730, 2020-04-01:
Looks like the commit is not merged yet. I did try removing the mount from the code and it didn't work.

Still says, `Cannot open /sys/class/uio/uio7/device/config: Read-only file system`

mythi, 2020-04-01:
> Still says, Cannot open /sys/class/uio/uio7/device/config: Read-only file system

Looks like I had an error in my test setup. I'll update the PR shortly. Thanks for checking!

mythi, 2020-04-01:
@sindhura730 the PR is now updated.

closed 2020-04-01 by rojkov

mythi, 2020-04-02:
@sindhura730 can you confirm it works now?

sindhura730, 2020-04-02:
@mythi  It seems to be working now. Thank you.

mythi, 2020-04-02:
@sindhura730 thanks!

PR #353 (2020-04-01): qat: fix UIO mounts
Fixes: #351