intel/intel-device-plugins-for-kubernetes #356: Review README files

rojkov, 2020-04-02:
Some files got outdated. For example `cmd/gpu_plugin/README.md` describes the installation procedure like we still use `dep` instead of `go mod`.

mythi, 2020-04-02:
We just did a bunch of updates. What is outdated?

> like we still use dep instead of go mod.

Where's this?

rojkov, 2020-04-02:
Particularly this
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/cmd/gpu_plugin/README.md#getting-the-source-code

It's not needed nowadays to clone the repo into a specific path.

rojkov, 2020-04-02:
This section https://github.com/intel/intel-device-plugins-for-kubernetes#running-e2e-tests needs to be included in the README's ToC.

mythi, 2020-04-02:
> It's not needed nowadays to clone the repo into a specific path.

true. I touched this part last time (f145541) and IIRC my thinking was that this gives a consistent starting point/path for each higher level section without any bigger re-work...

mythi, 2020-06-16:
Next steps:

- [ ] in all of the READMEs, highlight the container image availability and the possibility to deploy using `kubectl apply` without cloning the repo
- [ ] change git cloning (`go get` does not work) to be path agnostic (`GOPATH` no longer relevant)

pohly, 2020-06-16:
> the possibility to deploy using kubectl apply without cloning the repo

Also make sure that those deployment files always pull an image that is compatible with the deployment.

In PMEM-CSI, we ensure that for releases, but not for the development branch, and already had issue reports twice because users were using deployment files from an old revision of our master branch with the latest, incompatible "canary" images.

tkhred, 2020-06-17:
Hi, based on the following PR, the plugin has already supported k8s 1.18. 

https://github.com/intel/intel-device-plugins-for-kubernetes/pull/365

I think it is a critical problem.  (Actually, we need several days to know that the plugin supports 1.18 or not.)

Could you consider to update the top page README first?

mythi, 2020-06-17:
> Could you consider to update the top page README first?

I just opened a PR to work on updates in the top page README. What is the concern? In principle, the release branches should be used for work that requires stability.

tkhred, 2020-06-17:
Thanks for your quick response. 
Sorry for my ambiguous request. I just would like README to say it supports 1.18. (now it supports 1.11, 1.15, and 1.17)

tkhred, 2020-06-17:
Sorry that I asked without checking your project plan. Our team will test the master branch with k8s 1.18 and I believe we can wait for the official v0.18 release.

mythi, 2020-06-17:
@tkhred no problem at all. we're planning to make a v0.18 shortly (#251) you'r reminder is a valid task for #251!

PR #460 (2020-09-16): sync README.md files with the current state of the code
Closes #356
Closes #447
commit 1b82ab9d (2020-09-16): sync README.md files with the current state of the code

Closes #356
closed 2020-09-16 by mythi