intel/intel-device-plugins-for-kubernetes #368: Does intel-gpu-plugin work with Atom x5-Z8350 cpus?

todaywasawesome, 2020-04-19:
I get a segmentation fault when I try the demo but maybe I didn't install correctly. 

Here are the logs
`+ WORK_DIR=/root/6-1/fft
+ cd /root/6-1/fft
+ ./fft
Segmentation fault (core dumped)`

And the pod description
`kubectl describe pod intelgpu-demo-job-qxhxj              695ms  Sat Apr 18 19:16:06 2020
Name:         intelgpu-demo-job-qxhxj
Namespace:    default
Priority:     0
Node:         atomic1/10.0.0.77
Start Time:   Sat, 18 Apr 2020 19:15:43 -0600
Labels:       controller-uid=3e7e02ea-ee41-4dfe-a9de-385ccff86e33
              job-name=intelgpu-demo-job
              jobgroup=intelgpu-demo
Annotations:  <none>
Status:       Failed
IP:           10.42.3.65
IPs:
  IP:           10.42.3.65
Controlled By:  Job/intelgpu-demo-job
Containers:
  intelgpu-demo-job-1:
    Container ID:  containerd://2cb48404e6f016a6dcd975e2cfec6a39cec0d70a60a2c026ed8711a00b08f1cf
    Image:         rpbo/ubuntu-demo-opencl:devrel
    Image ID:      docker.io/rpbo/ubuntu-demo-opencl@sha256:3c89d01fc583562c662bf45dccfd95a51b365cafd65a96a92ca986086da916a6
    Port:          <none>
    Host Port:     <none>
    Command:
      /run-opencl-example.sh
      /root/6-1/fft
    State:          Terminated
      Reason:       Error
      Exit Code:    139
      Started:      Sat, 18 Apr 2020 19:15:45 -0600
      Finished:     Sat, 18 Apr 2020 19:15:46 -0600
    Ready:          False
    Restart Count:  0
    Limits:
      gpu.intel.com/i915:  1
    Requests:
      gpu.intel.com/i915:  1
    Environment:           <none>
    Mounts:
      /var/run/secrets/kubernetes.io/serviceaccount from default-token-7rz6h (ro)
Conditions:
  Type              Status
  Initialized       True 
  Ready             False 
  ContainersReady   False 
  PodScheduled      True 
Volumes:
  default-token-7rz6h:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  default-token-7rz6h
    Optional:    false
QoS Class:       BestEffort
Node-Selectors:  <none>
Tolerations:     node.kubernetes.io/not-ready:NoExecute for 300s
                 node.kubernetes.io/unreachable:NoExecute for 300s
Events:
  Type    Reason     Age        From               Message
  ----    ------     ----       ----               -------
  Normal  Scheduled  <unknown>  default-scheduler  Successfully assigned default/intelgpu-demo-job-qxhxj to atomic1
  Normal  Pulled     22s        kubelet, atomic1   Container image "rpbo/ubuntu-demo-opencl:devrel" already present on machine
  Normal  Created    22s        kubelet, atomic1   Created container intelgpu-demo-job-1
  Normal  Started    22s        kubelet, atomic1   Started container intelgpu-demo-job-1`

todaywasawesome, 2020-04-19:
Here's the description on one of the DaemonSet installer pods.

```
kubectl describe pod intel-gpu-plugin-28zcw               865ms  Sat Apr 18 19:19:13 2020
Name:         intel-gpu-plugin-28zcw
Namespace:    default
Priority:     0
Node:         atomic2/10.0.0.78
Start Time:   Sat, 18 Apr 2020 12:25:53 -0600
Labels:       app=intel-gpu-plugin
              controller-revision-hash=6567df68cd
              pod-template-generation=1
Annotations:  <none>
Status:       Running
IP:           10.42.1.58
IPs:
  IP:           10.42.1.58
Controlled By:  DaemonSet/intel-gpu-plugin
Containers:
  intel-gpu-plugin:
    Container ID:   containerd://67edc7d620a720baa12b086b2af104dd42e21ea83f4c1004edf7da797609d10b
    Image:          intel/intel-gpu-plugin:devel
    Image ID:       docker.io/intel/intel-gpu-plugin@sha256:88d6e75654c1427c9b86854692c05ff469e3e36d162b4faa45c15014ce36c483
    Port:           <none>
    Host Port:      <none>
    State:          Running
      Started:      Sat, 18 Apr 2020 12:26:05 -0600
    Ready:          True
    Restart Count:  0
    Environment:
      NODE_NAME:   (v1:spec.nodeName)
    Mounts:
      /dev/dri from devfs (ro)
      /sys/class/drm from sysfs (ro)
      /var/lib/kubelet/device-plugins from kubeletsockets (rw)
      /var/run/secrets/kubernetes.io/serviceaccount from default-token-7rz6h (ro)
Conditions:
  Type              Status
  Initialized       True 
  Ready             True 
  ContainersReady   True 
  PodScheduled      True 
Volumes:
  devfs:
    Type:          HostPath (bare host directory volume)
    Path:          /dev/dri
    HostPathType:  
  sysfs:
    Type:          HostPath (bare host directory volume)
    Path:          /sys/class/drm
    HostPathType:  
  kubeletsockets:
    Type:          HostPath (bare host directory volume)
    Path:          /var/lib/kubelet/device-plugins
    HostPathType:  
  default-token-7rz6h:
    Type:        Secret (a volume populated by a Secret)
    SecretName:  default-token-7rz6h
    Optional:    false
QoS Class:       BestEffort
Node-Selectors:  <none>
Tolerations:     node.kubernetes.io/disk-pressure:NoSchedule
                 node.kubernetes.io/memory-pressure:NoSchedule
                 node.kubernetes.io/not-ready:NoExecute
                 node.kubernetes.io/pid-pressure:NoSchedule
                 node.kubernetes.io/unreachable:NoExecute
                 node.kubernetes.io/unschedulable:NoSchedule
Events:          <none>
```

rojkov, 2020-04-20:
Thank you for the report! I guess we need to implement better HW detection if it's possible.

Even though Atom x5-Z8350 uses the same i915 driver it doesn't support OpenCL. Currently the OpenCL runtime supports CPUs with Intel HD, Iris and Iris Pro graphics.

todaywasawesome, 2020-04-20:
Got it, thanks!

todaywasawesome, 2020-04-21:
@rojkov Thanks again for the answer. In this case, I actually don't need OpenCL. I'm just looking for access to Intel Quick Sync. I've tried mounting the `/dev/dri` directory in Kubernetes, running the container in privileged mode, making sure the node OS has the intel drivers installed, tested using ffmpeg and found quick sync is working on the host. 

I'm reading through the source here to try to understand exactly what the plugin is doing. Any ideas on how I might expose quick sync in a way my Kubernetes container understands? 

I've seen people run this same container from Docker command line and get Quick Sync working only by mounting the `/dev/dri` directory. Doesn't seem as simple in Kubernetes.

mythi, 2020-04-21:
> I'm just looking for access to Intel Quick Sync. I've tried mounting the /dev/dri

bind mounting the device directory is not enough. The plugin returns the list of devices that are added to the container using device cgroup controller. On my system the list is:

```
I0421 04:23:14.673901       1 gpu_plugin.go:129] Adding /dev/dri/card0 to GPU card0
I0421 04:23:14.673943       1 gpu_plugin.go:129] Adding /dev/dri/renderD128 to GPU card0
```

What are the device nodes your app needs?

todaywasawesome, 2020-04-21:
@mythi Oh, I'm not sure how you got that output. How would I check this?

I need to expose `card0` and `renderD128`

Great news to report! I actually got it working!! To test, I started a ubuntu container, installed ffmpeg, and then ran a render and watched the hw acceleration happen on the host node using `intel_gpu_top`. 

So the answer to the question is, YES, this plugin actually does expose everything needed for Intel Quick Sync on an Atom x5-Z8350 cpu!

mythi, 2020-04-21:
@todaywasawesome great that it works!

For more verbose logging, I added `-v=4` args to my deployment.

eero-t, 2020-04-21:
I think this can be closed.  GPU plugin just provides the device node for K8s containers.  Rest is up to your container contents.  If you want to actually use the GPU mapped to the container, it needs to have a suitable HW driver.

Note that for Media, there are two drivers.  Older HW is supported by the default i965 VA-API driver, newer HW by the iHD intel-media driver [1], with some overlap on which HW they support, see:
* https://en.wikipedia.org/wiki/Intel_Quick_Sync_Video#Hardware_decoding_and_encoding
* https://github.com/intel/media-driver/

To use the new iHD driver in Ubuntu container, use 19.10 or newer, install `intel-media-va-driver` package and set the following environment variable for libva backend loading: `export LIBVA_DRIVER_NAME=iHD`

closed 2020-04-21 by mythi