intel/intel-device-plugins-for-kubernetes #371: ServiceAccount creation failed.

xxinran, 2020-04-27:
When I tried to create a service account by:
**kubectl create -f deployments/fpga_plugin/fpga_plugin_service_account.yaml**, it occurs:
"error: error validating "fpga_plugin_service_account.yaml": error validating data: ValidationError(ServiceAccount.metadata.namespace): invalid type for io.k8s.apimachinery.pkg.apis.meta.v1.ObjectMeta.namespace: got "map", expected "string"; if you choose to ignore these errors, turn validation off with --validate=false
"

It seems that "{namespace}" is not loaded correctly, as I know, this is hard code here the "kube-system" as the namespace. 

Did I missed some operations before creating service account?

bart0sh, 2020-04-27:
@xxinran thank you for reporting it. Yes, you need to replace {namespace} with the namespace name like it's done here: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/scripts/deploy-fpgaplugin.sh#L71 you can also just run the script scripts/deploy-fpgaplugin.sh to deploy the plugins

Alternatively you can use latest release tag v0.17.0. master branch contains unreleased code, so this kind of issues can easily happen.

xxinran, 2020-04-27:
> @xxinran thank you for reporting this. Yes, you need to replace {namespace} with the namespace name like it's done here: https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/scripts/deploy-fpgaplugin.sh#L71
> You can also just run the script scripts/deploy-fpgaplugin.sh to deploy the plugin
> 
> Alternatively you can use latest release tag v0.17.0.
> master branch contains unreleased code, so this kind of issues can easily happen.

Thank you for your quick reply. 
I saw that in [https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/scripts/deploy-fpgaplugin.sh] the namespace is set to "default" by default. But it was set to kube-system before this change. 
What is the reason of this change?

bart0sh, 2020-04-27:
It's more secure to deploy plugin in default namespace. However, if you prefer you can use tube-system or any other namespace of your choice.

xxinran, 2020-04-27:
> It's more secure to deploy plugin in default namespace. However, if you prefer you can use kube-system or any other namespace of your choice.

Thanks for your explaination!

closed 2020-04-27 by xxinran