intel/intel-device-plugins-for-kubernetes #373: Admission webhook can not map to correct AFU ID

xxinran, 2020-04-29:
Hi guys, I am trying Intel FPGA Device PlugIn with A10 FPGA. 
The FPGA resource has report correctly, I checked it by `kubectl describe nodes 127.0.0.1 | grep -A4 Alloc`, the output is :

```
Allocatable:
  cpu:                                                 32
  ephemeral-storage:                                   3188670572180
  fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18:  1
  hugepages-1Gi:                                       4Gi

```
I have succeed run a pod with this yaml file:
```
apiVersion: v1
kind: Pod
metadata:
  name: test-nlb3
spec:
  containers:
  - name: test-nlb3
    image: intel/opae-nlb-demo:devel
    imagePullPolicy: IfNotPresent
    command: ['nlb3']
    securityContext:
      capabilities:
        add:
          [IPC_LOCK]
    resources:
      limits:
        fpga.intel.com/af-f7df405cbd7acf7222f144b0b93acd18: 1
        cpu: 1
        hugepages-2Mi: 20Mi
  restartPolicy: Never
```
But if I changed the resources section to: 
```
    resources:
      limits:
        fpga.intel.com/arria10.dcp1.0-nlb3: 1
        cpu: 1
        hugepages-2Mi: 20Mi

```
The pod will be pending with an schedule error. 
The related logs are the following:

Run `kubectl logs intel-fpga-webhook-deployment-7f6b8474c-nd64g`
```
2020/04/29 12:15:55 http: TLS handshake error from 172.17.0.1:23786: remote error: tls: bad certificate
```

Run ` tail -f kube-apiserver.log`:
```
failed calling webhook "fpga.mutator.webhooks.intel.com": Post "https://intel-fpga-webhook-svc.default.svc:443/pods?timeout=30s": x509: certificate signed by unknown authority (possibly because of "crypto/rsa: verification error" while trying to verify candidate authority certificate "ca")
```
Run `kubectl describe pod test-nlb3`:
```

Events:
  Type     Reason            Age   From               Message
  ----     ------            ----  ----               -------
  Warning  FailedScheduling  6s    default-scheduler  0/1 nodes are available: 1 Insufficient fpga.intel.com/arria10.dcp1.0-nlb3.
  Warning  FailedScheduling  6s    default-scheduler  0/1 nodes are available: 1 Insufficient fpga.intel.com/arria10.dcp1.0-nlb3.
````

Also, I wonder the notion of DCP, I noticed there are 3 version of DCPs, I have tried each of them, and had always the same error.  What does DCP exactly mean?

rojkov, 2020-04-29:
I see you are testing with a local setup. In this case the CA certificate is wrong usually as mentioned in fpga_admissionwebhook's README. You can try deploy the webhook with
```
$ ./scripts/webhook-deploy.sh --ca-bundle-path /var/run/kubernetes/server-ca.crt
```
The actual path to the correct CA bundle may vary in your setup.

Honestly I don't remember what DCP stands for, but it is a release of the Intel Acceleration Stack for an Arria10 device which includes firmware (or so called "blue bitstream"). Interface IDs differ for different firmware releases (AFU IDs don't).

rojkov, 2020-04-29:
Also you might want to follow #358 as it changes the format of the name mappings and the way how the webhook is configured.

mythi, 2020-07-01:
@xxinran can you try with the latest master (and follow the new/improved instructions)?

bart0sh, 2020-09-16:
Closing as name mappings format was changed in the latest release.

closed 2020-09-16 by bart0sh