intel/intel-device-plugins-for-kubernetes #385: Publish One Device Operator in Operator Hub

msivosuo, 2020-06-03:
Publish One Device Operator in Operator Hub

rojkov, 2020-09-15:
I have to pause this task due to https://github.com/operator-framework/operator-lifecycle-manager/issues/1315#issuecomment-692597350

rojkov, 2020-11-25:
The progress can be tracked at https://github.com/operator-framework/community-operators/pull/2750 now.

msivosuo, 2020-12-15:
@rojkov - could we ping operator-framework people on this to get some traction?

rojkov, 2020-12-15:
I've managed to find to [this instruction](https://operator-framework.github.io/community-operators/#test-your-operator-before-submitting-a-pr) on how to reproduce the CI tests locally. But I'm busy with a QAT engine issue ATM. Will try it later.

pohly, 2020-12-15:
@avalluri has added CI testing of the PMEM-CSI operator. He can provide further pointers and help in case that you need it.

rojkov, 2021-01-05:
Closing as https://github.com/operator-framework/community-operators/pull/2750 got merged.

closed 2021-01-05 by rojkov