intel/intel-device-plugins-for-kubernetes #392: [linter] Enable wsl check

rojkov, 2020-06-08:
in ${PROJECT_ROOT}/.golangci.yml to improve readability.

kad, 2020-10-14:
Suggestion: maybe check the `gocognit` that seems to be doing same as wsl + more.

rojkov, 2020-10-14:
@kad it's enabled already (see https://github.com/intel/intel-device-plugins-for-kubernetes/blob/master/.golangci.yml#L15)
I even had to relax it a bit to make it pass. The problem was 
```
pkg/topology/topology.go:257:1: cognitive complexity 31 of func `GetTopologyInfo` is high (> 30) (gocognit)
func GetTopologyInfo(devs []string) (*pluginapi.TopologyInfo, error) {
```

mythi, 2021-12-15:
With the current `main` we get 50 hits which are mostly about adding empty lines. I don't think this is meaningful item to track/add.

@rojkov are you fine closing?

rojkov, 2021-12-15:
After a bit of internal struggle... Yes, I'm fine. :)

bart0sh, 2021-12-15:
Can we fix those 50 hits or I'm missing something?

bart0sh, 2021-12-15:
I started to work on it and after fixing bunch of hundreds I realized that the check outputs only 50 issues by default :)

PR #807 (2021-12-17): lint: enable wsl check
Fixes: #392
commit cec004c3 (2021-12-17): lint: enable wsl check

Fixes: #392
closed 2021-12-17 by mythi