intel/intel-device-plugins-for-kubernetes #394: Consider using go-bindata to keep object definitions in YAMLs rather than in Go code

rojkov, 2020-06-09:
Currently in the operator we construct DaemonSet objects directly as a Go structure:
```go
ds := &apps.DaemonSet{
    ObjectMeta: metav1.ObjectMeta{...},
    Spec: apps.DaemonSetSpec{...},
}
```
This way we end up with very long though simple functions and this prevents us from enabling some linter checks like `funlen`.

Evaluate moving object definitions to YAML files and generating Go files with go-bindata the way it's done in OpenShift operators: https://github.com/openshift/cluster-ingress-operator/tree/master/pkg/manifests.

Try to annotate the generated go-bindata files with buildtags so that linter checks don't go crazy: e.g. generate debug and release versions of the Go files and enable linter checks only for the debug version.

mythi, 2021-02-18:
@rojkov https://golang.org/pkg/embed/

rojkov, 2021-02-18:
Nice new feature. Looks cleaner than go-bindata

mythi, 2021-02-24:
@rojkov I gave it a try using tips from [this blog post](https://blog.carlmjohnson.net/post/2021/how-to-use-go-embed/) 

```
package main

import (
	_ "embed"
	"fmt"
	"sigs.k8s.io/yaml"
	apps "k8s.io/api/apps/v1"

)
var (
    //go:embed deployments/qat_plugin/base/intel-qat-plugin.yaml
    b []byte
    s = func() (s apps.DaemonSet) {
	   err := yaml.Unmarshal(b, &s)
	   if err != nil {
		   panic(err)
	   }
	   return
    }()
)

func main() {
    fmt.Printf("s: %#v\n", s)
}
```

mythi, 2021-06-17:
Updated the title: Go embed likely better option than go-bindata.

rojkov, 2021-06-22:
Moving the Go code to YAML files may enable us to add the [funlen](https://github.com/ultraware/funlen) check to `.golangci.yaml`.

mythi, 2021-11-15:
@bart0sh one problem to solve first is how to make `embed` to find the YAMLs because the embedded files must be under same dir hierarcy (and symlinks are not allowed) as the code that embeds them.

pohly, 2021-11-15:
In PMEM-CSI, we treat our `deploy` directory as a Go package. That package then embeds the YAML files and exports them with a suitable API.

bart0sh, 2021-11-15:
@mythi it worked this way: https://github.com/intel/intel-device-plugins-for-kubernetes/pull/747
We can either symlink other embed.go for other plugins or just copy this file as it should be identical for all of them.

mythi, 2021-11-15:
@bart0sh this looks good to me (and thanks @pohly!)

> We can either symlink embed.go for other plugins or just copy this file as it should be identical for all of them.

or have `deployments/embed.go` that pulls in all devices?

bart0sh, 2021-11-15:
@mythi thanks. that's even better as yamls are not always cooperate with that simple wildcard pattern. will do.

PR #747 (2021-11-23): operator: use go:embed to generate daemonset objects
Used go:embed to generate plugin daemonset objects instead of creating them manually in controller.go for all plugins.

Fixes: #394
closed 2021-11-23 by rojkov