intel/intel-device-plugins-for-kubernetes #400: harmonize imagePullPolicy usage

mythi, 2020-06-16:
it looks our `imagePullPolicy` usage isn't consistent between the plugins. Define a setting/kustomize overlay that suits all the plugins.

mythi, 2020-09-23:
Based on our experience, `imagePullPolicy: Always` for the master branch is the least error prone.

The opens are:
- is this OK for the release branches too?
- should we document how to add a kustomize overlay to change it if `Always` is not desired?

bart0sh, 2020-09-23:
I think it's ok to have 'Always' policy for all branches.

bart0sh, 2020-11-12:
> should we document how to add a kustomize overlay to change it if Always is not desired?

That would be great to have it documented. It would be even better to have ready to use kustomize overlays to switch to IfNotPresent policy.

PR #504 (2020-11-13): Change imagePullPolicy to Always
These patches change imagePullPolicy to Always in base deployments
and demos as discussed in #400.

In case a specific imagePullPolicy is required, it can be provided
through the relevant kustomization.yaml.

Closes #400.
closed 2020-11-13 by rojkov