intel/intel-device-plugins-for-kubernetes #419: FPGA Plugin e2e test fails

bart0sh, 2020-07-02:
It fails with below output. ```kubectl apply -f https://raw.githubusercontent.com/intel/intel-device-plugins-for-kubernetes/master/deployments/fpga_admissionwebhook/mappings-collection.yaml``` doesn't help.

```
> KUBECONFIG=~/.kube/config go test ./test/e2e/... -args -ginkgo.focus "FPGA Plugin"
I0702 18:57:18.832669    2848 test_context.go:423] Tolerating taints "node-role.kubernetes.io/master" when considering if nodes are ready
Jul  2 18:57:18.832: INFO: The --provider flag is not set. Continuing as if --provider=skeleton had been used.
=== RUN   TestDevicePlugins
Running Suite: E2E Device Plugins Suite
=======================================
Random Seed: 1593705438 - Will randomize all specs
Will run 1 of 7 specs

Jul  2 18:57:18.833: INFO: >>> kubeConfig: /home/ed/.kube/config
STEP: Deleting namespaces
Jul  2 18:57:18.868: INFO: namespace : intelfpgawebhook-system api call to delete is complete 
Jul  2 18:57:18.922: INFO: namespace : intelfpgaplugin-system api call to delete is complete 
Jul  2 18:57:18.922: INFO: Waiting for deletion of the following namespaces: [intelfpgaplugin-system intelfpgawebhook-system]
STEP: Waiting for namespaces to vanish
Jul  2 18:57:24.927: INFO: Waiting up to 30m0s for all (but 0) nodes to be schedulable
Jul  2 18:57:24.933: INFO: Waiting up to 10m0s for all pods (need at least 0) in namespace 'kube-system' to be running and ready
Jul  2 18:57:24.982: INFO: 8 / 8 pods in namespace 'kube-system' are running and ready (0 seconds elapsed)
Jul  2 18:57:24.982: INFO: expected 2 pod replicas in namespace 'kube-system', 2 are Running and Ready.
Jul  2 18:57:24.982: INFO: e2e test version: v0.0.0-master+$Format:%h$
Jul  2 18:57:24.983: INFO: kube-apiserver version: v1.18.5
SSSSS
------------------------------
FPGA Plugin E2E tests 
  Run FPGA plugin tests
  /srv/demo/intel-device-plugins-for-kubernetes/test/e2e/fpga/fpga.go:64
[BeforeEach] FPGA Plugin E2E tests
  /srv/demo/go/pkg/mod/k8s.io/kubernetes@v1.18.2/test/e2e/framework/framework.go:178
STEP: Creating a kubernetes client
Jul  2 18:57:24.984: INFO: >>> kubeConfig: /home/ed/.kube/config
STEP: Building a namespace api object, basename fpgaplugin-e2e
Jul  2 18:57:25.111: INFO: No PodSecurityPolicies found; assuming PodSecurityPolicy is disabled.
STEP: Waiting for a default service account to be provisioned in namespace
[It] Run FPGA plugin tests
  /srv/demo/intel-device-plugins-for-kubernetes/test/e2e/fpga/fpga.go:64
STEP: namespace fpgaplugin-e2e-8081: deploying webhook
Jul  2 18:57:27.118: INFO: Running '/usr/bin/kubectl --kubeconfig=/home/ed/.kube/config apply -k /tmp/fpgawebhooke2etest-fpgaplugin-e2e-8081218558422'
Jul  2 18:57:28.125: INFO: stderr: ""
Jul  2 18:57:28.125: INFO: stdout: "namespace/intelfpgawebhook-system created\ncustomresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com unchanged\ncustomresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com unchanged\nmutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgawebhook-mutating-webhook-configuration configured\nclusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role configured\nclusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding unchanged\nservice/intelfpgawebhook-webhook-service created\ndeployment.apps/intelfpgawebhook-webhook created\ncertificate.cert-manager.io/intelfpgawebhook-serving-cert created\nissuer.cert-manager.io/intelfpgawebhook-selfsigned-issuer created\n"
Jul  2 18:57:30.130: INFO: Got 0 pods running and ready, expect: 1
Jul  2 18:57:32.129: INFO: Got 0 pods running and ready, expect: 1
Jul  2 18:57:34.129: INFO: Got 0 pods running and ready, expect: 1
Jul  2 18:57:36.129: INFO: Got 0 pods running and ready, expect: 1
STEP: deploying mappings
Jul  2 18:57:38.129: INFO: Running '/usr/bin/kubectl --kubeconfig=/home/ed/.kube/config apply -n fpgaplugin-e2e-8081 -f /srv/demo/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default/../mappings-collection.yaml'
Jul  2 18:57:39.073: INFO: stderr: ""
Jul  2 18:57:39.073: INFO: stdout: "fpgaregion.fpga.intel.com/arria10.dcp1.0 created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.0-nlb0-orchestrated created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.0-nlb3-orchestrated created\nfpgaregion.fpga.intel.com/arria10.dcp1.1 created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.1-nlb0-orchestrated created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.1-nlb3-orchestrated created\nfpgaregion.fpga.intel.com/arria10.dcp1.2 created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.2-nlb0-orchestrated created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.2-nlb0-preprogrammed created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.2-nlb3-orchestrated created\nacceleratorfunction.fpga.intel.com/arria10.dcp1.2-nlb3-preprogrammed created\nfpgaregion.fpga.intel.com/d5005 created\nacceleratorfunction.fpga.intel.com/d5005-nlb0-orchestrated created\nacceleratorfunction.fpga.intel.com/d5005-nlb3-orchestrated created\nacceleratorfunction.fpga.intel.com/d5005-nlb3-preprogrammed created\nacceleratorfunction.fpga.intel.com/d5005-matrix-mult-orchestrated created\n"
STEP: namespace fpgaplugin-e2e-8081: deploying FPGA plugin in region mode
Jul  2 18:57:39.073: INFO: Running '/usr/bin/kubectl --kubeconfig=/home/ed/.kube/config delete -k /tmp/fpgaplugine2etest-fpgaplugin-e2e-8081399445821'
Jul  2 18:57:39.546: INFO: rc: 1
Jul  2 18:57:39.546: INFO: Running '/usr/bin/kubectl --kubeconfig=/home/ed/.kube/config apply -k /tmp/fpgaplugine2etest-fpgaplugin-e2e-8081399445821'
Jul  2 18:57:41.050: INFO: stderr: ""
Jul  2 18:57:41.050: INFO: stdout: "namespace/intelfpgaplugin-system created\ncustomresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com unchanged\ncustomresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created\nmutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgaplugin-mutating-webhook-configuration created\nclusterrole.rbac.authorization.k8s.io/intelfpgaplugin-manager-role created\nclusterrole.rbac.authorization.k8s.io/intelfpgaplugin-node-getter created\nclusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-get-nodes created\nclusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-manager-rolebinding created\nservice/intelfpgaplugin-webhook-service created\ndeployment.apps/intelfpgaplugin-webhook created\ndaemonset.apps/intelfpgaplugin-fpgadeviceplugin created\ncertificate.cert-manager.io/intelfpgaplugin-serving-cert created\nissuer.cert-manager.io/intelfpgaplugin-selfsigned-issuer created\n"
STEP: waiting for intel-fpga-plugin availability
Jul  2 18:57:43.120: INFO: Got 0 pods running and ready, expect: 1
Jul  2 18:57:45.090: INFO: Got 0 pods running and ready, expect: 1
Jul  2 18:57:47.061: INFO: Got 0 pods running and ready, expect: 1
STEP: checking if the resource is allocatable
Jul  2 18:57:49.073: INFO: Waiting up to 30s for any positive allocatable resource "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6"
Jul  2 18:57:50.078: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 1.004500343s
Jul  2 18:57:51.089: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 2.015911449s
Jul  2 18:57:52.139: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 3.065171692s
Jul  2 18:57:53.078: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 4.00443698s
Jul  2 18:57:54.078: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 5.004224043s
Jul  2 18:57:55.103: INFO: Found 0 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 6.029356643s
Jul  2 18:57:56.078: INFO: Found 2 of "fpga.intel.com/region-69528db6eb31577a8c3668f9faa081f6". Elapsed: 7.004356915s
STEP: submitting a pod requesting correct FPGA resources
[... 103 lines cut ...]
      }
      admission webhook "fpga.mutator.webhooks.intel.com" denied the request: no such resource: "fpga.intel.com/arria10.dcp1.2-nlb3-orchestrated"
  occurred

  /srv/demo/intel-device-plugins-for-kubernetes/test/e2e/fpga/fpga.go:140
------------------------------
S

Summarizing 1 Failure:

[Fail] FPGA Plugin E2E tests [It] Run FPGA plugin tests 
/srv/demo/intel-device-plugins-for-kubernetes/test/e2e/fpga/fpga.go:140

Ran 1 of 7 Specs in 37.455 seconds
FAIL! -- 0 Passed | 1 Failed | 0 Pending | 6 Skipped
--- FAIL: TestDevicePlugins (37.46s)
FAIL
[... cut ...] gins-for-kubernetes/test/e2e	37.495s
?   	github.com/intel/intel-device-plugins-for-kubernetes/test/e2e/fpga	[no test files]
?   	github.com/intel/intel-device-plugins-for-kubernetes/test/e2e/fpgaadmissionwebhook	[no test files]
?   	github.com/intel/intel-device-plugins-for-kubernetes/test/e2e/gpu	[no test files]
?   	github.com/intel/intel-device-plugins-for-kubernetes/test/e2e/qat	[no test files]
?   	github.com/intel/intel-device-plugins-for-kubernetes/test/e2e/utils	[no test files]
FAIL
```

rojkov, 2020-07-02:
```
Jul  2 18:57:56.150: INFO: intelfpgawebhook-webhook-7d6bcb8b57-psv5t started at 2020-07-02 18:57:29 +0300 EEST (0+1 container statuses recorded)
Jul  2 18:57:56.150: INFO: 	Container manager ready: true, restart count 0
Jul  2 18:57:56.150: INFO: intelfpgaplugin-webhook-7d6bcb8b57-tbqkx started at 2020-07-02 18:57:40 +0300 EEST (0+1 container statuses recorded)
Jul  2 18:57:56.150: INFO: 	Container manager ready: true, restart count 0
```

I see there two webhooks running. Obviously the line
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/0171c39da95f012e9ab4f9b3753cc7ecf4e0ee67/test/e2e/fpga/fpga.go#L67
Is not needed now in the test. And the mappings need to be added after the plguin is deployed
https://github.com/intel/intel-device-plugins-for-kubernetes/blob/0171c39da95f012e9ab4f9b3753cc7ecf4e0ee67/test/e2e/fpga/fpga.go#L94

bart0sh, 2020-07-02:
@rojkov thanks. I've submitted the fix. Please, review.

PR #420 (2020-07-03): e2e: fix FPGA tests
Fixes: #419
commit 49d3514e (2020-07-03): e2e: fix FPGA tests

Fixes: #419
closed 2020-07-03 by rojkov