intel/intel-device-plugins-for-kubernetes #444: Publish DSA operator on Operator Hub

msivosuo, 2020-09-02:
Publish DSA operator on Operator Hub

bart0sh, 2021-04-14:
@mythi @rojkov Created PR: https://github.com/operator-framework/community-operators/pull/3573
please, review

mythi, 2021-04-14:
@bart0sh LGTM, one nit: maybe add dsa to CSV keywords.

bart0sh, 2021-04-14:
@mythi thank you for the review. will do.

BTW, some tests fail on the plugin. If you know how to fix that please let me know

bart0sh, 2021-04-15:
Done: https://operatorhub.io/operator/intel-device-plugins-operator

closed 2021-04-15 by bart0sh

msivosuo, 2021-04-15:
Great to see it published, thanks! Any reason we do not have YAML example for DSA? For other devices there's one.

bart0sh, 2021-04-15:
No particular reason. Thanks for pointing out. Will fix.

bart0sh, 2021-04-15:
Here is the fix: https://github.com/operator-framework/community-operators/pull/3590
I'll need to convince maintainers that this is a cosmetic change, otherwise we'll have to release 0.20.1 or wait until 0.21.0 is ready