intel/intel-device-plugins-for-kubernetes #452: update to k8s 1.19 APIs

mythi, 2020-09-11:


rojkov, 2020-10-15:
K8s 1.19 has updated dependency on `github.com/go-logr/logr` to v0.2.0, but the latest released `sigs.k8s.io/controller-runtime` (0.6.3) still depends on logr v0.1.0.

controller-runtime 0.7.0, which seems to have the dependency updated, hasn't been released yet.

This prevents us from updating to k8s 1.19.

mythi, 2020-10-16:
> controller-runtime 0.7.0, which seems to have the dependency updated, hasn't been released yet.

Let's wait 0.7.0. It looks it's not too far away...

mythi, 2020-11-02:
> controller-runtime 0.7.0, which seems to have the dependency updated, hasn't been released yet.

@rojkov can you start with a draft PR that updates k8s 1.19 and controller-runtime 0.7.0-alpha4. Or would `Replace` v0.1.0 -> v0.2.0 work?

rojkov, 2020-11-02:
Ok. I'll start with moving to controller-runtime 0.7.0-alpha4. It seems to be a safer bet.

rojkov, 2020-11-02:
Now all deps seem to resolve ok. I pushed the new `go.mod` to my private `release-0.19` branch. The next step is to make the code compile.

PR #493 (2020-11-10): Upgrade to k8s v1.19.3
Closes #452
closed 2020-11-10 by bart0sh