intel/intel-device-plugins-for-kubernetes #461: FPGA: webhook deployment fails with "TLS certificate file is not set" error

bart0sh, 2020-09-16:
Steps to reproduce:

```
> k create -f https://github.com/jetstack/cert-manager/releases/download/v1.0.0/cert-manager.yaml
customresourcedefinition.apiextensions.k8s.io/certificaterequests.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/certificates.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/challenges.acme.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/clusterissuers.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/issuers.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/orders.acme.cert-manager.io created
namespace/cert-manager created
serviceaccount/cert-manager-cainjector created
serviceaccount/cert-manager created
serviceaccount/cert-manager-webhook created
clusterrole.rbac.authorization.k8s.io/cert-manager-cainjector created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-issuers created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-clusterissuers created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-certificates created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-orders created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-challenges created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-ingress-shim created
clusterrole.rbac.authorization.k8s.io/cert-manager-view created
clusterrole.rbac.authorization.k8s.io/cert-manager-edit created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-cainjector created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-issuers created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-clusterissuers created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-certificates created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-orders created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-challenges created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-ingress-shim created
role.rbac.authorization.k8s.io/cert-manager-cainjector:leaderelection created
role.rbac.authorization.k8s.io/cert-manager:leaderelection created
role.rbac.authorization.k8s.io/cert-manager-webhook:dynamic-serving created
rolebinding.rbac.authorization.k8s.io/cert-manager-cainjector:leaderelection created
rolebinding.rbac.authorization.k8s.io/cert-manager:leaderelection created
rolebinding.rbac.authorization.k8s.io/cert-manager-webhook:dynamic-serving created
service/cert-manager created
service/cert-manager-webhook created
deployment.apps/cert-manager-cainjector created
deployment.apps/cert-manager created
deployment.apps/cert-manager-webhook created
mutatingwebhookconfiguration.admissionregistration.k8s.io/cert-manager-webhook created
validatingwebhookconfiguration.admissionregistration.k8s.io/cert-manager-webhook created

> k get pods -A
NAMESPACE      NAME                                       READY   STATUS    RESTARTS   AGE
cert-manager   cert-manager-b599fdb8f-ghrgb               1/1     Running   0          10s
cert-manager   cert-manager-cainjector-5df66c9c94-6xrm9   1/1     Running   0          10s
cert-manager   cert-manager-webhook-679cf5c9f9-rqjk8      0/1     Running   0          10s
kube-system    coredns-66bff467f8-2ngwt                   1/1     Running   0          76d
kube-system    coredns-66bff467f8-dp98f                   1/1     Running   0          76d
kube-system    etcd-r720-1                                1/1     Running   0          76d
kube-system    kube-apiserver-r720-1                      1/1     Running   0          76d
kube-system    kube-controller-manager-r720-1             1/1     Running   112        76d
kube-system    kube-proxy-5qqkh                           1/1     Running   0          76d
kube-system    kube-scheduler-r720-1                      1/1     Running   118        76d
kube-system    weave-net-8r8f9                            2/2     Running   0          76d

> k create -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default?ref=master
namespace/intelfpgawebhook-system created
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgawebhook-mutating-webhook-configuration created
clusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding created
service/intelfpgawebhook-webhook-service created
deployment.apps/intelfpgawebhook-webhook created
certificate.cert-manager.io/intelfpgawebhook-serving-cert created
issuer.cert-manager.io/intelfpgawebhook-selfsigned-issuer created

> k get pods -A |grep intelfpgawebhook
intelfpgawebhook-system   intelfpgawebhook-webhook-674bb4d5c7-p7b4m   0/1     Error     0          8s

> k logs intelfpgawebhook-webhook-674bb4d5c7-p7b4m -n intelfpgawebhook-system 
F0916 10:51:53.846360       1 fpga_admissionwebhook.go:218] TLS certificate file is not set

```

bart0sh, 2020-09-16:
@mythi I couldn't reproduce this with the latest release:
```
> k create -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default?ref=v0.18.0
namespace/intelfpgawebhook-system created
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgawebhook-mutating-webhook-configuration created
clusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding created
service/intelfpgawebhook-webhook-service created
deployment.apps/intelfpgawebhook-webhook created
certificate.cert-manager.io/intelfpgawebhook-serving-cert created
issuer.cert-manager.io/intelfpgawebhook-selfsigned-issuer created

> k get pods -A | grep fpga
intelfpgawebhook-system   intelfpgawebhook-webhook-b694c84f8-8bcz7   1/1     Running   0          4m12s
```

rojkov, 2020-09-16:
I couldn't reproduce it with the current master and cert-manager v1.0.0

mythi, 2020-09-17:
`make test-with-kind` worked for me too

mythi, 2020-09-21:
@bart0sh are the e2e tests passing for you?

msivosuo, 2020-09-21:
@bart0sh - would you check one more time and close if not reproducible?

bart0sh, 2020-09-22:
It is reproducible. Sometimes it works, but most of the time I can see the same error.

bart0sh, 2020-09-22:
@mythi I didn't try to run e2e tests because webhook container doesn't run. I'm not sure it even worth trying in these conditions.

bart0sh, 2020-09-22:
Just tried one more time. This time the webhook container was OOMKilled, but the log is the same.
Here is a full console log:
```
> k get pods -A
NAMESPACE     NAME                             READY   STATUS    RESTARTS   AGE
kube-system   coredns-66bff467f8-2ngwt         1/1     Running   0          82d
kube-system   coredns-66bff467f8-dp98f         1/1     Running   0          82d
kube-system   etcd-r720-1                      1/1     Running   0          82d
kube-system   kube-apiserver-r720-1            1/1     Running   0          82d
kube-system   kube-controller-manager-r720-1   1/1     Running   112        82d
kube-system   kube-proxy-5qqkh                 1/1     Running   0          82d
kube-system   kube-scheduler-r720-1            1/1     Running   118        82d
kube-system   weave-net-8r8f9                  2/2     Running   0          82d

> k create -f https://github.com/jetstack/cert-manager/releases/download/v1.0.0/cert-manager.yaml
customresourcedefinition.apiextensions.k8s.io/certificaterequests.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/certificates.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/challenges.acme.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/clusterissuers.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/issuers.cert-manager.io created
customresourcedefinition.apiextensions.k8s.io/orders.acme.cert-manager.io created
namespace/cert-manager created
serviceaccount/cert-manager-cainjector created
serviceaccount/cert-manager created
serviceaccount/cert-manager-webhook created
clusterrole.rbac.authorization.k8s.io/cert-manager-cainjector created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-issuers created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-clusterissuers created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-certificates created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-orders created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-challenges created
clusterrole.rbac.authorization.k8s.io/cert-manager-controller-ingress-shim created
clusterrole.rbac.authorization.k8s.io/cert-manager-view created
clusterrole.rbac.authorization.k8s.io/cert-manager-edit created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-cainjector created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-issuers created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-clusterissuers created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-certificates created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-orders created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-challenges created
clusterrolebinding.rbac.authorization.k8s.io/cert-manager-controller-ingress-shim created
role.rbac.authorization.k8s.io/cert-manager-cainjector:leaderelection created
role.rbac.authorization.k8s.io/cert-manager:leaderelection created
role.rbac.authorization.k8s.io/cert-manager-webhook:dynamic-serving created
rolebinding.rbac.authorization.k8s.io/cert-manager-cainjector:leaderelection created
rolebinding.rbac.authorization.k8s.io/cert-manager:leaderelection created
rolebinding.rbac.authorization.k8s.io/cert-manager-webhook:dynamic-serving created
service/cert-manager created
service/cert-manager-webhook created
deployment.apps/cert-manager-cainjector created
deployment.apps/cert-manager created
deployment.apps/cert-manager-webhook created
mutatingwebhookconfiguration.admissionregistration.k8s.io/cert-manager-webhook created
validatingwebhookconfiguration.admissionregistration.k8s.io/cert-manager-webhook created

> k get pods -A
NAMESPACE      NAME                                       READY   STATUS    RESTARTS   AGE
cert-manager   cert-manager-b599fdb8f-h9gr4               1/1     Running   0          9s
cert-manager   cert-manager-cainjector-5df66c9c94-nrw4m   1/1     Running   0          9s
cert-manager   cert-manager-webhook-679cf5c9f9-sqv92      0/1     Running   0          9s
kube-system    coredns-66bff467f8-2ngwt                   1/1     Running   0          82d
kube-system    coredns-66bff467f8-dp98f                   1/1     Running   0          82d
kube-system    etcd-r720-1                                1/1     Running   0          82d
kube-system    kube-apiserver-r720-1                      1/1     Running   0          82d
kube-system    kube-controller-manager-r720-1             1/1     Running   112        82d
kube-system    kube-proxy-5qqkh                           1/1     Running   0          82d
kube-system    kube-scheduler-r720-1                      1/1     Running   118        82d
kube-system    weave-net-8r8f9                            2/2     Running   0          82d

> kubectl create -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default?ref=master
[... 4 lines cut ...]
clusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding created
service/intelfpgawebhook-webhook-service created
deployment.apps/intelfpgawebhook-webhook created
certificate.cert-manager.io/intelfpgawebhook-serving-cert created
issuer.cert-manager.io/intelfpgawebhook-selfsigned-issuer created

> k get pods -A
[... cut ...]
kube-system               kube-controller-manager-r720-1              1/1     Running     112        82d
kube-system               kube-proxy-5qqkh                            1/1     Running     0          82d
kube-system               kube-scheduler-r720-1                       1/1     Running     118        82d
kube-system               weave-net-8r8f9                             2/2     Running     0          82d

> k logs -n intelfpgawebhook-system  intelfpgawebhook-webhook-674bb4d5c7-nskvz
F0922 10:38:36.938687       1 fpga_admissionwebhook.go:218] TLS certificate file is not set
```

bart0sh, 2020-09-22:
I've never  been able to reproduce it with v0.18.0:
```
> kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_admissionwebhook/default?ref=v0.18.0
namespace/intelfpgawebhook-system created
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgawebhook-mutating-webhook-configuration created
clusterrole.rbac.authorization.k8s.io/intelfpgawebhook-manager-role created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgawebhook-manager-rolebinding created
service/intelfpgawebhook-webhook-service created
deployment.apps/intelfpgawebhook-webhook created
certificate.cert-manager.io/intelfpgawebhook-serving-cert created
issuer.cert-manager.io/intelfpgawebhook-selfsigned-issuer created

> k get pods -A
NAMESPACE                 NAME                                       READY   STATUS    RESTARTS   AGE
cert-manager              cert-manager-b599fdb8f-h9gr4               1/1     Running   0          8m18s
cert-manager              cert-manager-cainjector-5df66c9c94-nrw4m   1/1     Running   0          8m18s
cert-manager              cert-manager-webhook-679cf5c9f9-sqv92      1/1     Running   0          8m18s
intelfpgawebhook-system   intelfpgawebhook-webhook-b694c84f8-76v7j   1/1     Running   0          10s
kube-system               coredns-66bff467f8-2ngwt                   1/1     Running   0          82d
kube-system               coredns-66bff467f8-dp98f                   1/1     Running   0          82d
kube-system               etcd-r720-1                                1/1     Running   0          82d
kube-system               kube-apiserver-r720-1                      1/1     Running   0          82d
kube-system               kube-controller-manager-r720-1             1/1     Running   112        82d
kube-system               kube-proxy-5qqkh                           1/1     Running   0          82d
kube-system               kube-scheduler-r720-1                      1/1     Running   118        82d
kube-system               weave-net-8r8f9                            2/2     Running   0          82d

> k logs -n intelfpgawebhook-system intelfpgawebhook-webhook-b694c84f8-76v7j
I0922 10:44:55.543179       1 controller.go:91] Starting controller
I0922 10:44:55.844003       1 controller.go:109] Started controller workers
```

bart0sh, 2020-09-22:
@mythi @rojkov I think it worth investigating, especially after getting OOMKilled status. You have access to the host I guess.

bart0sh, 2020-09-22:
I can bisect it if needed.

rojkov, 2020-09-22:
> fpga_admissionwebhook.go:218] TLS certificate file is not set

This message means that the image with the binary used in the deployment doesn't correspond the new pod spec where the TLS cert is not specified. Could you try to refresh the image?

bart0sh, 2020-09-22:
@rojkov Thank you. Removing all our images and reinstalling webhook helped.

closed 2020-09-22 by bart0sh

mythi, 2020-09-22:
@bart0sh @rojkov #400 :-)

Good that it works!