intel/intel-device-plugins-for-kubernetes #499: updates to clean-up non-inclusive language

mythi, 2020-11-11:
Meta-issue to track inclusive language updates

- [ ] docs /cc @intelkevinputnam (can you help?)
- [ ] Dockerfiles (@mythi)
- [ ]  run-dpdk-test (@mythi)
- [ ] FGPA cri_hook test data, topology (@bart0sh)
- [ ] master -> main (all)

intelkevinputnam, 2021-01-14:
I've been trying out https://github.com/get-woke/woke to check for inclusive language problems. Found no issues with the docs.

mythi, 2021-01-14:
@intelkevinputnam we wanted ask help updating the docs when we rename 'master' to 'main'

mythi, 2021-01-25:
@intelkevinputnam is that something you could help with?

intelkevinputnam, 2021-01-26:
@mythi Since the new year my focus has shifted to tools development. @dougtw may be able to help.

mythi, 2021-02-23:
done!

closed 2021-02-23 by mythi