intel/intel-device-plugins-for-kubernetes #514: documentation fixes

mythi, 2020-11-25:
- [ ] release-0.19 specific docs generation
- [ ] top-level README: links to demo screencasts give 404
- [ ] top-level README: better anchor links for fpga/sgx device plugins (now it's 'device plugin' and 'SGX plugin', respectively)

intelkevinputnam, 2020-12-15:
Item 2 (links to demo screencasts give 404) is occurring because the links to the demo readme are linking to the directory and not the document.

Current reference: 
```
[GPU plugin demo video](demo#intel-gpu-device-plugin-demo-video)
```
Should be: 
```
[GPU plugin demo video](demo/readme.md#intel-gpu-device-plugin-demo-video)
```

intelkevinputnam, 2020-12-15:
Item 3 (better anchor links for fpga/sgx device plugins). I'm afraid I'm not sure what this is referring to.

mythi, 2020-12-15:
@intelkevinputnam item 3 refers to the sentences: "The [SGX|FPGA] plugin comes in three parts." and then anchor links to their respective subsections. I wanted to have these consistent between FPGA/SGX.

intelkevinputnam, 2021-01-14:
@mythi If I understand correctly, you want to change the naming convention? Create explicit anchors using HTML (preferably in the line above the header like this:

``` markdown
- the [plugin](#sgx-plugin)

<a name="sgx-plugin"></a>
####  plugin
```

mythi, 2021-01-14:
@intelkevinputnam ok, thanks!

mythi, 2021-01-15:
> Item 2 (links to demo screencasts give 404) is occurring because the links to the demo readme are linking to the directory and not the document.

@intelkevinputnam what's the right way to convert links that point to repo directories without an .md file

closed 2021-01-20 by rojkov

mythi, 2021-01-20:
I verified the generated docs and everything seems to be as expected!

PR #546 (2021-01-20): README: fix links and add docs for release-0.19
Closes: #514 

/cc @intelkevinputnam does this look right?