intel/intel-device-plugins-for-kubernetes #547: Imcompatibility of admissionReviewVersions

xxinran, 2021-01-19:
I have tried to deploy the FPGA plugin with the latest code, but an error occurs:

The MutatingWebhookConfiguration "intelfpgaplugin-mutating-webhook-configuration" is invalid: webhooks[0].admissionReviewVersions: Invalid value: []string{"v1beta"}: must include at least one of v1beta1

I have changed to v1beta1, and it works.  Is it a bug or I misconfigured something?

mythi, 2021-01-19:
It's a bug. Thanks for reporting! It is recommended to deploy released versions using the `?ref=release-0.19` suffix in your deployment URL.

mythi, 2021-01-19:
@xxinran what is your kubernetes version and do you have the latest `:devel` image on the node?

xxinran, 2021-01-19:
> @xxinran what is your kubernetes version and do you have the latest admission webhook `:devel` image on the node?

yes, I use intel/intel-fpga-admissionwebhook:devel  image, and my Kubernetes version is 1.16.

mythi, 2021-01-19:
> yes, I use intel/intel-fpga-admissionwebhook:devel image

is it possible that you have an old image cached on the node? the latest image is `Digest:sha256:ad367e9c1de79d5a7a14e096509ca7b2dcab2a4b9988fe3c5bf20441a6731c7f`

xxinran, 2021-01-19:
I don't think I have other old images cached, please see the details of my images.
```
$docker image inspect  a3af18aebe76
[
    {
        "Id": "sha256:a3af18aebe764c0331860a2bb9c95d1a1ae6550210d48a7a54beb37521d9078b",
        "RepoTags": [
            "intel/intel-fpga-admissionwebhook:devel"
        ],
        "RepoDigests": [
            "intel/intel-fpga-admissionwebhook@sha256:ad367e9c1de79d5a7a14e096509ca7b2dcab2a4b9988fe3c5bf20441a6731c7f"
        ],
...
```

mythi, 2021-01-19:
OK, thanks. Now that I think of it, it's probably not image related.

mythi, 2021-01-19:
> my Kubernetes version is 1.16.

based on the k8s release notes admissionreview v1 was added in v1.16. I'm not able to reproduce this problem with v1.19 so it sounds this could be related to the version you are using. Are you able to use some newer release? If not, our release branches should be used.

mythi, 2021-01-19:
```
$  kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/region
namespace/intelfpgaplugin-system created
Warning: apiextensions.k8s.io/v1beta1 CustomResourceDefinition is deprecated in v1.16+, unavailable in v1.22+; use apiextensions.k8s.io/v1 CustomResourceDefinition
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgaplugin-mutating-webhook-configuration created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-manager-role created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-node-getter created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-get-nodes created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-manager-rolebinding created
service/intelfpgaplugin-webhook-service created
deployment.apps/intelfpgaplugin-webhook created
daemonset.apps/intelfpgaplugin-fpgadeviceplugin created
certificate.cert-manager.io/intelfpgaplugin-serving-cert created
issuer.cert-manager.io/intelfpgaplugin-selfsigned-issuer created
```

or

```
$ kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/fpga_plugin/overlays/af
namespace/intelfpgaplugin-system created
Warning: apiextensions.k8s.io/v1beta1 CustomResourceDefinition is deprecated in v1.16+, unavailable in v1.22+; use apiextensions.k8s.io/v1 CustomResourceDefinition
customresourcedefinition.apiextensions.k8s.io/acceleratorfunctions.fpga.intel.com created
customresourcedefinition.apiextensions.k8s.io/fpgaregions.fpga.intel.com created
mutatingwebhookconfiguration.admissionregistration.k8s.io/intelfpgaplugin-mutating-webhook-configuration created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-manager-role created
clusterrole.rbac.authorization.k8s.io/intelfpgaplugin-node-getter created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-get-nodes created
clusterrolebinding.rbac.authorization.k8s.io/intelfpgaplugin-manager-rolebinding created
service/intelfpgaplugin-webhook-service created
deployment.apps/intelfpgaplugin-webhook created
daemonset.apps/intelfpgaplugin-fpgadeviceplugin created
certificate.cert-manager.io/intelfpgaplugin-serving-cert created
issuer.cert-manager.io/intelfpgaplugin-selfsigned-issuer created
```

xxinran, 2021-01-19:
> based on the k8s release notes admissionreview v1 was added in v1.16. I'm not able to reproduce this problem with v1.19 so it sounds this could be related to the version you are using. Are you able [...]

Thanks for your help, I think it is I tried with the master branch of the intel device plugin, I will switch to the stable branch to retry it.

mythi, 2021-01-19:
I'd more curious to know if you can get it working with the master branch but update the k8s version.

mythi, 2021-01-21:
@xxinran any updates?

xxinran, 2021-01-21:
> @xxinran any updates?

yes, I have try the master branch with Kubernetes 1.20. Everything works well.

mythi, 2021-01-21:
@xxinran Thanks for checking! I'm going to close this.

closed 2021-01-21 by mythi