intel/intel-device-plugins-for-kubernetes #551: mount /dev/ion error with using HDDL card in OpenNESS

jiangzhg, 2021-01-29:
When we use HDDL, we use this plugin and find that an error is reported in the container. "Mount "/dev/ion" fails".
The "/dev/ion" is not necessary, we use the "/dev/shm".

repo: https://github.com/open-ness/openness-experience-kits
tag : openness-20.12.01
flavor: minimal

`  Normal   Scheduled  51s                            default-scheduler  Successfully assigned default/openvino-cons-app-59db6474d6-8wqbw to node1
  Normal   Pulled     0s                             kubelet            Container image "alpine:3.12.0" already present on machine
  Normal   Created    0s                             kubelet            Created container alpine
  Normal   Started    0s                             kubelet            Started container alpine
  Normal   Pulled     <invalid>                      kubelet            Container image "certrequester:1.0" already present on machine
  Normal   Created    <invalid>                      kubelet            Created container certrequester
  Normal   Started    <invalid>                      kubelet            Started container certrequester
  Warning  BackOff    <invalid>                      kubelet            Back-off restarting failed container
  Normal   Pulled     <invalid> (x3 over <invalid>)  kubelet            Container image "openvino-cons-app:1.0" already present on machine
  Normal   Created    <invalid> (x3 over <invalid>)  kubelet            Created container openvino-cons-app
  Warning  Failed     <invalid> (x3 over <invalid>)  kubelet            Error: failed to start container "openvino-cons-app": Error response from daemon: error gathering device information while adding custom device "/dev/ion": no such file or directory
`

mythi, 2021-01-29:
@jiangzhg can you point to the documentation that guides to set-up the drivers you are using. `/dev/ion` was the only HDDL memory driver available when the vpu-plugin was added.

mythi, 2021-01-29:
/cc @alekdu

jiangzhg, 2021-01-29:
> @jiangzhg can you point to the documentation that guides to set-up the drivers you are using. `/dev/ion` was the only HDDL memory driver available when the vpu-plugin was added.

/dev/ion is not only HDDL memory driver, sometime we can use share memory to make hddldaemon work well.
I use the file setup. https://github.com/OpenVisualCloud/Dockerfiles/blob/master/VCAC-A/script/setup_hddl_daemonset.yaml

mythi, 2021-01-29:
> /dev/ion is not only HDDL memory driver, sometime we can use share memory to make hddldaemon work well.

is it documented in the OpenVINO release notes?

jiangzhg, 2021-02-02:
> > /dev/ion is not only HDDL memory driver, sometime we can use share memory to make hddldaemon work well.
> 
> is it documented in the OpenVINO release notes?

yes, please refer this note, [https://github.com/open-ness/specs/blob/master/doc/building-blocks/enhanced-platform-awareness/openness_hddl.md](url)

mythi, 2021-02-02:
@jiangzhg thanks. It seems to be an OpenNESS specific customization. @alekdu what do you think?

xxinran, 2021-02-03:
I have the same error without OPENNESS. When I install vpu plugin, it tried to mount /dev/ion on hostpath to the container, but there is no such file on the host.
Here's the output of `kubecrl describe pod intel-vpu-plugin-x28zw`
```
Events:
  Type     Reason       Age               From               Message
  ----     ------       ----              ----               -------
  Normal   Scheduled    15s               default-scheduler  Successfully assigned default/intel-vpu-plugin-x28zw to inspurserver
  Warning  FailedMount  7s (x5 over 14s)  kubelet            MountVolume.SetUp failed for volume "devion" : hostPath type check failed: /dev/ion is not a character device

```

alekdu, 2021-02-04:
Hi, I think originally /dev/ion is preferred by openvino, the driver is myd_ion in openvino package. Please refer to https://github.com/OpenVisualCloud/VCAC-SW-Analytics/blob/master/VCAC-A/Intel_Media_Analytics_Node/scripts/ovc_hddl_install.sh, line 156.

Later openvino is trying to deprecate ION starting from kernel 5.2 due to ION module is failed to compile against this kernel. We helped to fix the compiling issue and thus ION is still usable.

@xxinran, you need insmod myd_ion on the machine running the device plugin.

xxinran, 2021-02-04:
Thanks for your reply @alekdu , I didn't find myd_ion.ko on my host. 
Actually, I installed the kernel built by https://github.com/OpenVisualCloud/VCAC-SW-Analytics/blob/master/VCAC-A/Intel_Media_Analytics_Host/scripts/build.sh on my host where device plugin runs.  Here is my kernel info:
```
$ uname -r
4.18.0.0e222f9.VCA+
```
There is no myd_ion.ko file on the host. 
Meanwhile, I installed Ubuntu1804 built by https://github.com/OpenVisualCloud/VCAC-SW-Analytics/blob/master/VCAC-A/Intel_Media_Analytics_Node/scripts/vcad_build.sh, and I have install myd_ion module on system on card successfully. 

I am confused about it. My device plugin should run on host, but host don't have myd_ion.ko in kernel. Or should I download the .ko file somewhere?

alekdu, 2021-02-04:
@xxinran I think you need deploy the device plugin to your card OS,  and your card OS need join K8S network as worker node. This is also the setup that OpenVisualCloud uses.

alekdu, 2021-02-04:
@xxinran please refer to this: https://github.com/OpenVisualCloud/Dockerfiles/tree/master/VCAC-A#setup-the-vcac-a-as-kubernetes-node

jiangzhg, 2021-02-04:
> Hi, I think originally /dev/ion is preferred by openvino, the driver is myd_ion in openvino package. Please refer to [...]

@alekdu  did we install myd_ion on CentOS 7.8? The kernel version is 3.10.0-1160

xxinran, 2021-02-07:
@alekdu Thanks for your explanation.  According to the doc you sent, we can set up vca-a card as a worker node in k8s, and use hddl as the resource.
Is it possible that I deploy the server where vca-a card locates as a worker node in k8s, and run vca-a device plugin on this server directly, instead of running the device plugin in card OS.

mythi, 2021-02-08:
@alekdu @xxinran @jiangzhg can you confirm the VPU device plugin is OK and no changes needed there?

alekdu, 2021-02-08:
@mythi I think the myd_ion driver is still supported in openvino and thus we needn't change this.

alekdu, 2021-02-08:
@xxinran, I think there is a PCIe NTB bridge between VCAC-A card and Server this means they belong to two computer system. The device plugin should work closer to the OS who control the HW resource.

mythi, 2021-02-10:
> @mythi I think the myd_ion driver is still supported in openvino and thus we needn't change this.

@alekdu can you close this  issue then if you think this is OK?

alekdu, 2021-02-18:
@jiangzhg sorry for missing your questions about CentOS 7.8 for myd_ion. I guess your CentOS 7.8 is for the server that plugged with VCAC-A card. Actually we are talking about the OS that run on VCAC-A card itself. The formal release should be an Ubuntu OS image...

alekdu, 2021-02-18:
@mythi I wish I could close this issue but sounds like missing a permission...

mythi, 2021-02-18:
closing on behalf of @alekdu

closed 2021-02-18 by mythi