intel/intel-device-plugins-for-kubernetes #604: Device plugin operator build failed.

Ram-srini, 2021-03-26:
While building device plugin operator failed with following error.
Error Message:

/opt/go/src/crypto/tls/cipher_suites.go:18:2: unknown import path "golang_org/x/crypto/chacha20poly1305": cannot find module providing package golang_org/x/crypto/chacha20poly1305
/opt/go/src/crypto/tls/key_agreement.go:20:2: unknown import path "golang_org/x/crypto/curve25519": cannot find module providing package golang_org/x/crypto/curve25519
/opt/go/src/net/http/h2_bundle.go:46:2: unknown import path "golang_org/x/net/http2/hpack": cannot find module providing package golang_org/x/net/http2/hpack
/opt/go/src/net/http/h2_bundle.go:47:2: unknown import path "golang_org/x/net/idna": cannot find module providing package golang_org/x/net/idna
/opt/go/src/net/http/h2_bundle.go:48:2: unknown import path "golang_org/x/net/lex/httplex": cannot find module providing package golang_org/x/net/lex/httplex
/opt/go/src/net/http/transport.go:32:2: unknown import path "golang_org/x/net/proxy": cannot find module providing package golang_org/x/net/proxy
Makefile:81: recipe for target 'operator' failed
make: *** [operator] Error 1

Build command:
make deploy-operator

Branch: Tried on the branches - Master and v0.19.0
OS: Ubuntu 18
Following the instruction as in https://github.com/intel/intel-device-plugins-for-kubernetes#device-plugins-operator

Please let me know if I am missing something here?

mythi, 2021-03-29:
> Please let me know if I am missing something here?

@Ram-srini what's your Go version? We have the container image published so building is not necessary. `kubectl apply -k deployments/operator/default` is enough to deploy the operator.

Ram-srini, 2021-03-29:
I am using go version 1.12. Need to use latest go version 1.15 right?
I tried running kubectl command to deploy operator and got the below error.
kubectl apply -k deployments/operator/default/
..
Resource: "apiextensions.k8s.io/v1beta1, Resource=customresourcedefinitions", GroupVersionKind: "apiextensions.k8s.io/v1beta1, Kind=CustomResourceDefinition"
Name: "sgxdeviceplugins.deviceplugin.intel.com", Namespace: ""
for: "deployments/operator/default/": CustomResourceDefinition.apiextensions.k8s.io "sgxdeviceplugins.deviceplugin.intel.com" is invalid: spec.scope: Invalid value: "Namespaced": field is immutable
..
I am using kubernetes version: 1.20.4.

mythi, 2021-03-29:
> I am using go version 1.12. Need to use latest go version 1.15 right?

your build needs to use Go modules at least. 1.12 does not turn it on by default so `GO111MODULES=ON make deploy-operator` is likely needed.

> `for: "deployments/operator/default/": CustomResourceDefinition.apiextensions.k8s.io "sgxdeviceplugins.deviceplugin.intel.com" is invalid: spec.scope: Invalid value: "Namespaced": field is immutable`

are you using `v0.19.0` release?

Ram-srini, 2021-03-29:
> > I am using go version 1.12. Need to use latest go version 1.15 right?
> 
> your build needs to use Go modules at least. 1.12 does not turn it on by default so `GO111MODULES=ON make deploy-operator` is likely needed.
$go version
go version go1.15.5 linux/amd64
$ GO111MODULES=ON make deploy-operator
cd cmd/operator; go build -tags ""
controller-gen object:headerFile="build/boilerplate/boilerplate.go.txt" paths="./pkg/apis/..."
/bin/sh: 1: controller-gen: not found
Makefile:64: recipe for target 'generate' failed
make: *** [generate] Error 127

> 
> > `for: "deployments/operator/default/": CustomResourceDefinition.apiextensions.k8s.io "sgxdeviceplugins.deviceplugin.intel.com" is invalid: spec.scope: Invalid value: "Namespaced": field is immutable`
> 
> are you using `v0.19.0` release?
Yes. Since that is latest tagged release.

mythi, 2021-03-29:
> `/bin/sh: 1: controller-gen: not found`

We currently don't install this automatically so it needs to be installed separately. Anyway: your build error is fixed. Let's focus on why the CRD deployment fails.

> Yes. Since that is latest tagged release.

cool, let me check this shortly.

Ram-srini, 2021-03-29:
> controller-gen

Ram-srini, 2021-03-29:
Even after installing controller-gen separately, make is failing with same error
to:
Resource: "apiextensions.k8s.io/v1beta1, Resource=customresourcedefinitions", GroupVersionKind: "apiextensions.k8s.io/v1beta1, Kind=CustomResourceDefinition"
Name: "sgxdeviceplugins.deviceplugin.intel.com", Namespace: ""
for: "deployments/operator/default": CustomResourceDefinition.apiextensions.k8s.io "sgxdeviceplugins.deviceplugin.intel.com" is invalid: spec.scope: Invalid value: "Namespaced": field is immutable
Makefile:86: recipe for target 'deploy-operator' failed
make: *** [deploy-operator] Error 1

mythi, 2021-03-29:
@Ram-srini do you have that CRD already in the cluster? `kubectl get crd -A`

Ram-srini, 2021-03-29:
@mythi - Yes are CRDs are exists in the cluster. Here is the list, these are created once i followed the steps as in https://intel.github.io/intel-device-plugins-for-kubernetes/0.19/cmd/sgx_plugin/README.html

acceleratorfunctions.fpga.intel.com    
certificaterequests.cert-manager.io       
certificates.cert-manager.io              
challenges.acme.cert-manager.io            
clusterissuers.cert-manager.io             
dsadeviceplugins.deviceplugin.intel.com    
fpgadeviceplugins.deviceplugin.intel.com   
fpgaregions.fpga.intel.com                 
gpudeviceplugins.deviceplugin.intel.com    
issuers.cert-manager.io                    
orders.acme.cert-manager.io                
qatdeviceplugins.deviceplugin.intel.com    
sgxdeviceplugins.deviceplugin.intel.com

mythi, 2021-03-29:
> Here is the list, these are created once i followed the steps as in 

the list is not from `v0.19.0` release. 

You can get the operator deployed like this (delete the previous deployment first and make sure you start without the CRDs):
`kubectl apply -k https://github.com/intel/intel-device-plugins-for-kubernetes/deployments/operator/default?ref=release-0.19`

There's no need to run `make deploy-operator` after that unless you need to modify the operator.

mythi, 2021-03-30:
@Ram-srini did you get the operator deployed? OK to close this?

Ram-srini, 2021-03-30:
At the last step Create SgxDevicePlugin custom resource managed by the Operator, is not creating any pods.
kubectl apply -f https://raw.githubusercontent.com/intel/intel-device-plugins-for-kubernetes/v0.19.0/deployments/operator/samples/deviceplugin_v1_sgxdeviceplugin.yaml -n sgx-ecdsa-quote
sgxdeviceplugin.deviceplugin.intel.com/sgxdeviceplugin-sample created
But is not creating any pod

kubectl describe SgxDevicePlugin -n sgx-ecdsa-quote
Name:         sgxdeviceplugin-sample
Namespace:    sgx-ecdsa-quote
Labels:       <none>
Annotations:  <none>
API Version:  deviceplugin.intel.com/v1
Kind:         SgxDevicePlugin
Metadata:
  Creation Timestamp:  2021-03-30T11:00:34Z
  Generation:          1
  Managed Fields:
    API Version:  deviceplugin.intel.com/v1
    Fields Type:  FieldsV1
    fieldsV1:
      f:metadata:
        f:annotations:
          .:
          f:kubectl.kubernetes.io/last-applied-configuration:
      f:spec:
        .:
        f:enclaveLimit:
        f:image:
        f:initImage:
        f:provisionLimit:
    Manager:         kubectl-client-side-apply
    Operation:       Update
    Time:            2021-03-30T11:00:34Z
  Resource Version:  4500561
  UID:               3e219ebd-3521-4f68-8080-56d4ca4ca5d2
Spec:
  Enclave Limit:    110
  Image:            intel/intel-sgx-plugin:0.19.0
  Init Image:       intel/intel-sgx-initcontainer:0.19.0
  Provision Limit:  110
Events:             <none>

mythi, 2021-03-30:
> But is not creating any pod

Is the sgx device plugin daemonset created?  `kubectl get ds -A`

What's the status of it?

Ram-srini, 2021-03-31:
$ kubectl get ds -A
NAMESPACE                NAME                           DESIRED   CURRENT   READY   UP-TO-DATE   AVAILABLE   NODE SELECTOR              AGE
default                  sgxdeviceplugin-sample-5n5t7   1         1         0       1            0           kubernetes.io/arch=amd64   3m2s
kube-system              kube-flannel-ds                2         2         2       2            2           <none>                     45m
kube-system              kube-proxy                     2         2         2       2            2           kubernetes.io/os=linux     47m
node-feature-discovery   nfd-worker                     1         1         1       1            1           <none>                     8m26s

$  kubectl describe pod sgxdeviceplugin-sample-5n5t7-r56ln

Events:
  Type     Reason       Age                 From               Message
  ----     ------       ----                ----               -------
  Normal   Scheduled    115s                default-scheduler  Successfully assigned default/sgxdeviceplugin-sample-5n5t7-r56ln to gaia-desktop
  Warning  FailedMount  52s (x8 over 115s)  kubelet            MountVolume.SetUp failed for volume "sgx-enclave" : hostPath type check failed: /dev/sgx_enclave is not a character device
  Warning  FailedMount  52s (x8 over 115s)  kubelet            MountVolume.SetUp failed for volume "sgx-provision" : hostPath type check failed: /dev/sgx_provision is not a character device

@mythi One quick question. Does sgx device plugin operator needs to install SGX DCAP driver on the worker nodes? I am using NUC which is having isgx driver which support EPID based attestation.

mythi, 2021-03-31:
```
Warning FailedMount 52s (x8 over 115s) kubelet MountVolume.SetUp failed for volume "sgx-enclave" : hostPath type check failed: /dev/sgx_enclave is not a character device
Warning FailedMount 52s (x8 over 115s) kubelet MountVolume.SetUp failed for volume "sgx-provision" : hostPath type check failed: /dev/sgx_provision is not a character device
```

the plugin has a dependency to a driver that provides `/dev/sgx_[enclave|provision}` char devices. This is available in the latest DCAP driver release or in the 5.11 Linux with in-tree SGX.