intel/intel-device-plugins-for-kubernetes #618: Support for other FPGA vendor and product ids

rmr-silicom, 2021-04-13:
Within utils.go, only the intel vendor id is used. How can other vendor and product ids be added to support?

bart0sh, 2021-04-13:
FPGA plugin supports only 2 Intel FPGA cards at the moment:
- Intel Arria 10
- Intel Stratix 10

rmr-silicom, 2021-04-14:
Thanks for the response. I am adding support for a FPGA card. Another question I have...

https://github.com/OPAE/opae-sdk

What is the project difference? From my initial reading, this project and the opae-sdk kinda do the same thing.

bart0sh, 2021-04-14:
OPAE SDK is a set of libraries to work with FPGA. This project is a device plugin that enables various Intel accelerator devices in Kubernetes clusters. You can read about Kubernetes device plugins here: https://kubernetes.io/docs/concepts/extend-kubernetes/compute-storage-net/device-plugins/

rmr-silicom, 2021-04-15:
Got it. I am actually implementing an openshift operator for an FPGA PCIe device base on the Stratix 10, and have been pointed into the OPAE SDK direction, but it seems from a kubernetes point of view, this operator/golang pkg is where i should be using my time?

rojkov, 2021-04-15:
> it seems from a kubernetes point of view, this operator/golang pkg is where i should be using my time?

Yes, this project also provides a K8s operator for Intel FPGAs (and some other Intel HW). The latest release has been published at https://operatorhub.io/operator/intel-device-plugins-operator. Your contributions are welcome.

mythi, 2021-04-15:
> Your contributions are welcome.

@rmr-silicom or any feedback if the existing functionality does not meet your use-cases.

mythi, 2021-04-19:
@rmr-silicom did we answer your question and we can close this?

closed 2021-04-19 by rmr-silicom