intel/intel-device-plugins-for-kubernetes #633: Provide alternate image registry to dockerhub

onedr0p, 2021-05-07:
Hello,

In order to combat dockerhub's imposed rate limits it would be great to host these images on ghcr or quay.io. Are there on any thoughts from the community on this initiation?

mythi, 2021-05-10:
The device plugins release images are not updated that often and when combined with `IfNotPresent` (or even `Never`) `imagePullPolicy`, it should have minimal impact to your pull quota (the quota is IP based and not image based).

onedr0p, 2021-05-10:
Say you need to bootstrap a cluster with 10 nodes, each one using this Docker image. Now stack up all the other applications in your cluster also pulling from Dockerhub... it's VERY easy to hit their rate limits.

Yes there are ways to get around this on my side but it requires setting up a pull thru cache, but why not help solve the problem for everyone by pushing to an additional registry? GHCR is very easy to publish images to.

mythi, 2021-11-03:
Our device plugins images are now with the "verified publisher" badges and so free from any pull limits. Closing.

closed 2021-11-03 by mythi

onedr0p, 2023-01-23:
@mythi it looks like being part of the "Verified Publishers" only prevents your team from rate limits.

[image]