intel/intel-device-plugins-for-kubernetes #661: Missing init-container-tag, or wrongly renamed definition?

davralin, 2021-06-18:
Looking at [release-0.21/deployments/gpu_plugin/base/intel-gpu-plugin.yaml](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/release-0.21/deployments/gpu_plugin/base/intel-gpu-plugin.yaml#L18) for 0.21, I see both the regular, and the init-container was bumped to 0.21 - but there is no 0.21-tagged container on docker hub.

Is this a missing container, or a wrongly renamed tag?

For the record, it seems like everything works by editing the manifest with 0.20.0 for the initContainer.

mythi, 2021-06-19:
> but there is no 0.21-tagged container on docker hub.

`0.21.0` tags are now published.

closed 2021-06-19 by mythi

davralin, 2021-06-19:
Yeah, works.