intel/intel-device-plugins-for-kubernetes #667: OPAE v1.2.1 not working

stefanstoeckigt, 2021-06-24:
I cant seem to get the operator working with arria10 FPGA, opae 1.2.1 running interfaceId: 38d782e3b6125343b9342433e348ac4c. 

I created a new FPGARegion

```json

---
# DCP 1.2.1
apiVersion: fpga.intel.com/v2
kind: FpgaRegion
metadata:
  name: arria10.dcp1.2.1
  namespace: inteldeviceplugins-system
spec:
  interfaceId: 38d782e3b6125343b9342433e348ac4c

```

However the fpga-device-plugin fails to start

```
I0624 14:03:07.305399 1 fpga_plugin.go:360] FPGA device plugin (DFL) started in region mode
I0624 14:03:07.311300 1 manager.go:113] Received dev updates:{map[region-:map[dfl-fme.0:{Healthy [{/dev/dfl-port.0 /dev/dfl-port.0 rw {} 0}] [] map[] 0xc0003f9480}]] map[] map[]}
I0624 14:03:08.312724 1 server.go:248] Start server for region- at: /var/lib/kubelet/device-plugins/fpga.intel.com-region-.sock
E0624 14:03:08.316328 1 manager.go:140] Failed to serve fpga.intel.com/region-: rpc error: code = Unknown desc = the ResourceName "fpga.intel.com/region-" is invalid
Cannot register to kubelet service
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*server).registerWithKubelet
/go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/server.go:327
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*server).setupAndServe
/go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/server.go:260
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*server).Serve
/go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/server.go:191
github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin.(*Manager).handleUpdate.func1
/go/src/github.com/intel/intel-device-plugins-for-kubernetes/pkg/deviceplugin/manager.go:138
runtime.goexit
/usr/local/go/src/runtime/asm_amd64.s:1371
```

Is this not supported yet?

bart0sh, 2021-06-24:
@stefanstoeckigt thank you for reporting this issue.
It doesn't look like OPAE-related to me so far. Can you try to deploy FPGA plugin as a daemonset as its described [here](https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/cmd/fpga_plugin/README.md#deploying-as-a-daemonset)?

stefanstoeckigt, 2021-07-08:
@bart0sh I am getting the exact some issue when I deploy the FPGA plugin manually. 
So when I was running kernel 4.18, I was required to install the FPGA driver manually and I could get the operator working. 
However I needed to update the kernel to 5.4 which comes with repackaged driver, the operator stop working with the above error in the plugin device pod. 

Does the operator work with the dfl drivers?

stefanstoeckigt, 2021-07-08:
When i look at the ```/var/lib/kubelet/device-plugins/``` the following socket is created ```fpga.intel.com-region-.sock```. this seems to be incorrect. How does the plugin pod get the region id, to use in the socket creation?

bart0sh, 2021-07-15:
@seanporterwork sorry for the delay, I was on vacation. The plugin seems to use dfl driver on your system:
```
fpga_plugin.go:360] FPGA device plugin (DFL) started in region mode
```
I'd suggest to disable dfl kernel modules and use OPAE modules if you want to use OPAE in user space.

seanporterwork, 2021-07-15:
@bart0sh I'm guessing you meant to take @stefanstoeckigt :)

bart0sh, 2021-07-16:
@seanporterwork I'm sorry. thank you for pointing out!

stefanstoeckigt, 2021-07-19:
The issue with the OPAE driver is that it isn't compatible with kernel 5.4. So it seems that I am stuck with the dfl driver which I can't seem to get working with the kubernetes operator.

bart0sh, 2021-07-20:
Ah, I thought you're going to use OPAE driver.
Good news that the plugin works just fine on my Ubuntu 20.04 box with this kernel: 5.4.0-77-generic #86-Ubuntu SMP Thu Jun 17 02:35:03 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

Let's try to spot the difference in my and your configurations.
Can you show me the output of these commands:
```
lsmod  |grep dfl
ls -lR /sys/devices/*/*/*/fpga_region/
ls -la /dev/dfl*
```

stefanstoeckigt, 2021-07-22:
@bart0sh, many thanks for the help.

Please see below:  

##### ```lsmod  |grep dfl```
```
dfl_afu                32768  0
dfl_fme                24576  0
dfl_pci                16384  0
dfl                    28672  3 dfl_pci,dfl_fme,dfl_afu
fpga_region            20480  2 dfl_fme,dfl
fpga_bridge            20480  2 fpga_region,dfl_fme
fpga_mgr               16384  2 fpga_region,dfl_fme
```

##### ```ls -lR /sys/devices/*/*/*/fpga_region/```
```
'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/':
total 0
drwxr-xr-x 5 root root 0 Jul 22 16:10 region0

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0':
total 0
-r--r--r-- 1 root root 4096 Jul 22 16:10 compat_id
lrwxrwxrwx 1 root root    0 Jul 22 16:10 device -> ../../../0000:af:00.0
drwxr-xr-x 4 root root    0 Jul 22 16:10 dfl-fme.0
drwxr-xr-x 4 root root    0 Jul 22 16:10 dfl-port.0
drwxr-xr-x 2 root root    0 Jul 22 16:10 power
lrwxrwxrwx 1 root root    0 Jul 22 16:10 subsystem -> ../../../../../../class/fpga_region
-rw-r--r-- 1 root root 4096 Jul 22 16:10 uevent

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-fme.0':
total 0
-r--r--r-- 1 root root 4096 Jul 22 16:10 bitstream_id
-r--r--r-- 1 root root 4096 Jul 22 16:10 bitstream_metadata
-r--r--r-- 1 root root 4096 Jul 22 16:10 cache_size
-r--r--r-- 1 root root 4096 Jul 22 16:10 dev
lrwxrwxrwx 1 root root    0 Jul 22 16:10 driver -> ../../../../../../../bus/platform/drivers/dfl-fme
-rw-r--r-- 1 root root 4096 Jul 22 16:10 driver_override
drwxr-xr-x 2 root root    0 Jul 22 16:10 errors
-r--r--r-- 1 root root 4096 Jul 22 16:10 fabric_version
-r--r--r-- 1 root root 4096 Jul 22 16:10 modalias
-r--r--r-- 1 root root 4096 Jul 22 16:10 ports_num
drwxr-xr-x 2 root root    0 Jul 22 16:10 power
-r--r--r-- 1 root root 4096 Jul 22 16:10 socket_id
lrwxrwxrwx 1 root root    0 Jul 22 16:10 subsystem -> ../../../../../../../bus/platform
-rw-r--r-- 1 root root 4096 Jul 22 16:10 uevent

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-fme.0/errors':
total 0
-r--r--r-- 1 root root 4096 Jul 22 16:10 catfatal_errors
-r--r--r-- 1 root root 4096 Jul 22 16:10 first_error
-rw-r--r-- 1 root root 4096 Jul 22 16:10 fme_errors
-rw-r--r-- 1 root root 4096 Jul 22 16:10 inject_errors
-r--r--r-- 1 root root 4096 Jul 22 16:10 next_error
-r--r--r-- 1 root root 4096 Jul 22 16:10 nonfatal_errors
-rw-r--r-- 1 root root 4096 Jul 22 16:10 pcie0_errors
-rw-r--r-- 1 root root 4096 Jul 22 16:10 pcie1_errors

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-fme.0/power':
total 0
-rw-r--r-- 1 root root 4096 Jul 22 16:10 async
-rw-r--r-- 1 root root 4096 Jul 22 16:10 autosuspend_delay_ms
-rw-r--r-- 1 root root 4096 Jul 22 16:10 control
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_kids
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_enabled
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_status
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_suspended_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_usage

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-port.0':
total 0
-r--r--r-- 1 root root 4096 Jul 22 16:10 afu_id
-rw-r--r-- 1 root root 4096 Jul 22 16:10 ap1_event
-rw-r--r-- 1 root root 4096 Jul 22 16:10 ap2_event
-r--r--r-- 1 root root 4096 Jul 22 16:10 dev
lrwxrwxrwx 1 root root    0 Jul 22 16:10 driver -> ../../../../../../../bus/platform/drivers/dfl-port
-rw-r--r-- 1 root root 4096 Jul 22 16:10 driver_override
drwxr-xr-x 2 root root    0 Jul 22 16:10 errors
-r--r--r-- 1 root root 4096 Jul 22 16:10 id
-rw-r--r-- 1 root root 4096 Jul 22 16:10 ltr
-r--r--r-- 1 root root 4096 Jul 22 16:10 modalias
drwxr-xr-x 2 root root    0 Jul 22 16:10 power
-r--r--r-- 1 root root 4096 Jul 22 16:10 power_state
lrwxrwxrwx 1 root root    0 Jul 22 16:10 subsystem -> ../../../../../../../bus/platform
-rw-r--r-- 1 root root 4096 Jul 22 16:10 uevent
--w------- 1 root root 4096 Jul 22 16:10 userclk_freqcmd
--w------- 1 root root 4096 Jul 22 16:10 userclk_freqcntrcmd
-r--r--r-- 1 root root 4096 Jul 22 16:10 userclk_freqcntrsts
-r--r--r-- 1 root root 4096 Jul 22 16:10 userclk_freqsts

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-port.0/errors':
total 0
-rw-r--r-- 1 root root 4096 Jul 22 16:10 errors
-r--r--r-- 1 root root 4096 Jul 22 16:10 first_error
-r--r--r-- 1 root root 4096 Jul 22 16:10 first_malformed_req

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-port.0/power':
total 0
-rw-r--r-- 1 root root 4096 Jul 22 16:10 async
-rw-r--r-- 1 root root 4096 Jul 22 16:10 autosuspend_delay_ms
-rw-r--r-- 1 root root 4096 Jul 22 16:10 control
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_kids
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_enabled
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_status
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_suspended_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_usage

'/sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/power':
total 0
-rw-r--r-- 1 root root 4096 Jul 22 16:10 async
-rw-r--r-- 1 root root 4096 Jul 22 16:10 autosuspend_delay_ms
-rw-r--r-- 1 root root 4096 Jul 22 16:10 control
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_kids
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_active_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_enabled
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_status
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_suspended_time
-r--r--r-- 1 root root 4096 Jul 22 16:10 runtime_usage
```

##### ```ls -la /dev/dfl*```
```
crw------- 1 root root 236, 0 Jul 22 06:03 /dev/dfl-fme.0
crw------- 1 root root 235, 0 Jul 22 06:03 /dev/dfl-port.0
```

bart0sh, 2021-07-23:
@stefanstoeckigt I don't see dfl_fme_mgr and dfl_fme_br modules loaded in your. Here is my output:
```
# lsmod |grep dfl
dfl_fme_br             16384  0
dfl_fme_mgr            20480  1
dfl_fme_region         20480  0
dfl_afu                32768  0
dfl_fme                24576  0
dfl_pci                16384  0
dfl                    28672  4 dfl_pci,dfl_fme,dfl_fme_br,dfl_afu
fpga_region            20480  3 dfl_fme_region,dfl_fme,dfl
fpga_bridge            20480  4 dfl_fme_region,fpga_region,dfl_fme,dfl_fme_br
fpga_mgr               16384  4 dfl_fme_region,fpga_region,dfl_fme_mgr,dfl_fme
```

Can you also ` /opt/intel/fpga-sw/fpga_tool -d /dev/dfl-port.0 fpgainfo` this command and show its output here.
Here is what it outputs on my system:
```
# /opt/intel/fpga-sw/fpga_tool -d /dev/dfl-port.0 fpgainfo
//****** PORT ******//
Name                             : dfl-port.0
Device Node                      : /dev/dfl-port.0
SysFS Path                       : /sys/devices/pci0000:80/0000:80:01.0/0000:81:00.0/fpga_region/region0/dfl-port.0
PCIe s:b:d:f                     : 0000:81:00.0
Device Id                        : 0x8086:0x09c4
Device Class                     : 0x120000
Local CPUs                       : 1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33,35,37,39,41,43,45,47,49,51,53,55
NUMA                             : 1
SR-IOV Virtual Functions         : 0
SR-IOV maximum Virtual Functions : 1
FME Name                         : dfl-fme.0
Port Id                          : 0
Interface UUID                   : 9926ab6d6c925a68aabca7d84c545738
Accelerator UUID                 : f7df405cbd7acf7222f144b0b93acd18
Kernet API Version               : 0
Port Regions                     : 2
Port Region (Index/Size/Offset)  : 0 / 262144 / 0
Port Region (Index/Size/Offset)  : 1 / 4096 / 262144
```

stefanstoeckigt, 2021-07-23:
@bart0sh please see below.

##### ```/opt/intel/fpga-sw/fpga_tool -d /dev/dfl-port.0 fpgainfo```
```
//****** PORT ******//
Name                             : dfl-port.0
Device Node                      : /dev/dfl-port.0
SysFS Path                       : /sys/devices/pci0000:ae/0000:ae:00.0/0000:af:00.0/fpga_region/region0/dfl-port.0
PCIe s:b:d:f                     : 0000:af:00.0
Device Id                        : 0x8086:0x09c4
Device Class                     : 0x120000
Local CPUs                       : 1,3,5,7,9,11,13,15,17,19,21,23,25,27,29,31,33,35,37,39
NUMA                             : 1
SR-IOV Virtual Functions         : 0
SR-IOV maximum Virtual Functions : 1
FME Name                         : dfl-fme.0
Port Id                          : 0
Interface UUID                   :
Accelerator UUID                 : 66dae6f14f724fbaaeaa1b6facdfc39b
Kernet API Version               : 0
Port Regions                     : 2
Port Region (Index/Size/Offset)  : 0 / 262144 / 0
Port Region (Index/Size/Offset)  : 1 / 4096 / 262144
```

bart0sh, 2021-07-26:
@stefanstoeckigt Interface UUID is empty for some reason. That explains why plugin was not able to get it and failed with this error: ```E0624 14:03:08.316328 1 manager.go:140] Failed to serve fpga.intel.com/region-: rpc error: code = Unknown desc = the ResourceName "fpga.intel.com/region-" is invalid```

SR-IOV seems to be turned off, so theoretically you should have sysfs entries similar to what I have, i.e. 2 region dir  entries and compat_id file entry with non-zero interface id in it:
```
# ls -la /sys/class/fpga_region/
total 0
drwxr-xr-x  2 root root 0 Jun 23 19:45 .
drwxr-xr-x 70 root root 0 Jun 23 19:45 ..
lrwxrwxrwx  1 root root 0 Jun 23 19:47 region0 -> ../../devices/pci0000:80/0000:80:01.0/0000:81:00.0/fpga_region/region0
lrwxrwxrwx  1 root root 0 Jun 23 19:47 region1 -> ../../devices/pci0000:80/0000:80:01.0/0000:81:00.0/fpga_region/region0/dfl-fme.0/dfl-fme-region.2/fpga_region/region1

# cat /sys/class/fpga_region/region1/compat_id 
9926ab6d6c925a68aabca7d84c545738
```

First of all try to make sure all dfl modules are loaded. Second - I'd suggest to re-flash FPGA firmware on your device and cold reboot your machine.

bart0sh, 2021-07-26:
> I cant seem to get the operator working with arria10 FPGA, opae 1.2.1 running interfaceId: 38d782e3b6125343b9342433e348ac4c.

Can you explain how did you obtain this interface id?

bart0sh, 2021-08-20:
ok, closing the issue. Hopefully we've found the reason.

closed 2021-08-20 by bart0sh