intel/intel-device-plugins-for-kubernetes #681: qat kernel plugin doesn't work in VM

xxinran, 2021-08-31:
I am using qat kernel plugin in a **virtual machine** where I passthroughed a QAT PF.  
In VM, I can see the QAT is up,
```shell
$ sudo adf_ctl status
Checking status of all devices.
There is 1 QAT acceleration device(s) in the system:
qat_dev0 - type: c6xx,  inst_id: 0,  node_id: 0,  bsf: 0000:00:05.0,  #accel: 5 #engines: 10 state: up
```
But when I enter qat kernel plugin's container, the QAT device is down when I check with `adf_ctl status`, and the node id changes at each time I call `adf_ctl status` , another strange thing is the bsf is different from what I got from the sysfs, which is `0000:00:05.0`, the output is the following:
```shell
sh-5.1# adf_ctl status
Checking status of all devices.
There is 1 QAT acceleration device(s) in the system:
 qat_dev0 - type: ,  inst_id: 0,  node_id: -1406261856,  bsf: 0000:01:00.5,  #accel: 5 #engines: 10 state: down
sh-5.1# adf_ctl status
Checking status of all devices.
There is 1 QAT acceleration device(s) in the system:
 qat_dev0 - type: ,  inst_id: 0,  node_id: -1514506848,  bsf: 0000:01:00.5,  #accel: 5 #engines: 10 state: down
sh-5.1# adf_ctl status
Checking status of all devices.
There is 1 QAT acceleration device(s) in the system:
 qat_dev0 - type: ,  inst_id: 0,  node_id: -72010336,  bsf: 0000:01:00.5,  #accel: 5 #engines: 10 state: down
sh-5.1# adf_ctl status
Checking status of all devices.
There is 1 QAT acceleration device(s) in the system:
 qat_dev0 - type: ,  inst_id: 0,  node_id: 809616800,  bsf: 0000:01:00.5,  #accel: 5 #engines: 10 state: down
```
I think that's why my qat plugin doesn't discover the device, but I don't know how to fix it, could you please help on it?

mythi, 2021-08-31:
> I think that's why my qat plugin doesn't discover the device, but I don't know how to fix it, could you please help on it?

@xxinran what is the version of the QAT driver in your VM?

xxinran, 2021-09-01:
> > I think that's why my qat plugin doesn't discover the device, but I don't know how to fix it, could you please help on it?
> 
> @xxinran what is the version of the QAT driver in your VM?

@mythi  The versionfile is as the following: 
```shell

$ cat versionfile
PACKAGE_TYPE=QAT1.7

PACKAGE_OS=L

PACKAGE_VERSION_MAJOR_NUMBER=4

PACKAGE_VERSION_MINOR_NUMBER=14

PACKAGE_VERSION_PATCH_NUMBER=0

PACKAGE_VERSION_BUILD_NUMBER=00031
```
and the  kubernetes version is v1.22.1.

mythi, 2021-09-01:
@xxinran the driver had an ABI break and `adf_ctl`v4.11 (inside the container) fails to work with v4.14. See #666. If that works for you, maybe submit a PR for the qat kernel dockerfile too?

xxinran, 2021-09-01:
@mythi thanks for the information, I have tested with the latest driver and it works. please refer to https://github.com/intel/intel-device-plugins-for-kubernetes/pull/683

PR #683 (2021-09-01): Update qat driver in qat plugin image
this resolves the conflict in https://github.com/intel/intel-device-plugins-for-kubernetes/issues/681

Closes: #681
closed 2021-09-01 by mythi