intel/intel-device-plugins-for-kubernetes #706: Unable to build `openssl-qat-engine` container image: requires a shell more modern

devimc, 2021-09-15:
I'm using this Docker file https://github.com/intel/intel-device-plugins-for-kubernetes/blob/main/demo/openssl-qat-engine/Dockerfile

**docker build --rm -t openssl-qat-engine .**

```
Step 9/22 : RUN /bin/bash -c "sed -i -e 's/cmn_ko$//' -e 's/lac_kernel$//' quickassist/Makefile &&     KERNEL_SOURCE_ROOT=/tmp ./configure &&     make quickassist-all adf-ctl-all &&     install -m 755 build/libqat_s.so /usr/lib/ &&     install -m 755 build/libusdm_drv_s.so /usr/lib/ &&     install -m 755 build/adf_ctl /usr/bin/ &&     cd /ipp-crypto/sources/ippcp/crypto_mb &&     cmake . -B"../build"     -DOPENSSL_INCLUDE_DIR=/usr/include/openssl     -DOPENSSL_LIBRARIES=/usr/lib64     -DOPENSSL_ROOT_DIR=/usr/bin/openssl &&     cd ../build &&     make crypto_mb && make install &&     cd /intel-ipsec-mb &&     make && make install LIB_INSTALL_DIR=/usr/lib64"
 ---> Running in 560626e15fca
./configure: This script requires a shell more modern than all
./configure: the shells that I found on your system.
./configure: Please tell bug-autoconf@gnu.org and
./configure: qat-linux@intel.com about your system, including any
./configure: error possibly output before this message. Then install
./configure: a modern shell, or manually run the script under such a
./configure: shell if you do have one.
The command '/bin/sh -c /bin/bash -c "sed -i -e 's/cmn_ko$//' -e 's/lac_kernel$//' quickassist/Makefile &&     KERNEL_SOURCE_ROOT=/tmp ./configure &&     make quickassist-all adf-ctl-all &&     install -m 755 build/libqat_s.so /usr/lib/ &&     install -m 755 build/libusdm_drv_s.so /usr/lib/ &&     install -m 755 build/adf_ctl /usr/bin/ &&     cd /ipp-crypto/sources/ippcp/crypto_mb &&     cmake . -B"../build"     -DOPENSSL_INCLUDE_DIR=/usr/include/openssl     -DOPENSSL_LIBRARIES=/usr/lib64     -DOPENSSL_ROOT_DIR=/usr/bin/openssl &&     cd ../build &&     make crypto_mb && make install &&     cd /intel-ipsec-mb &&     make && make install LIB_INSTALL_DIR=/usr/lib64"' returned a non-zero code: 1
```

cc @mythi @eadamsintel

mythi, 2021-09-15:
@devimc hmm. I remember seeing something similar but cannot recall what is was. FWIW, we build this image part of our CI on Ubuntu 18. and 20.04 so it *should* be OK.

eadamsintel, 2021-09-16:
@devimc I just tried it on an Ubuntu 20.04 system and it built fine for me. I tried twice once with a 3 month old clearlinux:base image I had and once with the newer one 3 days ago and no issue.   I am using Docker 20.10.7 on an Ubuntu 20.04.2 system with a 5.4.0-81 kernel. 

There might be something interesting about your setup that would be worth knowing.  If you have time can you try running a base clearlinux docker image and then going through each of the steps manually until you get to step 9 and see if you can reproduce the same issue?   An alternative and maybe faster would be to set an environment variable in the Dockerfile  `ARG CONFIG_SHELL=/bin/bash` and try again.

devimc, 2021-09-20:
Fixed - I have updated my system (ubuntu 18.04) and rebooted it, thanks

closed 2021-09-20 by devimc