intel/intel-device-plugins-for-kubernetes #712: Scan yaml files with IaC security scanner tool https://www.checkov.io/

bart0sh, 2021-09-22:
This tool should help us to automate security checks for project yaml files.
It's integrated with github.com and can be configured to automatically scan github projects.

Here is an example of scanning demo/test-fpga-preprogrammed.yaml 
```
$ checkov -f ./test-fpga-preprogrammed.yaml 
Passed checks: 71, Failed checks: 18, Skipped checks: 0
Check: CKV_K8S_27: "Do not expose the docker daemon socket to containers"
	PASSED for resource: Pod.test-fpga-preprogrammed.default
	File: ./test-fpga-preprogrammed.yaml:1-20
	Guide: https://docs.bridgecrew.io/docs/bc_k8s_26

Check: CKV_K8S_18: "Containers should not share the host IPC namespace"
	PASSED for resource: Pod.test-fpga-preprogrammed.default
	File: ./test-fpga-preprogrammed.yaml:1-20
	Guide: https://docs.bridgecrew.io/docs/bc_k8s_17

Check: CKV_K8S_19: "Containers should not share the host network namespace"
	PASSED for resource: Pod.test-fpga-preprogrammed.default
	File: ./test-fpga-preprogrammed.yaml:1-20
	Guide: https://docs.bridgecrew.io/docs/bc_k8s_18
...

Check: CKV_K8S_23: "Minimize the admission of root containers"
	FAILED for resource: Pod.test-fpga-preprogrammed.default
	File: ./test-fpga-preprogrammed.yaml:1-20
	Guide: https://docs.bridgecrew.io/docs/bc_k8s_22

		1  | apiVersion: v1
		2  | kind: Pod
		3  | metadata:
		4  |   name: test-fpga-preprogrammed
		5  | spec:
		6  |   containers:
		7  |   - name: test-container
		8  |     image: intel/opae-nlb-demo:devel
		9  |     imagePullPolicy: IfNotPresent
		10 |     securityContext:
		11 |       capabilities:
		12 |         add:
		13 |           [IPC_LOCK]
		14 |     resources:
		15 |       limits:
		16 |         fpga.intel.com/arria10.dcp1.2-nlb3-preprogrammed: 1
		17 |         cpu: 1
		18 |         hugepages-2Mi: 20Mi
		19 | 
		20 |   restartPolicy: Never

...

```

hj-johannes-lee, 2022-02-03:
I made [a draft pr](#874) about this issue.
For now, I added only high severity ones.

Otherwise,, there are a few hundreds and thousands of fails.

bart0sh, 2022-02-06:
@hj-johannes-lee can you provide typical medium severity issues found by the tool? This would help us to decide whether we're going to fix them or not.

hj-johannes-lee, 2022-02-07:
@bart0sh 

Minimize admission of root containers (https://docs.bridgecrew.io/docs/bc_k8s_22)
- /demo/test-fpga-preprogrammed.yaml
- /demo/test-fpga-orchestrated.yaml
- /demo/intelfpga-job.yaml
- /demo/intelvpu-job.yaml
- ... total 81 files

Ensure containers do not share host process ID namespace (https://docs.bridgecrew.io/docs/bc_k8s_16)
Ensure containers do not share host IPC namespace (https://docs.bridgecrew.io/docs/bc_k8s_17)
- /deployments/gpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/dsa_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/vpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml

Ensure containers do not share the host network namespace (https://docs.bridgecrew.io/docs/bc_k8s_18)
- /deployments/gpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/dsa_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/vpu_plugin/overlays/namespace_kube-system/add-namespace-kube-system.yaml
- /deployments/sgx_enclave_apps/overlays/sgx_ecdsa_inproc_quote/add_hostnetwork.yaml
- /deployments/sgx_aesmd/base/kustomization.yaml
- /deployments/sgx_enclave_apps/overlays/sgx_ecdsa_inproc_quote/kustomization.yaml
- /deployments/sgx_aesmd/base/intel-sgx-aesmd.yaml

Ensure containers do not run with AllowPrivilegeEscalation (https://docs.bridgecrew.io/docs/bc_k8s_19)
- /deployments/sgx_admissionwebhook/manager/manager.yaml
- /deployments/sgx_admissionwebhook/base/manager_webhook_patch.yaml
- /deployments/sgx_plugin/overlays/epc-nfd/add-epc-nfd-initcontainer.yaml
- ... total 102 files

bart0sh, 2022-02-07:
@hj-johannes-lee thanks. From the first look all findings make sense and worth looking at.
@mythi wdyt?

mythi, 2022-02-08:
`bc_k8s_22`: the plugins cannot run as non-root users because they need to be able to write to kubelet device plugins socket. the webhooks run as user but we're going to be removing them too.

about the namespace checks: yes, worth checking. the current findings look strange to me: why it complains about overlay yamls that do not change much.

`bc_k8s_19`: worth checking but is this relevant if `bc_k8s_22` cannot be enabled.

Final thought: should we limit checks to `deployments/` only?

PR #906 (2022-03-01): implement terrascan check
Fixes: #712
closed 2022-03-01 by mythi